MarlinFirmware/Marlin #22260: AXIS4_NAME 'V'，G0 and G1 no move

diver197, 2021-06-29:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

#define LINEAR_AXES 4
#if LINEAR_AXES >= 4
  #define AXIS4_NAME 'V' // :['A', 'B', 'C', 'U', 'V', 'W']
#endif
When G28 is executed, all axes can go home, but when I axis is moved through G0 or G1, it cannot be moved

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin 2.0.9.1 (Jun 29 2021 14:05:30)

diver197, 2021-06-30:
In G2, the fourth axis is moving

ellensp, 2021-06-30:
Please attach your configuration files.

diver197, 2021-06-30:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6739762/Configuration.zip)

ellensp, 2021-06-30:
your config says   #define AXIS4_NAME 'A' // :['A', 'B', 'C', 'U', 'V', 'W']

ie A axis not U

diver197, 2021-06-30:
A and V, I've tried, the same result

ellensp, 2021-06-30:
what does M119 say, you seem to have a rather large number of endstops enabled

diver197, 2021-06-30:
[print_log.zip](https://github.com/MarlinFirmware/Marlin/files/6739980/print_log.zip)
This is my test log

ellensp, 2021-06-30:
cant even try and replicate, as you have modifed pins files....

diver197, 2021-07-01:
[pins_MKS_ROBIN_MINI.zip](https://github.com/MarlinFirmware/Marlin/files/6745910/pins_MKS_ROBIN_MINI.zip)

ellensp, 2021-07-02:
log does not show any G1 or G0  moves..

diver197, 2021-07-06:
Number of pulses emitted

DerAndere1, 2021-07-06:
@driver197 : Please change to #define AXIS4_NAME 'V' and upload Marlin again. 
Then connect to the printer via USB using a host software send the following G-codes sequence:
``` 
M502;
M500;
G28;
G1 V10;
```
Does the axis move after sending the last G-code command?

diver197, 2021-07-07:
> @diver197：请更改为#define AXIS4_NAME 'V' 并再次将 Marlin 上传到打印机。
> 然后使用主机软件（如 printrun 或 Repetier Host）通过 USB 连接到打印机，并发送以下 G 代码序列：
> 
> ```
> M502;
> M500;
> G28;
> G90;
> G1 V10;
> ```
> 
> 发送最后一个 G 代码命令后轴是否移动？

Using your method, G1 V10 I axis still does not move，
G1 X100 Y100
G2 X105 Y105 R5
I axis  move

ellensp, 2021-07-07:
V character not Y character

G1 X100 Y100 only moves X and Y axis no V axis
G2 X105 Y105 R5 only moves X and Y axis no V axis

diver197, 2021-07-07:
If it's right ：
G1 X100 Y100 moves X and Y axis，
G2 X105 Y105 R5 moves X and Y axis，
G1 V10    move Iaxis

My actual test results：
G1 X100 Y100 moves X and Y axis，ok
G2 X105 Y105 R5 moves X and Y axis，moves I axis ，error
G1 V10    not moves I axis，error

DerAndere1, 2021-07-07:
What happens if you disable ARC_SUPPORT (change `#definde ARC_SUPPORT` to `//#define ARC_SUPPORT` in Configuration_adv.h` and repeat my test?

DerAndere1, 2021-07-07:
Would be great if someone could extend ARC_SUPPORT to support more than 3 axes. Currently it looks like it is incompatible with LINEAR_AXES >= 4:
https://github.com/MarlinFirmware/Marlin/blob/304a926b0a2c5f9edb8adac93557758115d6b004/Marlin/src/gcode/motion/G2_G3.cpp#L57-L67

diver197, 2021-07-07:
G1 V10 not moves I axis

DerAndere1, 2021-07-07:
Please ask for help with configuration in the Marlin Facebook group or elsewhere on the internet. Your pins_MKS_ROBIN_MINI.h file has many pins assigned to multiple functionalities. That does not make sense for the *_STEP_PIN and *_DIR_PIN. E.g. PE4  is also used as FAN_PIN in cour pin file.

diver197, 2021-07-09:
Adding serial port printing, it is found that the v-axis value is not extracted when the axis target value is extracted
[image]

ellensp, 2021-07-09:
You say command was G1 X10 Y10 Z10 V10, but can see command line N148 is only G1 X10 Y10 Z10
So i'm confused.

diver197, 2021-07-09:
I guess this is G0G1, which has filtered out micro walk when parsing发自我的华为手机-------- 原始邮件 --------发件人： ellensp ***@***.***>日期： 2021年7月9日周五 18:12收件人： MarlinFirmware/Marlin ***@***.***>抄送： diver ***@***.***>, Mention ***@***.***>主    题： Re: [MarlinFirmware/Marlin]  AXIS4_NAME 'V'，G0 and G1 no move (#22260)
You say command was G1 X10 Y10 Z10 V10, but can see command line N148 is only G1 X10 Y10 Z10
So i'm confused.

ellensp, 2021-07-09:
where are half of the commands?, what are you using to send, can you try pronterface for a test?

diver197, 2021-07-09:
I use MC hair, also use serial assistant hair, vaaa on the front, middle, back are not发自我的华为手机-------- 原始邮件 --------发件人： ellensp ***@***.***>日期： 2021年7月9日周五 18:23收件人： MarlinFirmware/Marlin ***@***.***>抄送： diver ***@***.***>, Mention ***@***.***>主    题： Re: [MarlinFirmware/Marlin]  AXIS4_NAME 'V'，G0 and G1 no move (#22260)
where are half of the commands?, what are you using to send, can you try pronterface for a test?

closed 2021-07-16 by diver197

thinkyhead, 2021-07-17:
Extending arc support shouldn't be too hard.￼ The arc should be done with the two axes in the workspace plane, while all the others that are moving just get linear-interpolated.