MarlinFirmware/Marlin #22261: [BUG] Marlin 2.0.9.1 not priming after retractions

EricSchubert, 2021-06-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin 2.0.9.1 does not reprime nozzle after retractions.  After the nozzle moves, it continues extruding normally, it just does not reprime.

### Bug Timeline

Worked in 2.0.5.3, noticed in 2.0.8 and 2.0.9.1

### Expected behavior

If retraction happens, extruder should prime after the move is complete.

### Actual behavior

- Z-hop + Retraction happens
- Nozzle moves to new position and lowers to continue printing
- Extruder does not reprime.  It simply continues extruding as if already primed, resulting in underextrusion after retractions happen.

### Steps to Reproduce

1. Slice model with retraction and z-hop enabled.
2. Print model and observe extruder during moves.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Creality Ender 3

### Electronics

SKR Mini E3 v1.2 mainboard

### Add-ons

Mosquito hot end, Bondtech BMG-M extruder

### Your Slicer

Other (explain below)

### Additional information & file uploads

Slicers tested: Fusion 360 and Cura 4.9.1.  Both exhibited this issue.

Reverted back to Marlin 2.0.5.3 and issue disappeared.

Simple example g-code file attached.

[Limit Switch Bracket.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/6731376/Limit.Switch.Bracket.gcode.txt)

borland1, 2021-07-02:
Have you verified correct behavior (re-priming) when Z-hop feature is disabled?

EricSchubert, 2021-07-02:
I believe it is still not priming after retractions, even without z-hop.  I
created a test print and sliced it in Fusion 360, disabling their features
"Enable Rapid Z Lift" and "Rapid Lift Between Layers".  The nozzle no
longer lifts between layers, nor does it lift during moves.  Retractions
still appear to happen, but I'm not noticing priming happening.
Interestingly, I think I see a retraction and re-prime between layer changes

I shot an HD video at 60fps and am uploading now.  Will post that when it's
ready for review.  It might be easier to see what's going on when we slow
things down.  My retractions are so short and quick that they can be tough
to see in real-time.

borland1, 2021-07-02:
OK. Might try testing with different Slicer than Fusion 360's.  My printer has Titan 3:1 geared extruder so retraction/priming is easy to spot.  I don't see what you are observing on my printer, but I have not enabled Z-hoping (leadscrew has anti-backlash nut and zero-backlash bellows coupler), however do have mesh bed leveling enabled.

EricSchubert, 2021-07-02:
As I stated in my original write-up, I've tested this retraction behavior
in both Cura and Fusion 360.  I'll try this again, outputting g-code from
Cura instead.

I also enabled mesh bed leveling in my Marlin config, so I don't think
that's a factor.

For reference, I've already calibrated steps/mm for all 4 axes, as well as
calibrated flow rate.  Calibration was done when retraction worked fine in
Marlin 2.0.5.3.

EricSchubert, 2021-07-02:
Here's a video of my Fusion 360 model printing.  No z-hop enabled.  Retraction is 0.4mm @ 15mm/s.
https://youtu.be/4Ok9CKsJiMg

Same model sliced in Cura with same settings.  Retraction 0.4mm @ 15mm/s, "Z-hop when retracted" disabled, combing disabled:
https://youtu.be/CX1lbzGq2zY

Initially, I thought maybe the gears in the extruder had backlash, causing it to just not prime again when the extruder reversed directions.  But moving the extruder manually in the menu, I don't see any issues reversing direction.  Backlash appears nonexistent in that scenario.

borland1, 2021-07-02:
For your bondtech geared direct drive extruder, retraction/re-priming is very small and thus fast, which is what you should expect.  Setting the YouTube playback speed to 0.25, allow one to view the retraction/priming with more clarity.

After viewing your printer in slow motion for both Fusion360 and Cura slicers, retraction/re-priming looks normal to me.

Must be some other problem, like wet filament, clogged nozzle, wrong flow multiplier, wrong nozzle temperature, or wrong first layer z-offset.

EricSchubert, 2021-07-03:
I see several moves in both videos that show a clear retraction, but no
priming prior to regular printing again.

If you look at this instance right here:  https://youtu.be/CX1lbzGq2zY?t=52

Watch at 0.25x speed.  The amount of retraction is clearly larger than any
priming happening immediately afterwards.

EricSchubert, 2021-07-03:
Another short clip showing moves without repriming.
https://youtu.be/UIX1M-mFc2o

Another image showing the effects of this, as well.
[image]

EricSchubert, 2021-07-04:
Just a quick update:

I lost my build of Marlin 2.0.5.3, as it was overwritten when I grabbed the latest 2.0.x data for 2.0.9.1.  However, I still had a backup copy of 2.0.5.3.  I rebuilt my configuration and compiled it, then uploaded to my printer.  It's running smoothly through its first layer.  Retractions appear to be working just fine again, and moves produce clean, correctly-extruded lines after repriming.

Unless I botched something up in 2.0.9.1, which I don't believe I did as I checked over various settings multiple times, I'm still convinced there's an issue in the latest builds and how they handle retractions.

Scope666, 2021-07-04:
I had problems with 2.0.9.1 also, the 2 biggest being it simply stops extruding at some point during the print, and pauses while printing from Octoprint... rolling back to 2.0.7.2 fixed them both for me.

thinkyhead, 2021-07-30:
Is this related to `G10` / `G11` ? The slicer is responsible for retraction and priming otherwise.

EricSchubert, 2021-07-30:
To be honest, I don't know for sure.  But looking at the g-code sample, it
appears that retractions and primes are in the code.

As mentioned previously, this has not been an issue in 2.0.5.3.  I only ran
into it once I hit 2.0.8 and 2.0.9.1.  Both versions have been behaving
this way on my machine.

thinkyhead, 2021-07-30:
So, the problem appears somewhere in the span of time between 2.0.5.3 and 2.0.8. We will have to try and narrow it down further…. I would rather try to find the commit that broke it than go on a hunt for the bug by running tests and studying the code, especially since there is a strong possibility that this is hardware-dependent or due to interaction with some other feature. If the retract and prime is in the G-code then maybe it is a result of `FWRETRACT_AUTORETRACT`, or maybe it is a result of some stepper direction issue….

If you are able, please try to narrow down as close as you can to the day and hour when the bug first appears in the `bugfix-2.0.x` branch. The technique to do this is to test code from a point half-way in-between the last-known-working date and the first-known-broken date until the exact commit is discovered. This can take as few as three tests, but will usually require no more than 8 or 9 tests.

spaaghetti, 2021-08-20:
Update on my issue. It is now resolved. The issue was actually incorrect Vref on my mainboard (Creality 4.2.2). I lowered it from the factory setting (1.38V) to the TMC recommended setting (0.85V) on the extruder axis. I have not experienced a failed print since that change.

This was previously not translating to failed prints because I previously had a stock Creality extruder. I recently swapped to a Titan clone extruder. The increased esteps were causing an overheat condition (presumably) in either the stepper or driver. This was hard to detect because the extruder would simply stop without any clog condition and restart on a subsequent print without issue. It appeared to be firmware related. Sorry to have cluttered this issue.

closed 2022-04-12 by github-actions[bot]