MarlinFirmware/Marlin #22263: [BUG] MMU2S on SKR Pro and Sensorless homing

Minims, 2021-06-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When enabling MMU2S on SERIAL 3 or 6, the sensorless homing ( X & Y) on a SKR Pro 1.1 with TMC2209 fails.
When home X run, it bumps but don't stop.
Same thing for home Y.

If I disable MMU2S in the configuration, sensorless homing works again.

This setup was working few months ago, but since the last bugfix2.0 I can't make it works. 
Maybe it can be the 3rd serial implementation. Don't what to check.

### Bug Timeline

New Bug, Discover in latest bugfix2.0, was working a the beginning of the year.

### Expected behavior

Sensorless homing working with MMU2S enabled.

### Actual behavior

Sensorless homing fails with MMU2S enabled.

### Steps to Reproduce

Compile & Upload.
Reset EEPROM to default.
HOME X
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/6735775/Archive.zip)

### Version of Marlin Firmware

Latest bugfix (29/06/2021) 

### Printer model

BLV Cube 

### Electronics

BTT SKR Pro 1.1 , TMC2209

### Add-ons

MMU2S

### Your Slicer

Prusa Slicer

JohnnyTheOne, 2021-06-29:
Same issue [#22012](https://github.com/MarlinFirmware/Marlin/issues/22012)

ellensp, 2021-06-30:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22012

closed 2021-06-30 by ellensp