MarlinFirmware/Marlin #22265: SanityCheck warning errors running build for LPC1768 and LPC1769

xeddog, 2021-06-29:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I get a large amount of warning messages when running build for LPC1768 and LPC1769 processors.  This is related to #22259.  Line 3595 of SanityCheck.h looks like:

#error "SLIM_LCD_MENUS disables "Advanced Settings > Probe Offsets > PROBE_OFFSET_WIZARD."

It looks like it is missing the quote marks after SLIM_LCD_MENUS.  I added the closing quotes and reran the builds and they ran clean with no warning.

### Bug Timeline

new today after running pull

### Expected behavior

Build without the warnings

### Actual behavior

Many warning messages during build

### Steps to Reproduce

Using the latest VSCode, I Ran Clean followed by Build. 

### Version of Marlin Firmware

2.0.9.1

### Printer model

Monoprice Mini and Creality CR-10S

### Electronics

Mini running the SKR V1.3 and the Creality running the SKR V1.4

### Your Slicer

Cura

### Host Software

OctoPrint

slowbro, 2021-06-29:
Looks like this was just fixed a few minutes ago.

https://github.com/MarlinFirmware/Marlin/commit/cd01421ac32041c7f775ec37dd8d00b29a5d335b

xeddog, 2021-06-29:
I see that.  Not the way I did it so I need to go back and fix my fix.  :-)

thisiskeithb, 2021-06-29:
Patched in cd01421.

closed 2021-06-29 by thisiskeithb