MarlinFirmware/Marlin #22267: [BUG] E driver error detected / coil short circuit

rick-ryerson, 2021-06-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

About 70% into the first layer, my print halts with a driver error detected. This happens every time. I reset the printer and try again.

Attached is the output from OctoPrint

[octopi-output.txt](https://github.com/MarlinFirmware/Marlin/files/6737792/octopi-output.txt)

Lines 238 and 239 show the error message.

### Bug Timeline

new

### Expected behavior

I expected it to complete the print.

### Actual behavior

The printer halted with "driver error detected" shown on the screen.

### Version of Marlin Firmware

Marlin bugfix-2.0.x (Jun 29 2021 22:46:08) 

### Printer model

Ender 3 Pro

### Electronics

SKR 1.4 Turbo

### Add-ons

dual Z motors (single driver output)

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[octopi-output.txt](https://github.com/MarlinFirmware/Marlin/files/6737812/octopi-output.txt)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6737814/Configuration.zip)

thisiskeithb, 2021-06-30:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-06-30 by thisiskeithb

rick-ryerson, 2021-06-30:
I'm not looking for support. Looking at commits, this is has happened in the past. https://github.com/MarlinFirmware/Marlin/pull/20731/commits

Dist0rtion, 2021-07-05:
I have the same error with the latest bugfix 2.0x
I've reverted back to Marlin 2.0.8.2 - this Version runs without problems

sjirafje, 2021-08-05:
I have the same problem with SKR 1.4 and Marlin 2.0.9.1

rick-ryerson, 2021-08-05:
> I have the same problem with SKR 1.4 and Marlin 2.0.9.1

> I have the same error with the latest bugfix 2.0x - Got also an SKR 1.4 Turbo with TMC2208
> I've reverted back to Marlin 2.0.8.2 - this Version runs without problems

Notice this issue is closed.

Go write your comments on 
https://github.com/MarlinFirmware/Marlin/issues/22312