MarlinFirmware/Marlin #22270: [FR] Create a G-code command for a relative increase in feedrate or flowrate percentage

Wreck-3D, 2021-06-30:
### Is your feature request related to a problem? Please describe.

Sometimes when making a 3D print, the exact flow rate or feed rate you want may not be known ahead of time, especially if you're using a new brand of filament or even a new roll of filament you're used to. Therefore you may have to adjust the flow rate or feed rate via the tuning feature at the beginning of your print to make sure its extruding/printing properly.

Additionally, I personally find myself wanting to reduce or increase the feed rate or flow rate for certain parts of my prints. So for example, if the top part of a print has more intricate details, that need to be printed slower, or need to be printed at a higher or lower flow rate to ensure proper details, then I will generally change the feed rate or flow rate by hard coding it into the G-code using M220, or M221 because the slicer may not have the option to do something like that. 

Now say if I know I want the upper portion to be extruded at a flow rate 5% higher than the bottom, and I then set that in the G-code file with M221 S105. However, at the same time, I find at the beginning of the print that it is under-extruding, and I end up increasing the flow to 107% to get proper extrusion. Well now, when it gets to the M221 S105 G-code, it will end up decreasing the flow by 2% rather than increasing it by 5% like I had desired. I would like the ability to make a relative increase/decrease the feed rate or flow rate rather than only being able to set an absolute value.

### Describe the feature you want

I would like if there was a G-code command to perform a relative change in the feed rate or flow rate percentage. For example, currently the commands M220, and M221 exists to set the absolute feed rate or flow rate. So if you send M221 S108; it will set the flow rate to 108%. However, what I am asking for is a command to make a relative increase, so for example, if the flowrate was already set to 103%, you could send a command M### S8 and it would increase the current flow rate by 8% to 111%, or you could send M### S-3 to reduce the flow rate by 3% to 100%. A similar thing with would work with feed rate also.

ellensp, 2021-06-30:
https://marlinfw.org/docs/gcode/M220.html

has been part of marlin since near the start

ellensp, 2021-06-30:
never mind....  miss read it.

ellensp, 2021-06-30:
This really doesn't need a new gcode though, just new switch so that value is interpreted as relative. It a shame R is already taken

ellensp, 2021-06-30:
Also do you realize when on the main status screen of most lcd's if you rotate the knob left or right you increase and decrease the federate?

Wreck-3D, 2021-07-01:
Yes I know that you can do it manually on the screen. I was looking for a way to do it through the GCode though. If I want to increase the flow rate by 10% on the last layer, I would have to be standing there watching it print. If I could do it through the GCode, then I could just add it to the beginning of the last layer.

You make a good point about just needing a new switch though, I never even thought of that.