MarlinFirmware/Marlin #22272: [BUG] (bug summary)

altometer, 2021-06-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin fails to build for the Bigtree Tech SKR 2 using the ender 5 plus pre-config.

### Bug Timeline

new

### Expected behavior

Marlin should compile without errors

### Actual behavior

```In file included from Marlin/src/lcd/extui/dgus/origin/../../../../inc/MarlinConfigPre.h:37,
                 from Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp:27:
Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp:211:16: error: 'VP_Fan2_Percentage' was not declared in this scope; did you mean 'VP_Fan1_Percentage'?
...
...
*** [.pio/build/BIGTREE_SKR_2/src/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp.o] Error 1
Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp: In static member function 'static void DGUSScreenHandler::DGUSLCD_SD_ResumePauseAbort(DGUS_VP_Variable&, void*)':
Marlin/src/lcd/extui/dgus/origin/DGUSScreenHandler.cpp:88:20: error: 'MKSLCD_SCREEN_PAUSE' was not declared in this scope; did you mean 'DGUSLCD_SCREEN_UNUSED'?
   88 |         GotoScreen(MKSLCD_SCREEN_PAUSE);
      |                    ^~~~~~~~~~~~~~~~~~~
      |                    DGUSLCD_SCREEN_UNUSED

```

### Steps to Reproduce

Environment:

- Visual Studio Code 1.57.1
- Marlin Firmware Auto Build 2.1.34
- POP OS 20.04(ubuntu Linux) Host OS (reproduced on a windows machine)

1. Git clone current Marlin build
2. Download the config examples
3. overwrite Configuration.h and Configuration_adv.h with examples for (Ender-5 Plus Configurations for Marlin Firmware)
4. Fix incorrect setting in Configuration.h Line 808, `#define Z_MIN_ENDSTOP_INVERTING true` must be set to false due to BLTOUCH sensor
5. Click build, errors are produced.

Workaround to successfully build:

Alter  Marlin/src/lcd/extui/dgus/origin/DGUSDisplayDef.cpp to change the // Fan Data if statement for FAN_VPHELPER(N)
```
// Fan Data
  #if HAS_FAN
    #define FAN_VPHELPER(N) \
      VPHELPER(VP_Fan1_Percentage, &thermalManager.fan_speed[N], ScreenHandler.DGUSLCD_PercentageToUint8, ScreenHandler.DGUSLCD_SendPercentageToDisplay), \
      VPHELPER(VP_FAN1_CONTROL, &thermalManager.fan_speed[N], ScreenHandler.HandleFanControl, nullptr), \
      VPHELPER(VP_FAN1_STATUS, &thermalManager.fan_speed[N], nullptr, ScreenHandler.DGUSLCD_SendFanStatusToDisplay),
    REPEAT(FAN_COUNT, FAN_VPHELPER)
  #endif
```

### Version of Marlin Firmware

2.0.9.1

### Printer model

Creality Ender 5 Plus

### Electronics

BIGTREE_SKR_2

### Add-ons

N/A

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6742112/Configuration.zip)

slowbro, 2021-06-30:
Hmm, interesting. I see `MKSLCD_SCREEN_PAUSE` used in most of the other non-mks DGUS displays, but those displays all don't have a PAUSE screen, afaik. Maybe it was copy/pasted from the MKS one..? Seemingly, all the 'MKS*' things would need replacing in the other non-MKS DGUS files... but I'm not certain of that.

ellensp, 2021-07-01:
Ender 5 plus LCD has been broken a long time.

Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21576

closed 2021-07-01 by ellensp