MarlinFirmware/Marlin #22275: [BUG] error: 'MarlinBusyState' in 'class GcodeSuite' does not name a type

Andigallone, 2021-07-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

c:\temp\arduino_build_156155\sketch\src\lcd\extui\ui_api.h:82:15: error: 'MarlinBusyState' in 'class GcodeSuite' does not name a type
   GcodeSuite::MarlinBusyState getMachineBusyState();
               ^~~~~~~~~~~~~~~
exit status 1
Fehler beim Kompilieren für das Board Anet V1.0.

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Printer model

Anet A8

### Electronics

Stock; Anet A8 V1.0

### Your Slicer

Cura

ellensp, 2021-07-01:
Builds fine on platformio issue is only in Arduino IDE 
some code for TOUCH_UI_FTDI_EVE is being complied unconditional.

ellensp, 2021-07-01:
This seems to fix it
```DIFF
diff --git a/Marlin/src/lcd/extui/ftdi_eve_touch_ui/compat.h b/Marlin/src/lcd/extui/ftdi_eve_touch_ui/compat.h
index dd25af1e74..22f49a2565 100644
--- a/Marlin/src/lcd/extui/ftdi_eve_touch_ui/compat.h
+++ b/Marlin/src/lcd/extui/ftdi_eve_touch_ui/compat.h
@@ -21,6 +21,9 @@
 
 #pragma once
 
+#include "../../../inc/MarlinConfigPre.h"
+
+#if ENABLED(TOUCH_UI_FTDI_EVE)
 /**
  * This following provides compatibility whether compiling
  * as a part of Marlin or outside it
@@ -51,3 +54,5 @@
 class __FlashStringHelper;
 typedef const __FlashStringHelper *progmem_str;
 extern const char G28_STR[];
+
+#endif  // TOUCH_UI_FTDI_EVE

```

ellensp, 2021-07-01:
created a PR with the above patch to fix this.

closed 2021-07-01 by thinkyhead