MarlinFirmware/Marlin #22280: SKR 2 USB MSC

dotdash32, 2021-07-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Based on #17222, it would see that SDIO MSC/CDC composite support is availabe for STM32 boards.  Based on the example in `[env:STM32F407VE_black]`, I tried to copy the environment for the SKR 2 (STM32F407VGT6).

However, it's not working my the simple 512mb card.  Will follow up with testing larger SD cards, but it usually seems that smaller cards are better?

### Bug Timeline

New

### Expected behavior

2 USB devices become available on the computer, one for CDC serial and one MSC mass storage.

### Actual behavior

Only the CDC device appears.

### Steps to Reproduce

1. Set SKR 2 as mainboard
2. create "USB" env with additional build flags
```
# add USB??
[env:BIGTREE_SKR_2_USB]
extends             = env:BIGTREE_SKR_2
platform            = ${common_stm32.platform}
platform_packages   = ${stm_flash_drive.platform_packages} 
build_unflags       = ${common_stm32.build_unflags} 
  ; -DUSBD_USE_CDC
build_flags         = ${env:BIGTREE_SKR_2.build_flags} ${stm_flash_drive.build_flags}
  -DUSBCON
  -DUSBD_USE_CDC_COMPOSITE
  -DUSE_USB_FS
```
3. Load and check USB devices

### Version of Marlin Firmware

2.0.9.1

### Printer model

Ender 3

### Electronics

SKR 2 Rev B

### Additional information & file uploads

Additional question: what file needs to be modified for a new environment to show up in ABM?
[SK2_USB.zip](https://github.com/MarlinFirmware/Marlin/files/6751423/SK2_USB.zip)

thisiskeithb, 2021-07-02:
You don't need to create a new environment to use USB MSC. You just need to enable the correct options.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

dotdash32, 2021-07-02:
Will ask on Discord, but based on [issues on SKR2 github](https://github.com/bigtreetech/SKR-2/issues/26), it is not supported.  If it is, I will report this back.

dotdash32, 2021-07-08:
Bumping this.  Asked in Discord, and according to [this](https://discord.com/channels/461605380783472640/491165528295997450/860357860561780757) from @rhapsodyv, there are still some changes pending.  There are others who are experiencing the [same thing](https://discord.com/channels/461605380783472640/491105274464043026/862839482297352232) in `#bug-reports` without any feedback from BTT.

Can this be reopened until these changes have been fixed?  Or at least hear a little more about what the missing issues are?

thisiskeithb, 2021-07-14:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22354

closed 2021-07-14 by thisiskeithb