MarlinFirmware/Marlin #22281: Wont flash to Ender 6 just stays on opening screen

sparksplur, 2021-07-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

So i cant get the 2.0 bugfix to flash, ive copied the bin even copied the dwinset from the most currennt creality firmware but i cant get it to flash with the sd card... it just turns on and then never loads?

### Bug Timeline

new

### Expected behavior

to flash the firmware

### Actual behavior

nothing is happening

### Version of Marlin Firmware

2.0 bugix

### Printer model

ender 6

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

Cant figure out why its not flashing

thisiskeithb, 2021-07-02:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/21274

closed 2021-07-02 by thisiskeithb

sparksplur, 2021-07-02:
Its not a duplicate and that solution also doesnt work it says file not found