MarlinFirmware/Marlin #22282: Solution for Cant falsh ender 6 doesnt work... adding ../ to the ccp makes is too many ../'s and will complie error

sparksplur, 2021-07-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When you try to flash marlin to ender 6 u just get stuck on loading screen

### Bug Timeline

still same issue

### Expected behavior

expect it to load

### Actual behavior

doesnt load

### Version of Marlin Firmware

Newest bug fix 2.0

### Additional information & file uploads

You marked it as duplicate but it wasnt... also in the solution you linked where your adding a extra ''../'' for the feature/powerloss.h'' will just yieild a cannot find source file. So that solution also doesnt work.

thisiskeithb, 2021-07-02:
Duplicate of #21274

thisiskeithb, 2021-07-02:
Please stop creating new issues. Ender-6 discussions should happen in the linked issue.

closed 2021-07-02 by thisiskeithb

sparksplur, 2021-07-02:
ohhhh sorry XD i didnt realize what was going on