MarlinFirmware/Marlin #22283: [BUG] No response on SKR v1.3 with SERIAL_PORT 0 and BAUDRATE > 250000

mihai-dumitresq, 2021-07-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Have Marlin on SKR v1.3 (uses LPC1768) and configure `SERIAL_PORT 0`  and `BAUDRATE > 250000`
Then I can only receive marlin's output over UART but any commands sent don't reach marlin (behaviour is similar to a disconnected RX0 on SKR 1.3). However comms over USB CDC (i.e. SERIAL_PORT_2 -1) work fine.

NOTE I have a working fix for this bug:
https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.0.x...mihai-dumitresq:skr-1.3-baud
I'll make a pull request but note that while it works and
`CLKPWR_SetPCLKDiv(CLKPWR_PCLKSEL_UART0, CLKPWR_PCLKSEL_CCLK_DIV_1);`
is correct I'm not sure of where to best place it.

It was developed by reading the troubleshooting in:
https://community.nxp.com/t5/LPCXpresso-IDE/LPC1768-uart-using-FDR-at-higher-baud-rates/m-p/544083
and trying to understand the changes in https://github.com/MarlinFirmware/Marlin/pull/19945/commits/dd10a9224bba9a6d267e5c17333a5cad624f9742

Based on the fix I assume SKR v1.4 (also LPC1768) to be affected too. Maybe also SKR v1.4 Turbo (LPC1769).

### Bug Timeline

Bug is on latest 2.0.x and bugfix-2.0.x but it was also on 2.0.5.3 IIRC. Reporting now after investigating the issue in detail.

### Expected behavior

M105 and other commands work.

### Actual behavior

No response to any commands.

### Steps to Reproduce

1. Have SKR v1.3
2. Connect using USB-TTL to 5V, GND, TX0, RX0 as per https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/140
3. Set a BAUDRATE > 250000, i.e. tried 460800, 500000, 921600, 1000000
4. Build (board = nxp_lpc1768) and flash.
5. Listen in using the baudrate picked: picocom --quiet --echo --imap lfcrlf -b 921600 /dev/ttyUSB3
6. Power up the board and notice Marlin output:

``` 
start
PowerUp
Marlin 2.0.9.1

echo: Last Updated: 2021-06-27 | Author: Mihai
echo:Compiled: Jun 30 2021
echo: Free Memory: 11703  PlannerBufferBytes: 1344
echo:V83 stored settings retrieved (633 bytes; crc 39877)
[...]
```
7. Type in commands.

### Version of Marlin Firmware

Marlin 2.0.9.1

### Printer model

Ender 3

### Electronics

SKR 1.3

### Host Software

SD Card (headless)

mihai-dumitresq, 2021-07-02:
I've now tested SKR v1.4 Turbo / LPC1769 and turns out it's only somewhat affected:

- on bugfix-2.0.x BAUDRATEs of 115200, 250000, 460800, 500000, 921600 work. BAUDRATE 1000000 does not work
- with pull/22284, BAUDRATEs of 57600, 115200, 250000, 460800, 500000, 921600, 1000000, 2000000 (!) work

thisiskeithb, 2021-07-07:
> Will be fixed in the LPC176x framework

https://github.com/MarlinFirmware/Marlin/pull/22284#issuecomment-875721468

closed 2021-07-07 by thisiskeithb