MarlinFirmware/Marlin #22289: Marlin 2.0.7+ Messure wrong Hotend Temperature with MAX31855 Thermocouple K-Type (~30-35°C below)

qrstu, 2021-07-03:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Same behavior 2.0.7.2 and 2.0.9
Used different K-Type Thermocouples from different Vendors and also different MAX31855 Boards.
Same issue.
Connected to BTT 1.4 Turbo SPI PIN0.26, PIN0.7, PIN0.8 --> Communication works fine.
Temperatures are about 35°C below as displayed on Marlin LCD.
If the temperature is 100°C difference is about -25°C
Cross checked with basic Arduino Nano Sketch MAX31855 from Adafrut Library with Serial connection and also IR Thermometer --> Same values (+/- 5°C) --> It can't be the hardware !!
For me it looks that some where a hidden temperature correction is done. Like for termistors.h --> But I can't find.
Or is there also a template for thermocouples hat has to be applied to get correct values ?

**Marlin-2.0.9\Marlin\src\pins**
  #define TEMP_0_PIN P0_26  		// 2020-10-09 MR Pin for Thermocouple K-Type MAX31855
//#endif
//#ifndef MAX6675_SS_PIN			//2020-10-09 MR define Thermocouple
  #define MAX6675_SS_PIN TEMP_0_PIN   //2020-10-09 MR define Thermocouple
  #define MAX31855_SS_PIN TEMP_0_PIN
**Configuration.h**
#define TEMP_SENSOR_0 -3 		// 2020-10-16 MROE active Thermocouple MAX31855 for Extruder0

Regards
Martin

### Expected behavior

Correct Temperatur readings with MAX31855 and K-Type Thermocouple and Marlin Release 2.9.0+

### Actual behavior

I have to guess my temperatures :-(
And adapt slicing software with faulty temperatures :-(

### Steps to Reproduce

1. Connect MAX31855 with K-Type Thermocouple to Marlin
2. Attach second Thermocouple also with MAX31855 and ArduinoNano Board and Basic Adafrut Example MAX31855 as reference, if available If Thermometer as reference
3. Messure temperaures should be similar +/- 5 °C (Temperaturees >200°c)

### Version of Marlin Firmware

2.0.9

### Printer model

Creality Ender 3

### Electronics

BTT 1.4 Turbo, MAX31885, Micro Swiss Hotend, TMC2130

### Add-ons

ESP3D

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

slowbro, 2021-07-03:
This should be fixed with #22196 if you'd like to build that branch & test - I am planning on doing some extended testing today, again, and will make sure the '55 boards are reporting correctly.

qrstu, 2021-07-04:
Hello all,
I was testing again with the bugfix 2.0.x (redundant-temp-fixup) from yesterday. 02-JUL-2021
It seems that it has partly solved. I assume it has something to do with the SPI_SPEED.
I added #define SD_SPI_SPEED SPI_HALF_SPEED --> The values got realy accurate compaired +/- 5°C with external TC.

But there is still an issue in pins_BTT_SKR_common.h --> Needs to be comment out, otherwise compile error
//#if HAS_MAX_TC  //2021-07-03 MROE Fix
//#endif      //2021-07-03 MROE Fix
**'TEMP_0_CS_PIN' was not declared in this scope; did you mean 'TEMP_0_PIN'?**

**features.ini** 
TEMP_.+_IS_MAX31855                    = https://github.com/GadgetAngel/Adafruit-MAX31855-V1.0.3-Mod-M.git
**lpc167x.ini**
build_flags       = ${common.build_flags} -DU8G_HAL_LINKS -IMarlin/src/HAL/LPC1768/include -IMarlin/src/HAL/LPC1768/u8g -DLIB_MAX31855=1

Im using this parameters:
**pins_BTT_SKR_common.h**
//#if HAS_MAX_TC  //2021-07-03 MROE Fix
#define TEMP_0_CS_PIN   P0_26    // 2021-07-03 MROE MAX31855 Thermocouple PIN Hotend
#define TEMP_0_MISO_PIN P0_08    // 2021-07-03 MROE MAX31855 Thermocouple PIN Hotend
#define TEMP_0_SCK_PIN  P0_07    // 2021-07-03 MROE MAX31855 Thermocouple PIN Hotend
//#endif      //2021-07-03 MROE Fix
**Configuration_adv.h:**
#define THERMOCOUPLE_MAX_ERRORS 20 // 2020-10-24 MROE
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10  // 2020-10-24 MROE Thermocouple fix
#define SD_SPI_SPEED SPI_HALF_SPEED  //MROE 2021-07-04
#define SDCARD_CONNECTION ONBOARD // MROE 2020-05-23 BTT SKR V1.4 TURBO Use on Board SDCard
#define TEMP_SENSOR_FORCE_HW_SPI  // MROE 2021-07-04 Termocouple HotEnd MAX31855 fix
**Configuration.h:**
#define TEMP_SENSOR_0 -3

Regards
Martin

slowbro, 2021-07-04:
Thanks for the additional information. I spent the majority of last night trying to get SPI working as I expected on LPC and was starting to lean in that direction (SPI speeds), as well. I'll mess with it a bit more today/tomorrow and will hopefully be able to come to a solution.

Regarding the Adafruit library, have you tried without it on the fixup branch? For '55, all the library adds is more verbose error output. The library itself uses non-HAL/non-Marlin-y SPI; the default implementation in Marlin may give you better results.

thisiskeithb, 2021-07-07:
Fixed in #22196.

qrstu, 2021-07-11:
@slowbro 
@thisiskeithb 
he following issues have not been fixed and are not working as expected MAX31855!! Marlin-bugfix-2.0.x (07-JUL-2021)
**pins_BTT_SKR_common.h**
  #if HAS_MAX_TC //2021-07-03 MROE Fix
     #define TEMP_0_CS_PIN   P0_26    // 2021-07-03 MROE MAX31855 Thermocouple PIN Hotend
  #endif //2021-07-03 MROE Fix

Marlin\src\HAL\LPC1768\../../inc/SanityCheck.h:2060:4: error: #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
It works only when --> Comment out //#if HAS_MAX_TC //2021-07-03 MROE Fix

That compile is working with MAX31855 you have to do the follwing:
**features.ini**
LIB_MAX31855                           = https://github.com/GadgetAngel/Adafruit-MAX31855-V1.0.3-Mod-M.git
**lpc176x.ini**
build_flags       = ${common.build_flags} -DU8G_HAL_LINKS -IMarlin/src/HAL/LPC1768/include -IMarlin/src/HAL/LPC1768/u8g
  -DLIB_MAX31855=1
**pins_BTT_SKR_common.h**
#define TEMP_0_CS_PIN P0_26 // 2021-07-03 MROE MAX31855 Thermocouple PIN Hotend
//#define TEMP_0_MISO_PIN P0_08 // 2021-07-03 MROE MAX31855 Thermocouple PIN Hotend
//#define TEMP_0_SCK_PIN P0_07 // 2021-07-03 MROE MAX31855 Thermocouple PIN Hotend
--> You may need to comment out MISO and SCK PIN's !!!!!

**configuration_avd.h**
#define TEMP_SENSOR_FORCE_HW_SPI  // 2021-07-04 MROE Termocouple HotEnd MAX31855 fix
//#define MAX31865_SENSOR_WIRES_0 2
//#define MAX31865_SENSOR_WIRES_1 2
#define MAX31865_50HZ_FILTER // 2021-07-04 MROE Termocouple HotEnd MAX31855 fix

Regards
Martin

slowbro, 2021-07-11:
Ah, sorry! I'll take a look.

qrstu, 2021-07-11:
@slowbro 
The tempearature messurement is correct and working. :-)
I assume there is only is an issue with the HAS_MAX_TC-Variable --> I could not fixure out the how it will e used and assigned. I only tested compile for MAX31855 probably it's correct for MAX31865.
Regards
Martin

slowbro, 2021-07-12:
Ok, I see the error. The file that set up HAS_MAX_TC is called after the pins files are loaded, so that block never gets ran. To be honest, those defines don't need to be wrapped in HAS_MAX_TC, but either way it may be better to move the MAX TC stuff to Conditionals_adv.h; this should be fixed in bugfix soon. Thanks for letting me know!

Also, FYI - `#define MAX31865_50HZ_FILTER` only applies to MAX31865 (not '55).

thisiskeithb, 2021-07-14:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22342?

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

slowbro, 2021-07-14:
Yep, #22342 fixes this.

closed 2021-07-14 by thisiskeithb

qrstu, 2021-07-16:
@slowbro 
@thisiskeithb 
Thanks, compile is working now with HAS_MAX_TC enabled :-)
I was a little concerened that you don't change the CONFIGURATION_H_VERSION when change library versions.
HAS_MARLINUI_U8GLIB                    = U8glib-HAL@~0.5.0 --> features.ini
I had compilation errors.
And in general for bugfixing I only copy config files for same release and don't do a diff to check if library has changed :-(
Thanks
Regards
Martin

thisiskeithb, 2021-07-16:
> And in general for bugfixing I only copy config files for same release and don't do a diff to check if library has changed :-(

*Always* pull down the latest `bugfix-2.0.x` when testing, especially when asked to do so.

qrstu, 2021-08-01:
@thisiskeithb 
@slowbro 
Hello together,
When are the fixes of Conditionals_adv.h for HAS_MAX_TC available in the main 2.0.9.1 branch ?
https://github.com/MarlinFirmware/Marlin/pull/22342
Or have this fixes been missed to commit to the main branch? I can only find them in "bugfix 2.0.x"
Thanks
Regards
Martin (qrstu)