MarlinFirmware/Marlin #22294: [BUG] (bug summary)

prophetmaster, 2021-07-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

hard coded string in english 
line 297: "Stow Probe" for "deploy probe" and 
line 298: "Stow Probe" 

my personal correction is replacing **"Stow Probe"** by var `ds_str`

### Bug Timeline

i don't know

### Version of Marlin Firmware

2.0.9.1

### Printer model

custom delta

### Electronics

mks sbase

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

ellensp, 2021-07-04:
Please update (bug summary) to related issue

skruppy, 2021-07-12:
Looking into the code I found some more strings which could probably be replaced by translated strings. But I have no idea where (Serial, UI, Ext UI, Host Promt) translations should happen and where the output should be English only. But maybe I could at least help with a janky script finding those strings:

```python
#!/usr/bin/env python3

import re
import os
import sys
from pathlib import Path

strings = set()
str_pattern = re.compile(r'"(?:[^"\\]|\\.)*"')
with open('Marlin/src/lcd/language/language_en.h') as f:
	for line in f:
		for match in str_pattern.finditer(line):
			strings.add(match[0])

pathes = []
for root, dirs, files in os.walk('.', topdown=True):
	dirs[:] = [d for d in dirs if not (d.startswith('.') or d == 'language')]
	for file in files:
		path = Path(root, file)
		if path.suffix.lower() in ['.h', '.c', '.cpp']:
			pathes.append(path)

for path in sorted(pathes):
	try:
		with open(path) as f:
			for i, line in enumerate(f):
				line = line.strip().split('//', 1)[0]
				for match in str_pattern.finditer(line):
					if match[0] in strings:
						print(f'{path}:{i+1}: {line}')
	except UnicodeDecodeError as ex:
		print(f'ERROR: {path}: {ex}', file=sys.stderr)
```

ellensp, 2021-07-12:
python does not use marlin C language files

skruppy, 2021-07-12:
Sorry, I should have given more explanations:
TL;DR: Execute this script in the Marlin project root directory. You will get an output of all source file locations (like grep) containings strings also defined in the English Marlin "translation".

Long read: I was about to make a small pull request fixing this issues and started digging around in the code and found some more places with untranslated strings. I wrote this script to find some more untranslated strings. But I also found some places where I noticed I have no idea how for example the ExtUI API works, since for example the anycubic_chiron module [compares](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/lcd/extui/anycubic_chiron/chiron_tft.cpp#L202) the messages with its own [hard coded messages](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/lcd/extui/anycubic_chiron/chiron_tft_defs.h#L101). Hence I stopped creating a PR.

<details>
  <summary>Output of the script</summary>
Marlin/Configuration.h:924: * "Jerk" specifies the minimum speed change that requires acceleration.
Marlin/Configuration_adv.h:757: * Note: If the probe is not deploying, do a "Reset" and "Self-Test" and then check the
Marlin/Configuration_adv.h:757: * Note: If the probe is not deploying, do a "Reset" and "Self-Test" and then check the
Marlin/Configuration_adv.h:3076: * ; This macro send the string "Marlin" to the slave device with address 0x63 (99)
Marlin/src/HAL/AVR/math.h:81: : [longIn1] "d" (longIn1),
Marlin/src/HAL/AVR/math.h:82: [longIn2] "d" (longIn2)
Marlin/src/HAL/AVR/math.h:108: : [charIn1] "d" (charIn1),
Marlin/src/HAL/AVR/math.h:109: [intIn2] "d" (intIn2)
Marlin/src/core/language.h:191: #define STR_ERR_HOTEND_TOO_COLD             "Hotend too cold"
Marlin/src/core/language.h:245: #define STR_ERR_PROBING_FAILED              "Probing Failed"
Marlin/src/core/language.h:246: #define STR_ZPROBE_OUT_SER                  "Z Probe Past Bed"
Marlin/src/core/language.h:263: #define STR_PASSWORD_SET                    "Password is "
Marlin/src/core/language.h:345: #define AXIS4_STR "V"
Marlin/src/core/language.h:379: #define AXIS5_STR "V"
Marlin/src/core/language.h:413: #define AXIS6_STR "V"
Marlin/src/core/macros.h:91: #define A(CODE) " " CODE "\n\t"
Marlin/src/core/serial.cpp:41: PGMSTR(SP_I_STR, " " AXIS4_STR);     PGMSTR(SP_J_STR, " " AXIS5_STR);     PGMSTR(SP_K_STR, " " AXIS6_STR);
Marlin/src/core/serial.cpp:41: PGMSTR(SP_I_STR, " " AXIS4_STR);     PGMSTR(SP_J_STR, " " AXIS5_STR);     PGMSTR(SP_K_STR, " " AXIS6_STR);
Marlin/src/core/serial.cpp:41: PGMSTR(SP_I_STR, " " AXIS4_STR);     PGMSTR(SP_J_STR, " " AXIS5_STR);     PGMSTR(SP_K_STR, " " AXIS6_STR);
Marlin/src/core/serial.cpp:42: PGMSTR(SP_I_LBL, " " AXIS4_STR ":"); PGMSTR(SP_J_LBL, " " AXIS5_STR ":"); PGMSTR(SP_K_LBL, " " AXIS6_STR ":");
Marlin/src/core/serial.cpp:42: PGMSTR(SP_I_LBL, " " AXIS4_STR ":"); PGMSTR(SP_J_LBL, " " AXIS5_STR ":"); PGMSTR(SP_K_LBL, " " AXIS6_STR ":");
Marlin/src/core/serial.cpp:42: PGMSTR(SP_I_LBL, " " AXIS4_STR ":"); PGMSTR(SP_J_LBL, " " AXIS5_STR ":"); PGMSTR(SP_K_LBL, " " AXIS6_STR ":");
Marlin/src/feature/bedlevel/ubl/ubl.cpp:46: void unified_bed_leveling::echo_name() { SERIAL_ECHOPGM("Unified Bed Leveling"); }
Marlin/src/feature/bedlevel/ubl/ubl.cpp:62: SERIAL_ECHO_TERNARY(planner.leveling_active, " System v" UBL_VERSION " ", "", "in", "active\n");
Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1064: SERIAL_ECHOLNPGM("Done Editing Mesh");
Marlin/src/feature/host_actions.cpp:71: PGMSTR(CONTINUE_STR, "Continue");
Marlin/src/feature/hotend_idle.cpp:79: SERIAL_ECHOLNPGM("Hotend Idle Timeout");
Marlin/src/feature/meatpack.cpp:173: SERIAL_ECHOPGM(MeatPack_ProtocolVersion " ");
Marlin/src/feature/mmu/mmu2.cpp:274: DEBUG_ECHOLNPAIR("MMU <= F", filament, " ", cmd_arg);
Marlin/src/feature/pause.cpp:168: * - Show "Purge more" / "Continue" menu
Marlin/src/feature/pause.cpp:168: * - Show "Purge more" / "Continue" menu
Marlin/src/feature/pause.cpp:169: * - Return when "Continue" is selected
Marlin/src/feature/pause.cpp:199: TERN_(EXTENSIBLE_UI, ExtUI::onUserConfirmRequired_P(PSTR("Load Filament")));
Marlin/src/feature/powerloss.cpp:592: DEBUG_ECHOLNPAIR("zraise: ", info.zraise, " ", info.flag.raised ? "(before)" : "");
Marlin/src/feature/powerloss.cpp:624: LOOP_L_N(i, EXTRUDERS) DEBUG_ECHOLNPAIR(" ", info.filament_size[i]);
Marlin/src/feature/probe_temp_comp.cpp:76: SERIAL_ECHOPGM_P(s == TSI_BED ? PSTR("Bed") :
Marlin/src/feature/probe_temp_comp.cpp:78: s == TSI_EXT ? PSTR("Extruder") :
Marlin/src/feature/runout.cpp:122: SERIAL_ECHOPGM(" " ACTION_REASON_ON_FILAMENT_RUNOUT " ");
Marlin/src/feature/runout.cpp:122: SERIAL_ECHOPGM(" " ACTION_REASON_ON_FILAMENT_RUNOUT " ");
Marlin/src/gcode/bedlevel/G35.cpp:141: SERIAL_ECHOPAIR(" ", (screw_thread & 1) == (adjust > 0) ? "CCW" : "CW", " by ", abs(full_turns), " turns");
Marlin/src/gcode/calibrate/M852.cpp:85: SERIAL_ECHOLNPGM(STR_SKEW_MIN " " STRINGIFY(SKEW_FACTOR_MIN) " " STR_SKEW_MAX " " STRINGIFY(SKEW_FACTOR_MAX));
Marlin/src/gcode/calibrate/M852.cpp:85: SERIAL_ECHOLNPGM(STR_SKEW_MIN " " STRINGIFY(SKEW_FACTOR_MIN) " " STR_SKEW_MAX " " STRINGIFY(SKEW_FACTOR_MAX));
Marlin/src/gcode/calibrate/M852.cpp:85: SERIAL_ECHOLNPGM(STR_SKEW_MIN " " STRINGIFY(SKEW_FACTOR_MIN) " " STR_SKEW_MAX " " STRINGIFY(SKEW_FACTOR_MAX));
Marlin/src/gcode/control/M211.cpp:42: print_pos(l_soft_min, PSTR(STR_SOFT_MIN), PSTR(" "));
Marlin/src/gcode/control/M80_M81.cpp:112: LCD_MESSAGEPGM_P(PSTR(MACHINE_NAME " " STR_OFF "."));
[... cut ...]
Marlin/src/sd/usb_flashdrive/lib-uhs2/masstorage.cpp:774: Notify(PSTR(" "), 0x80);
Marlin/src/sd/usb_flashdrive/lib-uhs2/masstorage.cpp:1107: Notify(PSTR(" "), 0x80);
Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host/UHS_BULK_STORAGE/UHS_BULK_STORAGE_INLINE.h:767: Notify(PSTR(" "), 0x80);
Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host/UHS_BULK_STORAGE/UHS_BULK_STORAGE_INLINE.h:1128: Notify(PSTR(" "), 0x80);
Marlin/src/sd/usb_flashdrive/lib-uhs3/UHS_host/UHS_hexdump.h:60: E_Notify(PSTR(" "), 0x80);
</details>

thisiskeithb, 2021-07-21:
Marlin doesn't translate serial strings because it'd be a nightmare for host software to deal with.

closed 2021-07-21 by thisiskeithb