MarlinFirmware/Marlin #22295: [BUG] (bug summary)

prophetmaster, 2021-07-04:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

pause before probe not working in right direction inverted...

replace this -> if (deploy == PROBE_TRIGGERED()) break;
by -> if (deploy != PROBE_TRIGGERED()) break;

### Version of Marlin Firmware

2.0.9.1

### Printer model

custom delta

ellensp, 2021-07-04:
Please update (bug summary) to related issue

ellensp, 2021-07-05:
really?  perhaps I was not clear.
Please change the title to describe the actual bug. not literally using that string.

thisiskeithb, 2021-07-21:
Since you've provided a fix, it'd be faster if you submitted a [Pull Request](https://marlinfw.org/docs/development/getting_started_pull_requests.html) so it can be reviewed.

thinkyhead, 2021-07-30:
Hey! It took me less than 5 minutes to get this patched. If a solution is very simple, let's not badger the submitter for clerical fixes and post comments asking for a PR that might never come. Admins can just post a PR and this gets the issue closed ASAP so it doesn't linger for 9 days or more. If you require extra permissions to be able to edit issue titles please let me know.

closed 2021-07-30 by thinkyhead

thisiskeithb, 2021-07-30:
> Hey! It took me less than 5 minutes to get this patched. If a solution is very simple, let's not badger the submitter for clerical fixes and post comments asking for a PR that might never come. [...]

I didn't feel comfortable submitting a fix for something that I couldn't verify was broken. That's why I generally ask the person who had the issue & fix, to submit a PR.

thinkyhead, 2021-07-30:
I reviewed with Sublime Merge the last changes to the relevant lines and saw that indeed, the logic had been done backwards, and that was enough to confirm the fix.

commit 6efd7285 (2021-07-30): 🐛 Fix PAUSE_PROBE_DEPLOY_WHEN_TRIGGERED

Fixes #22295. Regression from #20241.
commit bcc31f68 (2021-07-30): 🐛 Fix PAUSE_PROBE_DEPLOY_WHEN_TRIGGERED

Fixes #22295. Regression from #20241.