MarlinFirmware/Marlin #22298: [FR] In dual extruder config allow Z align with 5 drivers board

MLiv79, 2021-07-05:
### Describe the feature you want

I don't know if is possible to but would be really cool if in a dual extrusion configuration, the 5th driver that normally is used for E1, could be used both for E1 and Z2 motor to align Z axis. I mean, Z alignement is a one time operation, the last slot driver could be used to align Z and then released to be used for E1 extruder when printing. 
Is it possible or hardware locked?

ellensp, 2021-07-05:
That would require some electro/mechanical method to physically uncouple the Z2 stepper from the driver
Not something that marlin can do.

closed 2021-07-05 by ellensp

MLiv79, 2021-07-05:
No because in skr 1.4t that i own, i can use the 5th driver for z2 motor or use it for e1 motor, just changing a pin def.