MarlinFirmware/Marlin #22299: [BUG] DWIN_CREALITY_LCD breaks MultiSerial

blazewicz, 2021-07-05:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Compilation with multiple serial ports while having `DWIN_CREALITY_LCD` enabled results in a cascade of errors similar to this one:

```
Marlin/src/HAL/STM32F1/../../inc/../HAL/./STM32F1/../../core/serial_hook.h:225:68: error: request for member 'write' in '((MultiSerial<BaseSerial<MarlinSerial>, int, 0>*)this)->MultiSerial<BaseSerial<MarlinSerial>, int, 0>::serial1', which is of non-class type 'int'
```

It happens because `SERIAL_CATCHALL` is set here to an integer:

https://github.com/MarlinFirmware/Marlin/blob/cd01421ac32041c7f775ec37dd8d00b29a5d335b/Marlin/src/inc/Conditionals_LCD.h#L1079

And here it is expected to be a compatible Serial class instance:

https://github.com/MarlinFirmware/Marlin/blob/cd01421ac32041c7f775ec37dd8d00b29a5d335b/Marlin/src/core/serial.h#L93

I guess a simple fix like one below would help (it does compile).

```diff
--- a/Marlin/src/core/serial.h
+++ b/Marlin/src/core/serial.h
@@ -90,7 +90,7 @@ extern uint8_t marlin_debug_flags;
   #define SERIAL_ASSERT(P)    if(multiSerial.portMask!=(P)){ debugger(); }
   // If we have a catchall, use that directly
   #ifdef SERIAL_CATCHALL
-    #define _SERIAL_LEAF_2 SERIAL_CATCHALL
+    #define _SERIAL_LEAF_2 MSERIAL(SERIAL_CATCHALL)
   #elif HAS_ETHERNET
     typedef ConditionalSerial<decltype(MYSERIAL2)> SerialLeafT2;  // We need to create an instance here
     extern SerialLeafT2 msSerial2
```

Yet I'm not sure of the function on `SERIAL_CATCHALL` and how it should be defined properly.

@thinkyhead it seems you added it in https://github.com/MarlinFirmware/Marlin/pull/17719, I can't find any other reference nor documentation.

I think it got broken in https://github.com/MarlinFirmware/Marlin/pull/21306, but I'm not sure if it worked before.

### Version of Marlin Firmware

bugfix-2.0.x

ellensp, 2021-07-05:
Please attach your configuration files.

blazewicz, 2021-07-05:
Any configuration with DWIN LCD and MultiSerial fails. It can be easily reproduced using [MarlinFirmware/Configurations/config/examples/Creality/Ender-3 V2](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20V2) with `SERIAL_PORT_2` enabled:

```diff
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -123,7 +125,7 @@
  * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
  * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-//#define SERIAL_PORT_2 -1
+#define SERIAL_PORT_2 2
```

ellensp, 2021-07-05:
Yes I can reproduce the original error.. But your diff to fix it,  I cant replicate that at all.

blazewicz, 2021-07-05:
With `STM32F103RET6_creality` env it fails because it can't find `MSerial0` symbol, probably because Serial port 0 is not defined for this platform and `SERIAL_CATCHAL` is declared to be `0`.

```
Marlin/src/HAL/STM32/../../inc/../HAL/./STM32/HAL.h:50:21: error: 'MSerial0' was not declared in this scope; did you mean 'MSerial10'?
```

It would compile if you changed:

```diff
--- a/Marlin/src/inc/Conditionals_LCD.h
+++ b/Marlin/src/inc/Conditionals_LCD.h
@@ -1076,7 +1076,7 @@
 #endif

 #if ENABLED(DWIN_CREALITY_LCD)
-  #define SERIAL_CATCHALL 0
+  #define SERIAL_CATCHALL 2
   #ifndef LCD_SERIAL_PORT
     #define LCD_SERIAL_PORT 3 // Creality 4.x board
   #endif
```

It is defined for `STM32F103RET6_creality_maple`'s platform, but I don't know which port it represents.

I would suggest a fix, but I don't know what `SERIAL_CATCHALL` is.

ellensp, 2021-07-24:
Bump...  with people using the DWIN_CREALITY_LCD or more and more hardware, this is becoming more of an issue

Gaiadan, 2021-07-24:
Bump

env STM32F103RC_btt_maple

"request for member 'write' in '((MultiSerial<ForwardSerial<USBSerial>, int, 0>*)this)->MultiSerial<ForwardSerial<USBSerial>, int, 0>::serial1', which is of non-class type 'int'",
	"startLineNumber": 225,
	"startColumn": 68,
	"endLineNumber": 225,
	"endColumn": 68
}

if i disable serial port 2 it works but i want the second serial for my raspberry pi

thinkyhead, 2021-07-30:
Let me try to follow the bouncing ball and try to make sense of this.

This concept comes from #21336 by @X-Ryl669 and the value of `SERIAL_CATCHALL` gets assigned to `_SERIAL_LEAF_2` which would otherwise be `msSerial2` or `MYSERIAL2`. The value of `_SERIAL_LEAF_2` is used in one of two ways:
```cpp
// Hook Meatpack if it's enabled on the second leaf
#if ENABLED(MEATPACK_ON_SERIAL_PORT_2)
  typedef MeatpackSerial<decltype(_SERIAL_LEAF_2)> SerialLeafT2;
  extern SerialLeafT2 mpSerial2;
  #define SERIAL_LEAF_2 mpSerial2
#else
  #define SERIAL_LEAF_2 _SERIAL_LEAF_2
#endif
```
Later, `SERIAL_LEAF_2` is used to define the `MultiSerial` instance.

Should `SERIAL_CATCHALL` actually be an integer, because it doesn't fit the pattern of other things, or should it be something like `MYSERIAL0` or `msSerial0` ?

Or, is there some other condition, and the `SERIAL_CATCHALL` value should be based on the value of `SERIAL_PORT`  or `SERIAL_PORT_2`, or … ?

It's a lot to review and try to make sense of, but the fix is probably something simple.

X-Ryl669, 2021-07-30:
What is the expected port to use in those printer ? 
If I understand the intend correctly from #17719, `SERIAL_CATCHALL` should be a serial port that's overwriting all serial configuration when defined.  

I think it can be removed and the proper serial port declared and used instead.
So maybe you can just undefine it from the Conditional file.

@thinkyhead  Can you explain what it was used for initially ?

GhostlyCrowd, 2021-09-09:
Bumping this, as is still an issue.

GhostlyCrowd, 2021-09-09:
> SERIAL_CATCHALL

`#if EITHER(HAS_DWIN_E3V2, IS_DWIN_MARLINUI)
  //#define SERIAL_CATCHALL 0
  #ifndef LCD_SERIAL_PORT
    #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_E3_TURBO)
      #define LCD_SERIAL_PORT 1
    #else
      #define LCD_SERIAL_PORT 3 // Creality 4.x board
    #endif
  #endif
#endif`

As you suggested, fixed to compile issue for me, cannot test until i get everything assembled.

SISLANGER, 2021-10-09:
Is there a fix for this yet ? I cant compile without multiSerial errors when enabling a second Serial Port

SISLANGER, 2021-10-10:
Just a quick update it seems like PlatformIO wasn't clearing / updating when re-compiling even when using the clean all function. However after removing and re-installing PlatformIO disabling the SERIAL_CATCHALL 0 has done the trick. Main build 02000902)

DaanvanBoven, 2022-04-16:
I seem to be getting this error after setting the following: `#define SERIAL_PORT_2 3`
This was when using the example/creality/ender3v2/crealityui configs.
Before setting `#define SERIAL_PORT_2 3`
I set default_envs = STM32F103RET6_creality
and #define SERIAL_PORT 1

Xlaits, 2022-07-10:
> Solved: [bigtreetech/BIGTREETECH-SKR-mini-E3#601 (comment)](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/601#issuecomment-1029852372)

How does this even remotely solve this? That uses fairly specific options for a specific board and stepper driver setup.
Where I am using the TMC2208 drivers, those instructions say to use otherwise. I am also using the BTT SKR 1.4 (Non Turbo), so some of those options also are incorrect for my board. 

Disabling the second serial port, and setting the first one to -1 seems to allow me to at least compile the firmware, but it refuses to load on my machine.

rarens, 2022-10-12:
Issue still exists

samiresa, 2023-03-27:
Issue still exists as of 3/26/2023 with bugfix build.

yuuahmad, 2023-08-14:
Marlin\src\HAL\shared\../../core/serial.h:87:28: error: 'MYSERIAL2' was not declared in this scope

VectorHasting, 2023-12-31:
I'm still having the error compiling using bugfix 2.1.x as of 12/9/2023.

I'm using #define MOTHERBOARD BOARD_BTT_SKR_V2_0_REV_B 
(which is the setting required by BigTree_SKR_2_F429)
I've tried compiling with BIGTREE_SKR2_F429 and BIGTREE_SKR2_F429_USB

My error is the same as @yuuahmad: 

`Marlin\src\HAL\STM32\../../inc/../core/serial.h:87:28: error: 'MYSERIAL2' was not declared in this scope; did you mean 'MYSERIAL1'?`

Neither the changes suggested by @denis1482 nor @Xlaits fixes it for me.

My fix was to fall back to 2.1.2.1.

nicwilson58, 2024-02-10:
I just upgraded to BugFix build 2.1.22 from 9th February 2024 and this issue still exists.  compile SKR Mini E3 V3.0 with Pro_UI or MarlinUI and this same error occurs.

ellensp, 2024-02-10:
This issue is specifically about having #define SERIAL_PORT and #define SERIAL_PORT_2 with any of the UI for Ender-3 v2 OEM display.  

If you do not have two serial ports enabled in marlin, you have a different issue 
ie your #define SERIAL_PORT or LCD_SERIAL_PORT is wrong for your motherboard

BIG-BRO-L, 2024-05-13:
Issue still exists as of 5/12/2024

(Please help us)

classicrocker883, 2024-11-13:
IIRC, try this:
in **ini/stm32f4.ini**
change to `toolchain-gccarmnoneeabi@1.120301.0`

ryan-makes-it, 2025-03-07:
Hi folks - I wanted to chime in that I'm still experiencing this issue as well - I added a Big Tree Tech SKR Mini E3 v3 controller board to my Ender 3 v2, which in my case has the TJC version of the DWIN LCD (see [this page about the stock Ender 3 v3 displays (link)](https://github.com/mriscoc/Ender3V2S1/tree/Ender3V2S1-Released/display%20assets)).  My goal was to connect a Raspberry Pi Zero 2 W via UART2 (the TFT port) to the SKR Mini E3 v3 for OctoPrint use.

Using [Miguel Risco-Castillo's excellent Ender 3 optimized firmware (link)](https://github.com/mriscoc/Ender3V2S1/tree/Ender3V2S1-Released) and a rewired display cable, I can get the TJC / DWIN display working just fine.  However, when I uncomment "#define SERIAL_PORT_2" and select any serial port (I've tried -1, 0, 1, 2 and 3), I get the same "Marlin/src/HAL/STM32/../../inc/../core/serial.h:72:26: error: 'MYSERIAL' was not declared in this scope; did you mean 'MYSERIAL2'?" error reported in [#23889 (link)](https://github.com/MarlinFirmware/Marlin/issues/23889).

I'm currently running Marlin Bugfix-2.1.x from 12/17/2024 (the current version of Miguel Risco-Castillo's optimized firmware), but I've experienced the error with the nightly Bugfix-2.1.x as well.  I went back to Miguel's version to eliminate any potential misconfigurations by me.  

Thanks for any help you can provide!

classicrocker883, 2025-03-08:
hey @ryan-makes-it , could you provide your configuration files?  
(select **Configuration.h**/**Configuration_adv.h** together and archive them in a `.zip`)  and also what `default_envs` youre using in **platformio.ini**. 

You can try my fork of [mriscoc's firmware](https://github.com/classicrocker883/MRiscoCProUI) because I may have solved that issue there.

ryan-makes-it, 2025-03-10:
Hello @classicrocker883 , thanks for the suggestions!

I actually ended up commenting out the #define SERIAL_CATCHALL in my testing, and then found I had to disable the M575 functionality too.  That's good to know that I can use the stock Marlin M575 file, I'll try that!  

In any case, that got me a working printer with the TFT port connecting to OctoPrint, so that's something.  

Also, my board has a STM32G0B1 RET6 chip, which apparently has 512kb of flash storage.  Marlin Auto Build is using the STM32G0 architecture, and STM32G0B1RE_btt as the environment.  So, that's default_envs = STM32G0B1RE_btt for reference.  

Do you still want the configuration files?  I suspect they'll look just like yours!

ryan-makes-it, 2025-03-10:
Yeah, I was looking at where SERIAL_CATCHALL was defined and wondering if some logic to not define it in case of multiple serial ports plus the Ender 3 v2 DWIN / DWIN variants screen might make sense too.

Do you happen to know why the SERIAL_CATCHALL exists?  

The only thing I've noticed not working properly with it commented out is that at the end of the print, the screen goes all yellow, and only items that are updated on the screen (temperature, position, things like that) are displayed.  This can be fixed by clicking the encoder button and bringing up another menu or forcing the screen to refresh in other ways.  Not a dealbreaker, just odd behavior by my TJC variant DWIN screen.

classicrocker883, 2025-03-11:
I suppose its purpose had been more useful in the past. it is because manufacturers use Marlin, so the code has to appeal to their needs. I don't get why the only issue is what you said about the screen after a print. it's possible that only happens on a TJC display.

closed 2026-10-13 by thinkyhead