MarlinFirmware/Marlin #22301: [BUG] Auto bed leveling bilinear not applied when printing Marlin 2.0.9.1

swatkxs, 2021-07-06:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

[](url)I did bed leveling following thess steps: 
G90 ; use absolute coordinates
M83 ; extruder relative mode
M107 ; turn off fan
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28
G29
M500
M503
G1 X0 Y-3.0 F1000.0 ; go outside print area
G92 E0.0
G1 X60.0 Z0.6 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line
G92 E0.0
M221 S{if layer_height<0.075}100{else}95{endif}
[config.zip](https://github.com/MarlinFirmware/Marlin/files/6770578/config.zip)

G29 
T:220.00 /220.00 B:92.36 /93.00 @:69 B@:126
echo:busy: processing
Bilinear Leveling Grid:
      0      1      2      3      4      5      6
 0 +0.096 +0.113 +0.064 -0.038 -0.157 -0.339 -0.513
 1 +0.026 +0.027 -0.006 -0.069 -0.146 -0.271 -0.426
 2 -0.001 +0.005 -0.014 -0.081 -0.146 -0.280 -0.413
 3 -0.106 -0.041 -0.016 -0.058 -0.184 -0.290 -0.393
 4 -0.014 +0.035 +0.032 -0.082 -0.159 -0.269 -0.378
 5 +0.011 +0.061 +0.003 -0.131 -0.196 -0.291 -0.418
 6 -0.088 -0.038 -0.100 -0.234 -0.364 -0.510 -0.591

 T:220.25 /220.00 B:92.26 /93.00 @:27 B@:127
echo:busy: processing
X:0.00 Y:-2.20 Z:5.00 E:503.50 Count X:5494 Y:4988 Z:2004
ok
 T:220.00 /220.00 B:92.20 /93.00 @:75 B@:127

M500
echo:Settings Stored (809 bytes; crc 25573)
ok

M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius

echo:; Filament settings: Disabled
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo: M92 X100.00 Y100.00 Z400.00 E280.00
echo:; Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z12.00 E120.00
echo:; Maximum Acceleration (units/s2):
echo:  M201 X1000.00 Y1000.00 Z1000.00 E5000.00
echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.00 R1250.00 T1000.00
echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E1.50
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S1 Z10.00
echo:  G29 W I0 J0 Z0.09600
echo:  G29 W I1 J0 Z0.11350
echo:  G29 W I2 J0 Z0.06350
echo:  G29 W I3 J0 Z-0.03800
echo:  G29 W I4 J0 Z-0.15650
echo:  G29 W I5 J0 Z-0.33900
echo:  G29 W I6 J0 Z-0.51300
echo:  G29 W I0 J1 Z0.02600
echo:  G29 W I1 J1 Z0.02700
echo:  G29 W I2 J1 Z-0.00650
echo:  G29 W I3 J1 Z-0.06900
echo:  G29 W I4 J1 Z-0.14650
echo:  G29 W I5 J1 Z-0.27150
echo:  G29 W I6 J1 Z-0.42650
echo:  G29 W I0 J2 Z-0.00150
echo:  G29 W I1 J2 Z0.00450
echo:  G29 W I2 J2 Z-0.01400
echo:  G29 W I3 J2 Z-0.08050
echo:  G29 W I4 J2 Z-0.14550
echo:  G29 W I5 J2 Z-0.28050
echo:  G29 W I6 J2 Z-0.41300
echo:  G29 W I0 J3 Z-0.10650
echo:  G29 W I1 J3 Z-0.04150
echo:  G29 W I2 J3 Z-0.01550
echo:  G29 W I3 J3 Z-0.05800
echo:  G29 W I4 J3 Z-0.18400
echo:  G29 W I5 J3 Z-0.29050
echo:  G29 W I6 J3 Z-0.39300
echo:  G29 W I0 J4 Z-0.01400
echo:  G29 W I1 J4 Z0.03450
echo:  G29 W I2 J4 Z0.03200
echo:  G29 W I3 J4 Z-0.08150
echo:  G29 W I4 J4 Z-0.15900
echo:  G29 W I5 J4 Z-0.26900
echo:  G29 W I6 J4 Z-0.37800
echo:  G29 W I0 J5 Z0.01100
echo:  G29 W I1 J5 Z0.06100
echo:  G29 W I2 J5 Z0.00350
echo:  G29 W I3 J5 Z-0.13150
echo:  G29 W I4 J5 Z-0.19650
echo:  G29 W I5 J5 Z-0.29150
echo:  G29 W I6 J5 Z-0.41800
echo:  G29 W I0 J6 Z-0.08800
echo:  G29 W I1 J6 Z-0.03800
echo:  G29 W I2 J6 Z-0.10050
echo:  G29 W I3 J6 Z-0.23400
echo:  G29 W I4 J6 Z-0.36400
echo:  G29 W I5 J6 Z-0.51050
echo:  G29 W I6 J6 Z-0.59150
echo:; Material heatup parameters:
echo:  M145 S0 H185.00 B65.00 F0
echo:  M145 S1 H230.00 B93.00 F0
echo:; PID settings:
echo:  M301 P13.39 I0.69 D64.61
echo:  M304 P190.00 I15.00 D928.31
echo:; Z-Probe Offset (mm):
echo:  M851 X23.00 Y5.00 Z-0.94
echo:; Stepper driver current:
echo:  M906 X650 Y650 Z450
echo:  M906 I1 Z450
echo:  M906 T0 E480

echo:; StallGuard threshold:
echo:  M914 X3 Y6
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 I1 Z
echo:  M569 S1 T0 E
echo:; Filament load/unload lengths:
echo:  M603 L0.00 U100.00
echo:; Filament runout sensor:
echo:  M412 S0
ok
 T:220.00 /220.00 B:93.51 /93.00 @:25 B@:56
[image]
[image]
[image]
[image]

### Version of Marlin Firmware

2.0.9.1

### Printer model

Prusa Bear with SKR 1.3

### Electronics

SKR 1.3

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

XDA-Bam, 2021-07-07:
Hi. Please attach logs as TXT in a ZIP or link to a Pastebin of the log. Otherwise posts get very crowded.

Also, whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:
- Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send M502 and M500 to ensure your Configurations are applied.
- Issue the command M111 S247 to enable maximum logging.
- Perform a G28 to do your standard homing procedure.
- Do a G29 to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a .TXT file and attach it to your next reply.

Independent of these additional steps, this looks like it could be related to #22175 and #21887.

salfter, 2021-07-11:
It's not just bilinear leveling, either.  I've noticed the same with unified bed leveling: https://github.com/MarlinFirmware/Marlin/issues/22175

DerAndere1, 2021-07-20:
This should be fixed in bugfix-2.0.x with commit https://github.com/MarlinFirmware/Marlin/commit/154decfc66c30808cfe7320e542fcf90427d7176 . Please close this issue if current bugfix-2.0.x works for you.

swatkxs, 2021-07-22:
need have a check,.....

thinkyhead, 2021-07-30:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

swatkxs, 2021-07-30:
Hi, thinkyhed，I found a new bug that might be related to this, the situation was: Homing first and waiting for heat up, then G29, the printer would be halted after some leveling points,  while, if I put Z a little bit higher, G29 will pass with no problem.

thinkyhead, 2021-07-30:
If you get halting try adjusting the value of `Z_PROBE_LOW_POINT`. The default is -5 so try -10.

thinkyhead, 2021-07-30:
@DerAndere1 — Your links to commits and other issues lately are seeming sketchy and it is not always clear how they are relevant. What are you up to?

swatkxs, 2021-09-03:
All confirmed, this bug has been fixed, thank you guys.

closed 2021-09-03 by thisiskeithb

Fordsierra88, 2021-09-22:
Sorry if I open the topic again but I have the same problem.  Therefore, I just write here underneath. 

From Marlin 2.0.5.2 it had started with me with the problem.  I switched to Bugfix every now and then.  Depending on the branch it worked without big problems.   A month ago I tried again with 2.0.9.1 and unfortunately nothing has changed.

Because I read here that the latest branch will fix the problem, I flashed Bugfix (the latest version).  It really worked in the beginning and a few prints it did not work anymore.  Mechanically it can not be, I checked everything, it runs everything clean. 

Someone maybe a solution ?  

My printer : 

Biqu B1
SKR 1.4 
TMC2209
BL Touch 3.1
Marlin Bugfix (21.09.2021)

many thanks 

Translated with www.DeepL.com/Translator (free version)