MarlinFirmware/Marlin #22302: [BUG] (bug summary)

Hanki06, 2021-07-06:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Huhu, gleich eins vorweg irgentwie bekomme ich es mit dem bugfix-2.0.x-Code nicht hin, sorry.
Hab meinen Ender 5 Plus mit BTT SKR V1.4 Turbo 32Bit Karte + TFT35 V3.0 sowie TMC 2209 und einem Direktextruder von Micro Swiss ausgerüsstet. Sobald ich den Homebutton drücke fährt der Kopf zu den Endschaltern u. anschliessend wieder zur mitte, das Heizbett fährt etwas hoch bleibt stehen und der BL-Touch fährt aus. OctoPrint gibt diese Fehler aus:

!! STOP called because of BLTouch error - restart with M999

Die Marlin Version ist 2.0.7.2, mit Version 2.0.9.1 gibt es keine Verbing zum Drucker.

Mit freundlichen Grüßen

 Christian Hanka

### Version of Marlin Firmware

2.0.7.2

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

Im Terminal:
Send: G91
Recv: ok
Send: G1 Z10 F1200
Recv: ok
Send: G90
Recv: ok
Send: G28

Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly

Recv: Error:!! STOP called because of BLTouch error - restart with M999
Changing monitoring state from "Operational" to "Error"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

ellensp, 2021-07-07:
Please attach your configuration files.

Hanki06, 2021-07-07:
Oh. Das hab ich wohl im eifer des gefechts vergessen.

 [Marlin-2.0.7.2-BTT-SKR-V1.4-Turbo-Ender5Plus-DualZ.zip](https://github.com/MarlinFirmware/Marlin/files/6774987/Marlin-2.0.7.2-BTT-SKR-V1.4-Turbo-Ender5Plus-DualZ.zip)

Hanki06, 2021-07-07:
Oh, das hab ich wohl im eifer des gefechts vergessen. 

[Marlin-2.0.7.2-BTT-SKR-V1.4-Turbo-Ender5Plus.zip](https://github.com/MarlinFirmware/Marlin/files/6775003/Marlin-2.0.7.2-BTT-SKR-V1.4-Turbo-Ender5Plus.zip)

ellensp, 2021-07-07:
Where did you plug in black and white bltouch cable? two options
1) in Z_MIN_ENDSTOP plug
or 
2) in probe plug.

Hanki06, 2021-07-07:
So habe ich es angeschlossen.

[image]

ellensp, 2021-07-07:
this is  wrong. #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
make it //#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

You are not using z-min pins, now it  will look at probe port pins

also
change //#define USE_PROBE_FOR_Z_HOMING
to #define USE_PROBE_FOR_Z_HOMING

ellensp, 2021-07-07:
Not a but, config issue.

closed 2021-07-07 by ellensp

Hanki06, 2021-07-07:
Hat leider nicht funktioniert.

ellensp, 2021-07-07:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).