MarlinFirmware/Marlin #22306: [BUG] (bug summary)enquecommand

espadan3d, 2021-07-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

HI to all
I used until of marline 1.0 but I decided use of marlin 1.1 or 2.0
but have a problem 
now i have a mechanical pause bouton is very important for me this active by some line order in marlin_main.cpp and other page this order is correct in marline version 1.0 and verify but when copy this line in similar place on marlin1.1 couldn't verify marlin I receive an error message
Should the shape of the commands change?(attached pic) 
I do not know programming🤦🏻‍♂️please help me can import in marlin1.1 or marlin2.0
[image]
 

### Version of Marlin Firmware

1.0

### Printer model

corexy

### Electronics

ramp1.4

### Your Slicer

Cura

### Host Software

SD Card (headless)

espadan3d, 2021-07-07:
I need a mechanical bouton when push it pause print and lock all axis until unpush (may be for 1 hours or more )pause in menu not working for me
if you have any idea please tell me

ellensp, 2021-07-07:
No woks is done on marlin v1 anymore.

in v2 you would use something like queue.enqueue_one_P(PSTR("M1001"))

But in v2 there is also button support  #define CUSTOM_USER_BUTTONS In Configuration_adv.h

closed 2021-07-07 by ellensp

espadan3d, 2021-07-08:
Thanks a lot
I found CUSTOM_USER_BUTTONS it good
But unfortunately I do not know programming
And I could not find place to specify the PIN number
Can you help me for write this order?
I want it when pin 42 is connected to GND
Execute commands simultaneously M109/M25/M24
and when disconnected run M109 and M24

ellensp, 2021-07-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).