MarlinFirmware/Marlin #22308: HELP! I want to connect Reprap discount smart controller to stock motherboard of ender 3 pro

n0moreCRY, 2021-07-07:
### Describe the feature you want

The LCD that I want to connect looks like that one: https://reprap.org/wiki/RepRapDiscount_Smart_Controller
I followed this tutorial and soldered cables like that: https://ender3.club/%D0%A3%D1%81%D1%82%D0%B0%D0%BD%D0%BE%D0%B2%D0%BA%D0%B0_%D1%8D%D0%BA%D1%80%D0%B0%D0%BD%D0%B0_Reprap_Full_Graphics_12864_%D0%BD%D0%B0_%D1%81%D1%82%D0%BE%D0%BA%D0%BE%D0%B2%D1%83%D1%8E_%D0%BF%D0%BB%D0%B0%D1%82%D1%83/SKR_Mini_E3

When I tried to turn it on, it didn't help. I noticed that the pin 10 on Reprap discount smart controller is VCC, while on RepRapDiscount Full Graphic Smart Controller VCC is 1. So, in this case the schematic doesn't work for my LCD model. Could you help me please with a correct one? (sd card and other features it is not so important, more important is that i could see image and could start printing).
[image]

Firmware I already uploaded (melzi_optiboot_optimized) on printer, and in marlin firmware I changed LCD model, so in firmware there shouldn't be issues, only connection.

### Additional context

Printer Ender 3 Pro
Stock board, melzi Creality V1.1.4

ellensp, 2021-07-07:
You cannot fully do this.

This requires 6 pins for LCD, 3 pins for encoder, 1 pin for buzzer, 5 for sdcard. 
Ie you need 15 IO pins.  
Your Controller only has 7 IO pins on EXP1 and a 6 pin SPI port (which has vcc, gnd and reset,  so only 3 data pins)
That is 10 IO pins that are easily available.

If you drop the SD and buzzer, you should be able to get just the LCD and encoder working.

closed 2021-07-07 by thisiskeithb

n0moreCRY, 2021-07-07:
> You cannot fully do this.
> 
> This requires 6 pins for LCD, 3 pins for encoder, 1 pin for buzzer, 5 for sdcard, 1 for reset (special rest line)
> Ie you need 15 IO pins + reset
> Your Controller only has 7 IO pins on EXP1 and a reset line. You also have a 6 pin SPI port (which has vcc, gnd and reset, so only 3 data pins)
> That is 10 IO pins + reset that are easily available.
> 
> If you drop the SD you should be able to get the LCD, encoder, reset and buzzer working.

Thanks for the reply. I don't need SD card let's say, only 6pins for LCD and 3 pins for encoder. 9 Pins will be enought?