MarlinFirmware/Marlin #22310: [BUG] (bug summary)

Bitenc, 2021-07-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, I want to install the Marlin Version 2.0.x on my old delta 3D printer (Makeblock mGiraffe), mainly for the Bilinear levelling feature. 
Previously it ran a forked version of the 1.1.0 Marlin Firmware (https://github.com/Makeblock-official/Marlin-for-mGiraffe)
The board is just an ATmega2560 board (called MegaPi from Makeblock), running supposedly DRV8825 motor drivers.
I have configured the printer with the correct I/O Pins, since the old custom pin file could nearly exactly be copied over.

The thing is now, I got nearly every functionality running: Fan, Buzzer, SD Card, Screen, Encoder etc...
Yet, I cannot get the motor drivers working. I attached my configuration files below.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6778768/Configuration.zip)

### Expected behavior

When calling e.g. the M17 command, I expect the motors to power on and lock up.

### Actual behavior

Seemingly nothing happens. The motors do not lock up and are still moveable. The drivers are supplied via 12 V DC and the enable pin switches from 5V to 0V, when the M17 command is called. Despite that, nothing happens on the motors.

Is there probably a timing issue on the step/ direction signal?
The motor drivers are not bad, since when I install the old V1.x.x firmware, they work!

### Steps to Reproduce

1. Get a Makeblock mGiraffe printer...
2. just kidding. I am hoping that someone can see obvious errors in my Config files or has another Idea where the issue might be.

### Version of Marlin Firmware

Marlin 2.0.9.1

### Printer model

Makeblock mGiraffe

### Electronics

DRV8825

Bitenc, 2021-07-07:
Okay, got the issue. My board has probably no pullups on the MS1, MS2, MS3, RESET and SLEEP lines. Therefore Makeblock added the following codeblock (for all axes) into their stepper.cpp file to mitigate this issue: 

```
  SET_OUTPUT(X_MS1_PIN);
  SET_OUTPUT(X_MS2_PIN);
  SET_OUTPUT(X_MS3_PIN);
  SET_OUTPUT(X_RESET_PIN);
  SET_OUTPUT(X_SLEEP_PIN);
  WRITE(X_MS1_PIN,1);
  WRITE(X_MS2_PIN,1);
  WRITE(X_MS3_PIN,1);
  WRITE(X_RESET_PIN,1);
  WRITE(X_SLEEP_PIN,1);
```
This initializes the signals with a logical 1. Could this be a potential feature request?

ellensp, 2021-07-08:
Two things here

Marlin alreadyhas software settable MS pins

You just need to define X_MS1_PIN, X_MS2_PIN, X_MS3_PIN in your boards pins file.

Then set the microstepping in Configuration_adv.h

```
// Microstep settings (Requires a board with pins named X_MS1, X_MS2, etc.)
#define MICROSTEP_MODES { 16, 16, 16, 16, 16, 16 } // [1,2,4,8,16]

```

Secondly, for the other pins Marlin has BOARD_PREINIT which you also just add to your boards pins file
This sets additional pin requirements

eg add something like

#define BOARD_PREINIT() { \
  OUT_WRITE(X_RESET_PIN,1); \
  OUT_WRITE(X_SLEEP_PIN, 1); \
  OUT_WRITE(Y_RESET_PIN,1); \
  OUT_WRITE(Y_SLEEP_PIN, 1); \
  OUT_WRITE(Z_RESET_PIN,1); \
  OUT_WRITE(Z_SLEEP_PIN, 1); \
}

closed 2021-07-08 by ellensp

Bitenc, 2021-07-08:
That was it. Thanks a lot!!!