MarlinFirmware/Marlin #22311: I want to connect my 3d printer to the database with serial communication mcu esp, I have a question about this, can you help me?

paxsipornax, 2021-07-07:
### Is your feature request related to a problem? Please describe.

hello, I have ender 3 v2 3d printer and I want to communicate with my printer via serial port, I am thinking of connecting an esp32, but can the parameters sent to the lcd via serial communication be sent to esp32?
Can you help me find which is the communication library in vsc?

### Are you looking for hardware support?

serial port

### Describe the feature you want

ıot

thisiskeithb, 2021-07-07:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-07-07 by thisiskeithb

paxsipornax, 2021-07-07:
> Bu Sorun Sırası, Marlin hata raporları ve geliştirmeyle ilgili sorunlar içindir ve burada kullanıcı desteği sorularını ele almamayı tercih ediyoruz. ( [Bu sayfada](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question) belirtildiği gibi .) Yapılandırma ve sorun giderme konusunda yardım almak için en iyi sonuçları almak için lütfen aşağıdaki kaynakları kullanın:
> 
> * [Marlin Belgeleri](https://marlinfw.org) , MarlinFW.org'da barındırılıyor
> * [Marlin Forum](https://forums.reprap.org/list.php?415) , RepRap.org'da barındırılıyor
> * [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Grubu
> * [3D Yazıcılar için Marlin Firmware](https://www.facebook.com/groups/3Dtechtalk/) Facebook Grubu
> * [](https://www.youtube.com/results?search_query=marlin+configuration)YouTube'da [Marlin Yapılandırması](https://www.youtube.com/results?search_query=marlin+configuration)
> * [Marlin Firmware](https://discord.gg/n5NJ59y) Discord sunucusu
> 
> Topluluktan yardım istedikten sonra, fikir birliği Marlin'de bir hataya işaret ediyorsa, o zaman bir [hata raporu](https://github.com/MarlinFirmware/Marlin/issues/new/choose) göndermelisiniz .

> Bu Sırası, Marlin hata sorunları ve geliştirmeyle ilgili tüm sorunlarını almamayı tercih ediyoruz. ( [Bu toplum](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question) gibi .) Yapılandırma ve sorunlu yardım almak için en iyi iyileşme için lütfen kullanılır kullanılır:
> 
> * [Marlin Belgeleri](https://marlinfw.org) , MarlinFW.org'da barındırılıyor
> * [Marlin Forum](https://forums.reprap.org/list.php?415) , RepRap.org'da barındırılıyor
> * [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Grubu
> * [3D Yazıcılar için Marlin Firmware](https://www.facebook.com/groups/3Dtechtalk/) Facebook Grubu
> * [](https://www.youtube.com/results?search_query=marlin+configuration)YouTube'da [Marlin Yapılandırması](https://www.youtube.com/results?search_query=marlin+configuration)
> * [Marlin Firmware](https://discord.gg/n5NJ59y) Uyuşmazlığı
> 
> Topluluktan yardım istedikten Sonra, fikir birliği Marlin'de Bir hataya İşaret ediyorsa, o zaman Bir [hata raporu](https://github.com/MarlinFirmware/Marlin/issues/new/choose) göndermelisiniz .

thanks for informing

paxsipornax, 2021-07-07:
> Bu Sırası, Marlin hata sorunları ve geliştirmeyle ilgili tüm sorunlarını almamayı tercih ediyoruz. ( [Bu [...]

thanks for informing

paxsipornax, 2021-07-07:
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this [...]

thanks for informing