MarlinFirmware/Marlin #22312: [BUG] E driver error detected: 0x1075052576 coil short circuit

rick-ryerson, 2021-07-07:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

After installing bugfix-2.0.x on SKR 1.4 Turbo and running a calibration cube, the printer halts and says to "reset" with the following message in octopi terminal.

Recv: E driver error detected: 0x1075052576.0000000000000000
Recv: coil short circuit

Attached are all the appropriate files 
[octoprint-output.txt](https://github.com/MarlinFirmware/Marlin/files/6780518/octoprint-output.txt)
[platformio.ini.txt](https://github.com/MarlinFirmware/Marlin/files/6780531/platformio.ini.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/6780532/Configuration.h.txt)
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/6780536/Configuration_adv.h.txt)

### Bug Timeline

Errors in bugfix-2.0.x; Works in 2.0.4

### Expected behavior

I expected the printer to print the calibration cube.

### Actual behavior

The printer halted and said to reset.

### Steps to Reproduce

Compile bugfix-2.0.x with the attached files
Deploy to SKR 1.4 Turbo
Verify BLTouch is working 
Autohome

Use OctoPi to load and print a calibration cube
Watch autobedleveling work
watch printer print rim of cube 

watch printer fail after attempting to print inside cube rim.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Ender 3 Pro

### Electronics

SKR 1.4 Turbo, TMC2208 drivers

### Add-ons

Creality BLTouch 3.x

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[files.zip](https://github.com/MarlinFirmware/Marlin/files/6780545/files.zip)

I tried 2.0.9, 2.0.8 and 2.0.7 plus the version that BTT hosts on their github. 2.0.6.x All these have the same results. 

Recv: E driver error detected: 0x1075052576.0000000000000000
Recv: coil short circuit

A quick search of reddit in /r/BIGTREETECH will show you other instances of people having the same issue and reverting to 2.0.4 for a working printer.

I tried to report this as a bug last week and you closed it. Don't ignore this again. It's a repeatable bug, and is correctable by going back to 2.0.4

GMagician, 2021-07-08:
This may be disabled by commenting, in configuration_adv.h, below line:
#define STOP_ON_ERROR
but I think its an error reporter by TMC driver (for 10 times consecutively)

rick-ryerson, 2021-07-08:
> This may be disabled by commenting, in configuration_adv.h, below line:
> #define STOP_ON_ERROR
> but I think its an error reporter by TMC driver (for 10 times consecutively)

Doing that just ignores the issue. The stepper stops advancing or retracting.

Fusseldieb, 2021-07-09:
I'm using an SKR 1.4 Turbo with TMC drivers and none of them ever presented this issue on Marlin 2.x

> Doing that just makes it so the printer doesn't halt. The stepper stops advancing or retracting.

Then you definitively have a hardware problem. Check your cables with a multimeter, maybe they short together when doing a specific curve. Your E stepper could also have gone bad - like it said, a coil short.

Holm-tech, 2021-07-13:
I have exactly the same issue.

Well fault code is a bit different but... 
Recv: E driver error detected: 0x1075380256
Recv: coil short circuit

**Version of Marlin Firmware**
bugfix-2.0.9

**Printer model**
FolgerTech FT-5

**Electronics**
SKR 1.4 Turbo, TMC2208 drivers

**Add-ons**
BLTouch

**Your Slicer**
Cura

**Host Software**
OctoPrint

[image]

DerAndere1, 2021-07-22:
The root cause is probably the bug described in https://github.com/MarlinFirmware/Marlin/issues/15797

rick-ryerson, 2021-07-24:
That'd be cool. Either way, it'd be nice to be able to upgrade.

thinkyhead, 2021-07-30:
Are you able to narrow down the version of Marlin or the date range where the bug starts to appear? That can help us to narrow down which change(s) led to this bad behavior.

sjirafje, 2021-08-05:
Same issue here. Printer halts in the first layer. 
>     E driver error detected: 0x1074921488.0000000000000000
>     coil short circuit
>     echo:Driver error
>     Error:Printer halted. kill() called!
>     [ERROR] Error:Printer halted. kill() called!
> 
>     //action:poweroff

Version of Marlin Firmware: 2.0.9.1
Printer model: Creality Ender 5 Plus
Electronics: SKR 1.4 Turbo, BTT TFT35,
Add-ons: TMC2208, Biqu H2 Extruder, dual Z

rsperry79, 2021-08-11:
> The root cause is probably the bug described in #15797

I get the same error when I am entering commands in the terminal manually. Turing STEALTHCHOP off for E  in the control panel did allow me to run the following commands without error.  With it on I get the above error.

G0 X10 Y10 F9000 
G0 Z0.15 
G1 Y90 E50 F500

rsperry79, 2021-08-22:
Pulled the latest bugfix and this is now working as expected

wundi22, 2021-09-06:
Unfortunately the exact same issue here for me. It just stops printing on the first layer and I need to reset the printer.

my fault code is again different to the others
Recv: E driver error detected: 0x1075314720 
Recv: coil short circuit

Version of Marlin Firmware
bugfix-2.0.x (Latest Revision 06 September 2021)

Printer model
Custom - CoreYX motion system

Electronics
BTT SKR Pro 1.2 with TMC2225 drivers (essentially the same as TMC2208)

Add-ons
Inductive probe connected to the BLTouch pin set

Your Slicer
PrusaSlicer2.3.3

Host Software
OctoPrint

elminster, 2021-09-17:
Same issue here. Although I get the error for Z rather than E. And the issue normally takes several hours before it occurs.

BTT SKR 1.4 Turbo with TMS2209, sensorless homing and BLTouch homing.

Changing Z (and I did E0 to be on the safe side) from stealthchop to spreadcycles fixed the issue, and allowed prints that never finished to complete.

Upgraded from 2.0.5_3 (ran for a year or so, no issues) to 2.0.9_1.

rick-ryerson, 2021-11-17:
Replying to keep alive. It would be nice to have corrected so I can update to latest version.

arminth, 2021-12-27:
Randomly, I have the same issue with a 2208 on E0. When I deactivate stealthchop, it seems to work. I will now try this solution:
https://techoverflow.net/2021/11/16/how-to-fix-marlin-random-driver-error-detected-coil-short-circuit-crash-trinamic-tmc/
will report on any success.
Cheers
Armin

arminth, 2021-12-29:
Didn't compile with latest bugfix. so no solution.

tockmos, 2022-01-09:
I have BTT SKR 2. I get: E driver error detected: 0x40190010, coil short circuit. after 2 min. 
Tried to change stepper and inserting st.diss2g(true); in trinamic.cpp no luck.
Can’t downgrade to Marlin 2.0.4 because it doesn’t recognize SKR 2.
NEED HELP!

arminth, 2022-01-09:
Try to switch the E Stepper to spread cycle. This helped for me and I didn't get any errors any more. Not that it is a great solution but a solution after all.

tockmos, 2022-01-09:
> stealthchop
Didn’t help :(

tockmos, 2022-01-09:
> Try to switch the E Stepper to spread cycle. This helped for me and I didn't get any errors any more. Not that it is a great solution but a solution after all.

It seems to work, (didn’t store it in EEPROM on first try). 
I have printed for 10 min now 😊
Thank You!

arminth, 2022-01-09:
Good luck for next 100 hours!

blindley125, 2022-01-12:
So I have this same issue when compiling the latest release 2.0.9.03. I have been able to determine that it is related to Linear Advance. If I leave it disabled everything runs as normal. If I enable Linear advance it faults out with the Same Error posted in this thread. For now I am just running without Linear Advance .

arminth, 2022-01-12:
You can use linear advance but your have to switch to spread cycle!

dannyglover, 2022-03-10:
Any progress on this issue? I'm also experiencing it.

blindley125, 2022-03-10:
Actually the driver burnt up and I switched to an SKR 2 board. However
after investigating the board I believe I had the jumpers in UART and SPI
mode which may be the leading issue so maybe start there.

dannyglover, 2022-03-10:
> Actually the driver burnt up and I switched to an SKR 2 board. However after investigating the board I believe I had the jumpers in UART and SPI mode which may be the leading issue so maybe start there.
> […](#)

Hey.

I don't have that problem at my end at least. I'm using an SKR 1.4 and have repeatedly checked that my 2208's are setup for UART both in hardware and firmware, and the correct jumpers are used.

I did have an interesting development though. With LIN_ADVANCE enabled, the 4 prints I had failed midway through the first layer, whilst printing the XYZ calibration cube, at the same exact place. Moaning about the E driver.

After disabling LIN_ADVANCE, I tried printing it again. It completed the print first go with no issues.
Thoughts?

blindley125, 2022-03-10:
That was my issue in a nut shell as soon as linear advance was used it would stop extruding or stop with E driver error . I don’t have this issue on the same board on my Ender 5 plus using 2209 and marlin 2.0.9.1. So maybe try an earlier version of marlin as well.

thisiskeithb, 2022-03-10:
@dannyglover If you're running TMC2208s/TMC2225s, this is a known issue and they will shutdown: https://github.com/MarlinFirmware/Marlin/issues/16076.

Disable `LIN_ADVANCE` or switch to a different driver.

dannyglover, 2022-03-10:
> @dannyglover If you're running TMC2208s/TMC2225s, this is a known issue and they will shutdown: #16076.
> 
> Disable `LIN_ADVANCE` or switch to a different driver.

Thanks Keith. Good to know.
Thankfully I don't use linear advance anyway, I just forgot to disable it initially in the config file.

I assume this issue isn't a "you can't use linear advance with these drivers" and more of a "we're looking into fixing it" situation?
Cheers.

dan-and, 2022-03-16:
Got the same issue when doing long prints. It takes at least 7hours before the Coil Error occurs. 
I use SKR 1.4 Turbo with TMC2209 with Linear Advance and Stealthchop enabled with bugfix-2.0.x compiled 
I will first switch to Spread Cycle (and keep Lin Advance activated) and inform the results in the next days. 

> FIRMWARE_NAME:Marlin bugfix-2.0.x (Mar 16 2022 07:43:33) [SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin](http://source_code_url:github.com/MarlinFirmware/Marlin) PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 Pro EXTRUDER_COUNT:1

> Send: N305830 G1 X31.73 Y44.156 F10800*127
Recv: ok
Send: N305831 G1 X31.807 Y44.017*30
Recv: ok
Send: N305832 M204 P800*82
Recv: ok
Send: N305833 G1 F900*73
Recv: 
Recv: E driver error detected: 0xC0190060
Recv: coil short circuit
Recv: 		X	Y	Z	Z2	E
Recv: Address		0	0	0	0	0
Recv: Enabled		true	true	true	true	true
Recv: Set current	1000	1000	750	750	800
Recv: RMS current	994	994	734	734	795
Recv: MAX current	1402	1402	1035	1035	1121
Recv: Run current	17/31	17/31	23/31	23/31	25/31
Recv: Hold current	8/31	8/31	11/31	11/31	12/31
Recv: CS actual	17/31	17/31	11/31	11/31	25/31
Recv: PWM scale
Recv: vsense		0=.325	0=.325	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true
Recv: msteps		16	16	16	16	16
Recv: interp		true	true	true	true	true
Recv: tstep		12876	26	max	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false	false
Recv: pwm scale sum	38	255	35	10	207
Recv: pwm scale auto	1	31	2	0	0
Recv: pwm offset auto	67	67	90	27	255
Recv: pwm grad auto	22	22	40	39	24
Recv: off time	4	4	4	4	4
Recv: blank time	24	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2	2
Recv:  -start		1	1	1	1	1
Recv: Stallguard thrs	0	0	0	0	0
Recv: uStep count	649	794	440	712	360
Recv: DRVSTATUS	X	Y	Z	Z2	E
Recv: sg_result	68	180	142	110	306
Recv: stst		*	*			*
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb						*
Recv: Driver registers:
Recv: 		X	0xC0:11:00:00
Recv: 		Y	0xC0:11:00:00
Recv: 		Z	0xC0:0B:00:00
Recv: 		Z2	0xC0:0B:00:00
Recv: 		E	0xC0:19:00:60
Recv: 
Recv: 
Recv: echo:Driver error
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Printing" to "Error"
Send: M112
Send: N305834 M112*24
Send: N305835 M104 T0 S0*25
Send: N305836 M140 S0*94
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor

azjose, 2022-04-14:
I have the same issues on Y though. This on MKS Robin Nano v1.2 using 2.0.9.3 2208 drivers

Not using Linear Advance and turned off stealth chop.

`Recv: Y driver error detected: 0x40190010
Recv: coil short circuit
Recv: 		X	Y	Z	Z2
Recv: Enabled		true	true	true	true
Recv: Set current	800	800	800	800
Recv: RMS current	795	795	795	795
Recv: MAX current	1121	1121	1121	1121
Recv: Run current	25/31	25/31	25/31	25/31
Recv: Hold current	12/31	12/31	12/31	12/31
Recv: CS actual	25/31	25/31	18/31	18/31
Recv: PWM scale
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: interp		true	true	true	true
Recv: tstep		161	159	max	max
Recv: PWM thresh.
Recv: [mm/s]
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: pwm scale sum	74	68	30	28
Recv: pwm scale auto	-2	0	2	0
Recv: pwm offset auto	58	45	51	50
Recv: pwm grad auto	12	19	30	48
Recv: off time	4	4	4	4
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv:  -end		2	2	2	2
Recv:  -start		1	1	1	1
Recv: Stallguard thrs
Recv: uStep count	8	504	968	424
Recv: DRVSTATUS	X	Y	Z	Z2
Recv: sg_result
Recv: stst
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa			*
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xC0:19:00:00
Recv: 		Y	0xC0:19:00:10
Recv: 		Z	0xC0:0E:00:00
Recv: 		Z2	0xC0:0E:00:00
Recv: 
Recv: 
Recv: echo:Driver error
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Printing" to "Error"`

ellensp, 2022-04-28:
An Interesting discovery on another board https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/issues/114
I don't know if this is related, but the 1.4  board also has tmc's butted together.
So give this a try. Put some kapton tape between the stepper module that errors and its neighbours

rick-ryerson, 2022-04-28:
@ellensp considering that link is to a "theory" of a hardware problem, and that this problem is corrected by going back to a previous version of the software, I'm going to say they aren't related.

charlesrg, 2022-06-13:
Having the same problem with BTT SKR PRO 1.2.  Trying to print the same pattern and both times errors on first layer.
Recv: E driver error detected: 0x40170060
I'm running 2.0.9.4

parallyze, 2022-06-13:
> Having the same problem with BTT SKR PRO 1.2. Trying to print the same pattern and both times errors on first layer. Recv: E driver error detected: 0x40170060 I'm running 2.0.9.4

Out of curiousity: Which "same problem" do you refer to? Looking at this thread there's the inital post with a
different driver error, various reports of people having "same problems" with either the e or the z axis and different
error codes.... and the ones who disabled StealthChop on the extruder, which seemed to solve their problem.
 And the ones who said they had StealthChop disabled and problems with the Y axis right before posting a log which says "true" for "stealthChop" on X, Y, Z and Z2.... maybe you'd like to add one or more details of your setup, the stepper drivers you're running and if you're using linear advance in combination with stealthchop? Maybe config files? It's a bit hard trying to keep all those "same problems" apart.... ^^

And unless I'm missing it the thread starter didn't tell if disabling StealthChop on E did change anything... according to the config files he's using TMC2208s, LIN_ADVANCE and STEALTHCHOP_E.

#16076 might be what's causing trouble for you, try disabling StealthChop on E if you're using TMC2208/TMC2225.

charlesrg, 2022-06-13:
Sure, I'm doing a print test now to confirm I can reproduce the problem just by changing firmware.  So far no issues with 2.0.9.3
I've tried 2.0.9.4 twice and both failed, now trying 2.0.9.3 and will try bugfix-2.0.x next.
My error is on the Extruder: Recv: E driver error detected: 0x40170060

I'm using TMC2209 with BIQU H2
Both binaries have #define STEALTHCHOP_E 

Linear Advance is enabled in the firmware but disabled on the GCODE as I didn't have time to tune it yet.

charlesrg, 2022-06-14:
I ran the same 5 hour print on bugfix-2.0.x and I've got no error. I've Configuration.h Configuration_adv.h and pins exactly the same as on 2.0.9.4. So I'm assuming the issue is resolved  in the bugfix.

joaoachando, 2022-06-21:
I have an skr 1.4 turbo with tmc2208 and can confirm that this error still exists in bugfix 2.1.x

AapoTahkola, 2022-07-05:
I got Fysetc Spider with a couple TMC2130 drivers to a state that it errors out immediately as I move X axis. I swapped out the offending motor and stepper once already. Disabling short circuit protection killed(I think) one of the drivers so no more playing around with that. marlinbft only works 20% or so of the time so cannot really proceed until I find a better way to upload firmware. I am not exactly sure which branch I was using earlier but CONFIGURATION_H_VERSION was 02000904 and it did not have this kind of issue. 2.1.x maybe.
I have lowered the steppings because I realized that there is no point in trying to go below 0.01mm with the mechanics I have.
Z is tmc2209 rest are tmc2130
`
Recv: X driver error detected: 0xB0150000
Recv: coil short circuit
Recv: 		X	Y	Z	E
Recv: Address				0
Recv: Enabled		true	true	true	false
Recv: Set current	700	900	800	1000
Recv: RMS current	673	887	795	994
Recv: MAX current	949	1251	1121	1402
Recv: Run current	21/31	28/31	25/31	17/31
Recv: Hold current	10/31	14/31	12/31	8/31
Recv: CS actual	21/31	28/31	12/31	8/31
Recv: PWM scale	0	48		55
Recv: vsense		1=.18	1=.18	1=.18	0=.325
Recv: stealthChop	true	true	true	true
Recv: msteps		8	8	4	16
Recv: interp		true	true	true	true
Recv: tstep		max	max	max	max
Recv: PWM thresh.	98	98	658	6325
Recv: [mm/s]		100	100	3	1
Recv: OT prewarn	false	false	false	false
Recv: triggered
Recv:  OTP		false	false	false	false
Recv: pwm scale sum			58
Recv: pwm scale auto			2
Recv: pwm offset auto			139
Recv: pwm grad auto			14
Recv: off time	5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv:  -end		0	0	0	0
Recv:  -start		6	6	6	6
Recv: Stallguard thrs	6	5	0	0
Recv: uStep count	912	944	992	456
Recv: DRVSTATUS	X	Y	Z	E
Recv: sg_result	0	0	0	0
Recv: stallguard
Recv: fsactive
Recv: stst
Recv: olb
Recv: ola		*
Recv: s2gb		*
Recv: s2ga
Recv: otpw
Recv: ot
Recv: 157C
Recv: 150C
Recv: 143C
Recv: 120C
Recv: s2vsa
Recv: s2vsb
Recv: Driver registers:
Recv: 		X	0xB0:15:00:00
Recv: 		Y	0x80:1C:00:00
Recv: 		Z	0xC0:0C:00:00
Recv: 		E	0x80:08:00:00
Recv: 
Recv: 
Recv: echo:Driver error
Recv: Error:Printer halted. kill() called!
Changing monitoring state from "Operational" to "Error"

`

AapoTahkola, 2022-07-13:
It seems that there is another bug which means that none of the defines STEALTHCHOP_XY etc. have no effect and stealth chop is always on.
#undef STEALTHCHOP_XY
#undef STEALTHCHOP_Z
#undef STEALTHCHOP_E
also has no effect
but "M569 S0 X Y Z E" followed by "M500" does disable them. More noisy and M122 has them disabled.

rick-ryerson, 2022-07-13:
It's been a year now. Can we expect someone to be assigned to this bug?

AapoTahkola, 2022-07-13:
I think the Linux kernel folks deal with this kind of stuff all the time. If things start getting worse then at some point the process of booting patches even at random starts until things start looking little bit better again.

closed 2022-07-14 by thisiskeithb