MarlinFirmware/Marlin #22315: [BUG] (bug summary)M42 s4 S255(not work)

espadan3d, 2021-07-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

hi
in older  version when import M42 s4 S255 in gcode active s4 but in new version (bugfix-2.0.x)not work!
should be change one of menu in marlinfirmware?

### Version of Marlin Firmware

bugfix-2.0.x

ellensp, 2021-07-08:
Incorrect syntax
M42 P4 S255 is probabably what your after

closed 2021-07-08 by ellensp

ellensp, 2021-07-08:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

espadan3d, 2021-07-08:
sorry my manes M42 P4 S255

ellensp, 2021-07-08:
Tested on ramps, default config + DIRECT_PIN_CONTROL enabled with a oscilloscope. 
with M42 P4 S255 Pin 4 does as expected goes high and stays there.
Also tested M42 P4 S128 and M42 P4  S0. 
All work as expected

espadan3d, 2021-07-08:
Checked not enable in my version ,enabled
Thanks a lot