MarlinFirmware/Marlin #22317: [BUG] When coolant flow is zero, laser stops but stays ON

Fusseldieb, 2021-07-08:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

While using Marlin on our 60W Laser Cutter I implemented a coolant flow meter and used the magic of Marlin to handle it.
It shows the flow properly, however as soon as as I turn off the chiller and the machine is midprint of an SD file, it stops, starts to buzz and the laser *stays on* the whole time until I forcefully turn the entire laser off.

### Bug Timeline

Since I used the feature

### Expected behavior

Machine should stop and turn off the laser "immediately"

### Actual behavior

Machine will stop, buzz and laser will stay on indefinitively

### Steps to Reproduce

1. Generate GCODE file with "Lightburn" and put on SD card ("Laser inline" mode, if that matters)
2. Start file on laser and wait until its in the middle of a cut
3. Stop coolant flow
4. Observe

### Version of Marlin Firmware

2.0.x

### Printer model

60W Laser, adapted

### Electronics

SKR 1.4 Turbo

### Add-ons

FLOWMETER

### Your Slicer

Other (explain below)

### Host Software

SD Card (headless)

descipher, 2021-07-08:
Inline mode does matter, the laser is disabled via cutter.disable(), this does not control inline mode at the moment. Inline mode has a separate control. I am addressing that problem  in another issue

descipher, 2021-07-08:
Also the laser does not always turn off when pausing.

descipher, 2021-07-08:
See #22152

thinkyhead, 2021-07-30:
> Inline mode has a separate control…

If there is a simple ON/OFF state that the inline code uses to decide whether the laser is on or off —and to ignore further inline power blocks— then that flag should be set OFF as part of the coolant safety thing.

descipher, 2021-07-30:
Yes, I have read the other PR that deals with this issue. Not sure thats the best method. I have resolved the current issues but there is more to fix.

thinkyhead, 2021-07-30:
I had planned to go through all that code and study it and try to understand how it all works, taking lots of notes, then doing some cleaning up and re-writing…. But there has just not been the time, and I no longer have a laser here to test with.

descipher, 2021-07-30:
I will do testing on my laser in about 2 weeks. We can go over the functional elements once I verify what I have done so far. I think we can add dynamic power like GRBL as an inline option. I did fix the flag to control enabling inline mode and can now address the coolant safety control issue with inline mode. Right now I am just using leds to validate function. I am not near the laser right now which means I can’t do real timing validation with the stepper/planner pwm output control.

Fusseldieb, 2021-08-01:
> I will do testing on my laser in about 2 weeks. We can go over the functional elements once I verify what I have done so far. I think we can add dynamic power like GRBL as an inline option. I did [...]

Do you want me to test anything? I have access to the laser monday onwards...

descipher, 2021-08-01:
I will submit a PR this week to address the flow sensor safety and the M3/4/5 compatibility issue. Testing will be appreciated, thanks.

descipher, 2021-08-10:
Update, my preliminary testing resulted in more work on my end. Most of the laser mode compatibility issues have been addressed. There are multiple calls that do the same function and to many features that have zero adoption by control software products e.g. Lightburn, LaserGRBL, LaserWEB etc. so I am prepping a PR to do a discussion of the current features and what we can consider to do with  them. Should be in this week

descipher, 2021-10-10:
22690 Should be merged soon, all tests passed, update docs in progress.

closed 2021-12-21 by github-actions[bot]