MarlinFirmware/Marlin #22322: [BUG] z probe offset isn't working

CMcKenzie74, 2021-07-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

1. I run a bed level using BLTouch and store setting.
2. do a test print and use babystepping to find the z probe offset.
3. put the value from previous step into the z probe offset under configuration and store setting.
4. when doing a print after these steps the z probe offset seems to be ignored and I have to run through step 2 again.
5. if i power off the printer the prints seem to work fine unless I run another bed level which forces me to run through step 2 again and the z probe offset value keeps getting bigger.
[My Marlin setup files.zip](https://github.com/MarlinFirmware/Marlin/files/6792784/My.Marlin.setup.files.zip)

### Bug Timeline

new I have just upgradded my ender 3 Max with a BTT E3 turbo and BL Touch

### Expected behavior

I expected the setting for the Z probe to stay in config and retain it for every print even after a g29.

### Actual behavior

the z probe offset seems to be ignored.

### Steps to Reproduce

1 run a bed level procedure (g29).
2 run a print. 

### Version of Marlin Firmware

bugfix-2.0.x downloaded 09\07\21

### Printer model

creality ender 3 Max

### Electronics

BTT e3 turbo board

### Add-ons

BL Touch

### Your Slicer

Cura

### Host Software

SD Card (headless)

lmocanu, 2021-07-10:
I can confirm I have been having the same issue with Z-offset babystepping not being taken into account after restarting print.(bugfix branch, with E3 Mini v1.2 and BLTouch  clone).

CBDesignS, 2021-07-11:
(quote)
I run a bed level using BLTouch and store setting.
do a test print and use babystepping to find the z probe offset.

Is the bltouch rotuine saving the mesh  for you  ? Once you have worked out the Z offset are you saving the settings then reloading the settings to make the new settings active ?? m500 saves eeprom then you need to m501 to load and activate the new settings ? if you do it in terminal. If you use lcd then once you enter offset then you need to store settings then you need to Load settings so that the controller knows it is working with new offset. 

you say it works after a reset (because it is forced to use the new updated / saved offset) It may be the fact you are missing 1 step and that is possibly causing this problem.

lmocanu, 2021-07-11:
It is not an issue of saving/loading configuration; the issue is that after reset, saved Z-babysteps values are not added to current Z value. Thus, Z-babysteps need to be set again; Since the old value is still there, its value increases again and again.

CMcKenzie74, 2021-07-11:
After the test print and changing the z offset using baby stepping I did then store the setting, I wasn’t aware that you then needed to load the setting to make the new z offset setting active I will try it and let you know.

CBDesignS, 2021-07-11:
lmocanu I must have been miss informed because I was told that babysteps were only an "on the fly" adjustment to help get a good first layre if you could not dial in your settings while configuring the Z height & offsets, Are you using a tft in tft mode ? as in btt tft35/mks tft50 etc with the posh looking menus and icons or a standard 12864 Marlin type lcd ?? I can remember a while ago that the babysteps had problems on the tft screens because of the way they work (they are a controller & not just an information display) like the good ole 12864 screens and the problem was in the screen firmware & not with marlin firmware. " the reason I changed back from posh tft to the good ole 12864 lcd screen"

lmocanu, 2021-07-11:
Yes, I use the TFT35 v3 screen in GUI mode. I have not tried the old LCD format tbh.

CBDesignS, 2021-07-11:
I would have a try in Marlin mode instead of gui/tft mode. My tft35 is in the box as I could never get it to work as I wanted. I spent many an unhappy hour in the firmware trying to get it working and gave up. Those screens are mini controllers that have firmware  and the firmware used to be buggy as hell (the problem may be fixed in newer firmware updates) but I gave up because the bugs were reported and never fixed unless a user could code and fix the mess left by MKS & BTT .. Check for the latest screen firmware and try Marlin mode to see if it helps.

nsarzyns, 2021-07-12:
I have a similar issue with the Z probe being incorrect and the Z probe offset wizard just not doing anything useful. When I updated, I didn't change anything on my printer or with the mount but using my old Z probe offset of -1.31 was no longer valid. I used the Z probe offset wizard and that gave me a value of -1.45, still very wrong. After trial and error, I found a working value to be -1.91

Nothing changed on the printer except the firmware. Eeprom with reset and reinitialized as a test didn't fix the issue.

loetefix, 2021-07-18:
same here. z offset takes no effect on skr1.4turbo.

CMcKenzie74, 2021-07-19:
> lmocanu I must have been miss informed because I was told that babysteps were only an "on the fly" adjustment to help get a good first layre if you could not dial in your settings while configuring [...]

I am aware that babystepping is a way of getting your reference offset, which is what I put in my initial description of this issue. I do have a MKS TFT35 installed on my printer but I mostly use it in marlin mode as I have found that it can be problematic in touch screen mode.

CBDesignS, 2021-07-19:
I think this still all relates to the firmware for the screen. Maybe the screen is saving the babystep data & if adjusted it is not overwriting it but just adding to the saved data (this may explain the problem you are getting) I do not use babystepping but I have the same hardware and have no problem with it. I have spent many an hour getting the glass beds level as possible tho.  I hope you find the solution and get printing without tinkering..

DerAndere1, 2021-07-20:
This looks like a duplicate of issue https://github.com/MarlinFirmware/Marlin/issues/21833

thinkyhead, 2021-07-30:
More details are needed. Please log information using `M420` after each adjustment, before and after saving to EEPROM, after reboot, etc. Of course, make sure you are actually using `BABYSTEP_ZPROBE_OFFSET` and that the display interface is actually doing the right thing with respect to this option. The more data we have, the better able we are to narrow down the true problem.

Zloyuzver, 2021-08-04:
Marlin 2.0.9.1 and buffix for him. The printer does not remember the configured Z-offset. Configured from marlin 12864 screen emulation on TFT43 from BTT. With every new print, you have to re-adjust the Z-offset. Automatic bed leveling is not activated in firware settings (Configuration.h and Configuration_adv.h). After setting up the Z-offset, I saved them to the EEPROM and immediately loaded them from there. Nothing helped.

GJSchaller, 2021-08-12:
I'm running into similar issues with an EZABL and a SKR mini E3 v2.0 - no matter what I seem to set the Z Offset to, the print will begin (which includes a G29), and then the Z offset is wrong - under 2.0.8 (the BTT fork), it would be too close to the bed, under 2.0.9.1, the nozzle would be too high.  I'm using Bilinerar leveling if that matters at all, and not even doing Babystepping - the initial Wizard results are close, and then the nozzle is off to the point that the filament is either entirely blocked, or so loose it doesn't squish / adhere to the bed at all.

I'll try to log some M420s over the next few nights when I am home from work and have the free time.

GJSchaller, 2021-08-14:
OK, a bit more information: not what I was looking for, but interesting none the less.  This was my original plan:

- Level Bed using G29
- Power Cycle Printer
- M420 (Part 1)
- Run Bed Leveling Probe
- M420 (Part 2)
- Print Test Cube WITHOUT G29
- M420 (Part 3)
- Power Cycle Printer
- M420 (Part 4)
- Run Bed Leveling Probe
- M420 (Part 5)
- Print Test Cube WITH G29
- M420 (Part 6)

I finished up to Part 4 without an issue - the test print worked like a charm.  I power cycled the printer, did an M420, and ran the probe again... and that's when things got weird.  No bed leveling was involved.

When I ran the probe prior to the second print, I had the following values on the screen:

Z0 5.00
Probe Z Offset 3.60

I will note that the result of the prior probe was to set the offset to 1.40.  It looks like when you run the probe again, it somehow pulls the current value and doesn't start fresh?

GJSchaller, 2021-08-14:
Ok, did some more testing.  The issue I mentioned with the Wizard above may be a separate issue.  Here's the testing I did:

- Compiled 2.0.9.1 w/o Probe Wizard or Babystepping defined, flashed to printer.
- Manually set Z Offset
- Ran test print w/ G29 commented out in gcode
- Test Print w/o G29 worked perfectly
- No changes to setup, ran exact same file just with G29 enabled
- Nozzle too close to bed, Z Probe offset not adhered to
- No changes to setup, ran with G29 commented out again
- Test Print w/o G29 worked perfectly

It looks like running G29 at the start of a print causes the nozzle to be too close to the bed somehow, even if the Z Offset is hardcoded and nothing is changed between prints.

GJSchaller, 2021-08-15:
I inspected my starting Gcode, and did another test - I noticed there was a M420 Z0 in my starting code.  After commenting that line out, everything works fine.

+++

With G29 enabled, Z Offset bad:

M117 Homing All...
G28 ; home all axes
M117 Homing Z Probe...
G28 Z ; home z again
M117 Generating ABL Mesh...
G29; Auto Bed Leveling
; M117 Heaters Recovering...
; G4 S10; wait for heaters to recover
M420 Z0; Z fade
M900 K0; LA
M117 By Ender be Purged!

+++

With G29 enabled, Z Offset works:

M117 Homing All...
G28 ; home all axes
M117 Generating ABL Mesh...
G29; Auto Bed Leveling
; M117 Homing Z Probe...
; G28 Z ; home z again
; M117 Heaters Recovering...
; G4 S10; wait for heaters to recover
; M420 Z0; Z fade
; M900 K0; LA
M117 By Ender be Purged!

GJSchaller, 2021-08-15:
OK, this is getting weirder.  I ran a print yesterday without issue.  After that, I ran a second one, and noticed that the nozzle was too close to the bed, but after the first layer it was fine.  This morning, I re-measured the Z Offset manually, not using a probe, and it had gone from -1.5mm up to +0.5mm over the course of two prints.

I suspect something is changing the value of Z / Home after each print?

GJSchaller, 2021-08-15:
I had a bunch of comments earlier that were more confusing than helpful, so I deleted them and will summarize here.

It seems like when G29 is run, it causes the position of the Z Home to change somehow.  Although the Z Offset itself is the same, the nozzle will hit the bed, and if you recalculate the offset, it's higher than it was before.  Each time you re-run G29, the problem repeats, so the offset value needs to be recalculated between each print.

As a work around, I tried adding a G28 Z to my start code in Cura after the G29, and this seems to be a workaround.  Whatever the bug is that's causing the position of Z's home to change, re-homing Z forces it to update and be correct, so the print can proceed without needing to change the offset value.

So the problem is not Z Offset, it's what the printer considers "Home" for Z is changing when G29 is run.

My setup is an Ender 3 Pro, BTT SKR mini E3 v.20, EZABL, Marlin 2.0.9.1 Vanilla.

GJSchaller, 2021-08-19:
Update - even after a print finishes, and I re-home at the beginning of the next print, the Z Offset needs to be manually updated and recalculated.  It looks like the only way to correct the Z Offset Bug is to potentially reboot / reset / initialize the EEPROM.
I'll try to test more when I am back next week, see if I can confirm what specific actions will clear the issue.

thinkyhead, 2021-08-21:
What kind of LCD display are you using @GJSchaller? Is that display actually updating the value of `M851 Z` or is it actually modifying the value of `M206 Z`?

GJSchaller, 2021-08-21:
> What kind of LCD display are you using @GJSchaller? Is that display actually updating the value of `M851 Z` or is it actually modifying the value of `M206 Z`?

Stock Ender 3 Pro supply.  I can attach / send my Configs if you’d like.

GJSchaller, 2021-08-24:
Z Probe Offset Bug Hunt Process:

Display is defined as: CR10_STOCKDISPLAY

* Power on printer
* Init. EEPROM
* Z Height is currently 5.00
* Run Z Offset Wizard
*** Result is -2.10mm
* Run test print w/o G29
* Run Z Offset Wizard
*** Result is -1.30mm
* Run (same) test print w/o G29
* Run Z Offset Wizard
*** Result is -0.50mm (I think, cat distracted me on the way back to my laptop, but it was definitely not the same.)

Even without the G29, the value of Z is changing after each print.  I can send you my Configuration.h file and the test print file, if you'd like.

GJSchaller, 2021-08-25:
Update - ran some more testing tonight.

* Started by powering on the printer, and flashing the latest bugfix compiled this afternoon.  Loaded defaults.
* Ran Z Probe Offset Wizard.  Staring values: Z0 = 5.00, Probe Offset is 0.00
* Results: Offset: -1.90mm
* Ran test print (no G29)
* Ran Z Probe Offset Wizard again.  Starting values: Z1 = 5.00, Probe Offset is listed as 3.10

Note that 3.10 = 5.00 + -1.90... is it possible the Offset is being re-calculated somewhere?  It's happening outside the Wizard, as it was happening when I set it manually as well.

GJSchaller, 2021-09-01:
In regards to #21833 - my setup is using an EZABL, and it uses the Z-Endstop connection to trigger (as opposed to the BLTouch port on the board).

GJSchaller, 2021-09-22:
There was another bug (#22524) related to the EEPROM on the SKR mini v2 that may be the cause of this?  If it's having problems committing the value when saving it, that may be the cause here.  Is anyone having this problem on a board other than the SKR mini E3 v2?

nsarzyns, 2021-09-22:
Hmm I'm not sure. It seems to get committed to eeprom. I can read back the values between power offs and it does persist. It just needs to be changed with almost every print.

Something just occurred to me, one thing that changed was the environment definition in .pio file. It's late right now so I might not be verbally clear but this issue might have appeared when I had to change which processor I selected the board had. I think before there was some 512k option a lot of people used for the skr v2 but that wrong and was fixed.

Am I not remembering correctly?

GJSchaller, 2021-09-22:
That's actually a good point - even when I reset the EEPROM and re-calculate the Z Offset, it varies from session to session.  Sometimes it's over -2mm, sometimes closer to -1.  It's certainly not as consistent as fixed hardware should be.  That might be a clue to solve this.

moham96, 2021-09-22:
Facing the same issue, very annoying.
I have to change the offset each time i turn on the printer, I go from -1.900 to -1.300 to -1.500, note that the offset is saved fine to EEPROM, it's just somehow used incorrectly each time and that's why i need to change it each time :(

nsarzyns, 2021-10-03:
Today again, between multiple prints I had to change the offset 3 different time. Largest swing between values was 0.7. Again, nothing is changing mechanically.

GJSchaller, 2021-10-11:
@thisiskeithb - It looks like you have a good amount of experience with the SKR mini E2 v2 - are you able to assist with this bug?  What further information can we provide to help track it down and eliminate it?

numpad0, 2021-11-04:
I just encountered what seems a similar problem, albeit in 1.1.9(not 2.x). Z probe offset worked fine for years within -1mm to +1mm range, but today I had to change the value to way over 5mm and it simply didn't work. Maybe some sanity checks or value clamping code is causing it?

psychoph, 2021-11-16:
Any update on this issue? I think I am running into the same issue with a reality silent board 4.2.7 on an ender 3 pro with Marlin 2.0.9.2. I will run the z probe wizard to come up with a consistent -0.7 offset and then when I print I scratch the crap out of the bed until I stop the print. I manually tune the print and I come out with a -0.25 offset that seems to work for short prints.

SwagDad99, 2021-12-25:
I am fighting this on my JGMaker Artist-D Pro running latest upstream 2.0.9.2BugFix.

jurajdanis, 2022-01-11:
Same problem here. Seems like z offset is commited to memory, however same z offset value produces different nozzle-bed gap height. I my case the margin ranges from 0.05 to 1.7. Problematic part about this is, that when I go G28 a then G0 Z0, nozzle hits the bed despite offset was set up fine during the last print (does not matter whether there was a printer shutdown or not). 

M500/M501 does not change anything, Z offset value is stored correctly, seems like (as mentioned in earlier comments) that problem is in Z0/Home position changing randomly during the prints. 

Right now, running marlin 2.0.1/E3V2 4.2.2 with CRTouch. I can't find any higher marlin version with crtouch support to test.

C0rn3j, 2022-01-16:
@jurajdanis 
> I can't find any higher marlin version with crtouch support to test

What do you mean? 2.0.9.3 supports CR Touch just fine

SQL-enwiki, 2022-01-23:
> Rolling back to a previous firmware doesn't have the issue so it's not a physical printer issue.

@nsarzyns What version are you rolling back to that works?

nsarzyns, 2022-01-24:
@SQL-enwiki Honestly I don't remember. I moved to klipper after fighting this issue for too many versions of Marlin. As toxic as the Klipper community can be it's actually really good. Haven't seen this issue ever again since switching. Marlin is great, but this issue pushed me away for at least my current printer

descipher, 2022-02-11:
I can confirm that there is an issue with the offset function manual or wizard on a bltouch. It does not compute correctly.

Loaded  Creality stock  with manual offset:
```
      0      1      2
 0 -0.101 +0.041 -0.234
 1 -0.062 -0.028 -0.254
 2 -0.253 -0.188 -0.289
```
bugfix-2.0.x with manual offset:
```
      0      1      2
 0 +3.987 +4.027 +3.792
 1 +3.988 +4.001 +3.772
 2 +3.853 +3.913 +3.842
```
Same offset M851Z-1.4

descipher, 2022-02-11:
Looks like this issue occurs when the #define PROBE_OFFSET_WIZARD_START_Z -4.0 is not set to start below the probe deploy point. By setting the value to 0.0 I can consistently cause the miscalculation to occur.

After setting it to -4.0 here is the result:
```
Bilinear Leveling Grid:
      0      1      2
 0 -0.106 +0.027 -0.132
 1 -0.084 +0.004 -0.147
 2 -0.179 -0.036 -0.022
```

sarvenn, 2022-03-06:
I have reported an issue about Z offset consistency and adjustment delay. I think ABL function is broken somehow.
https://github.com/MarlinFirmware/Marlin/issues/23781

sarvenn, 2022-03-06:
> I can confirm I have been having the same issue with Z-offset babystepping not being taken into account after restarting print.(bugfix branch, with E3 Mini v1.2 and BLTouch clone).

I have the same issue and in addition to this issue while using the TFT touch mode (BTT TFT35) there is too much delay whenever I adjust my Z offset while printing, it takes like 20 seconds for Z stepper to react.
https://github.com/MarlinFirmware/Marlin/issues/23781

thinkyhead, 2022-04-21:
> Looks like this issue occurs when PROBE_OFFSET_WIZARD_START_Z is not set to start below the probe deploy point….

We could do a sanity-check of the configuration, anyway, to make sure that `PROBE_OFFSET_WIZARD_START_Z` is set below the probe offset point, and maybe some kind of runtime check. Or, maybe there is some way that `PROBE_OFFSET_WIZARD_START_Z` can be ignored or overridden when it's too small.

thinkyhead, 2022-04-21:
> too much delay whenever I adjust my Z offset while printing

That is to be expected. You can try the experimental `INTEGRATED_BABYSTEPPING` option to see if it improves babystepping generally.

SwagDad99, 2022-04-25:
'#define BABYSTEP_ZPROBE_OFFSET // Combine M851 Z and Babystepping' Fixed mine...

DiskDoctor, 2022-04-29:
Any response to the additional info I posted above to help isolate the bug? I can upload configurations is desired

JeremyDWilliams, 2022-05-06:
> When soft endstop for Z is enabled, any command to move below ZERO is stopped at 0, or G28 or simply Z home is resetting Z zero point without including negative z probe offset<

This sounds like a possible suspect.

It's as if the offset amount just isn't getting to where it needs to go, or the wrong (old) variable is being passed.

I was honestly about to ask if UBL was still supported or if it was abandoned, but figured I'd get my head bit off... until I saw this post after hours of searching. I'm glad it's still being supported, but I've gone from having a working machine to one that I can't get to print. (sigh).

The documentation didn't help as it doesn't really explain the correct and full procedure to obtain and enter the correct z probe offset and how to use UBL, or the wizard.

It ignores z offset. It keeps printing about 1.5mm above the bed. It doesn't seem to matter what I enter for the offset. I always get the same printing in space (1.5mm).

During a test to see if turning off soft endstops with the LCD menu, it had no affect. And curiously hitting pause during a test print dropped it into "stop" print, not pause. And the print was fully cancelled.

I also tried adding G29 to gcode to see if that would 'wake up the mesh and z offset, but nope. My initial assumption was that I would not need any special (or new, additional) gcode to let UBL work.

I did try using the wizard, which ended up spending 15 minutes doing nothing while displaying 'printing' complete with minutes to go. This after forcing me to go through a empty/load filament cycle that emptied a half meter or filament on the bed while I have 100mm in the config.h file. 

I mention this other stuff, not as a definitive smoking gun, but as hopefully a clue that some other items may be involved.

Regardless, I do appreciate any and all help I can get to fix this. 

As I've already put more time into this than I care to admit, I would be happy to help improve documentation and/or try to help find the cause of this bug (or user error causing this problem)

ManuelMcLure, 2022-05-06:
A couple of questions:
* Are you homing with the probe, or with a separate end stop?
* If homing with the probe, do you have it connected to the dedicated probe port or to the Z min end stop port?

If it's connected to the probe port and it is easy to do (i.e. the black/white wire from the probe is separate from the servo wire) try switching to the Z min port and enable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` and see if that fixes it.
If it isn't easy to move the probe to Z min you can edit the pins file for your board and swap the Z min and probe pins instead of physically switching it. In either case enable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. There's some weirdness involving the dedicated probe port that strikes some users and not others.

Just to ensure clarity on what's expected when homing with a Z offset, after homing Z the nozzle should move up `Z_AFTER_HOMING` mm (default 10) and should display on the LCD the value of `Z_AFTER_HOMING` _minus_ your configured probe offset (as displayed by `M851`). For example, if your Z probe offset is -1.77 then the Z should show 11.7 after homing.

JeremyDWilliams, 2022-05-07:
> A couple of questions:
> 
> * Are you homing with the probe, or with a separate end stop?
> * If homing with the probe, do you have it connected to the dedicated probe port or to the Z min end stop port?

1) Homing with the probe (BL touch v3.1)

2) The probe is connected to the probe port (not the 2 wire Z stop) on the BTT SKR Mini E3 V3

I will try as you suggest.

And yes, the auto homing shows 10mm minus the negative offset (10mm - (-1.3) = 11.3mm displayed Z height after homing

ManuelMcLure, 2022-05-07:
OK - another couple of questions:
* If after homing, you move Z to Z=0, is the nozzle at the right place (i.e. almost touching the bed)?
* Was the UBL mesh generated before or after the probe offset was set up? The measurement in the mesh point nearest the Z safe homing point should be very close to 0. If it isn't, you should try rebuilding the mesh with the new Z offset configured.

JeremyDWilliams, 2022-05-07:
> OK - another couple of questions:
>
> * If after homing, you move Z to Z=0, is the nozzle at the right place (i.e. almost touching the bed)?
> * Was the UBL mesh generated before or after the probe [...]

1)  yes
2) I have tried both, before and after with no notable difference. At the center of the bed, where it z-homes, the value is Z: 0.052

And a huge THANK YOU! for helping.

I will do the items above as soon as possible... it might be an hour or so.

DiskDoctor, 2022-05-07:
> If it's connected to the probe port and it is easy to do (i.e. the black/white wire from the probe is separate from the servo wire) try switching to the Z min port and enable [...]

This was in the back of my mind, too. It should make zero difference, but this problem should not exist, either.

My goal is to get the z-homing working properly, then use slicer (Cura for now) to set Z-offset for variable first layer thickness. 

I expect to be testing UBL this weekend, which is supposed to not have this problem. 

As long as we keep working this problem, something will turn out. 

I wonder if there is a particular combination of settings causing this incorrect behavior or messing with the variables?

DiskDoctor, 2022-05-07:
> * If after homing, you move Z to Z=0, is the nozzle at the right place (i.e. almost touching the bed)?

Negative on mine. The nozzle is too high and won't move below that point (Z0), until printing.

I could try enabling Babysteps everywhere and experimental, but trying to keep it simple and that would be a workaround, not a solution.

Important to note: Before running z offset wizard, be sure to set z offset to 0.00 and store the settings. Otherwise it just all adds together, which also seems incorrect behavior.

midnightsmith89, 2022-05-10:
I can confirm same issues here, but using an SKR 1.3 board, constant adjustment of the offset for every print is needed.  Attached is my config. 
Note this happens on my TRONXY machine as well, using an SKR 1.4 board.  I have ran through all suggestions here to the same results.  Something is changing the Z home position, because the offset value persists, but is now wrong after each home or print.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/8656119/Marlin.zip)

linuxsense, 2022-05-10:
I have the same issue with a SKR 1.4 Turbo. It will be fine then for some reason it just starts to ignore the probe offset. If I re-flash it will be fine untill suddenly it isn't.

GJSchaller, 2022-05-10:
For what it's worth, I turned off the Probe Offset Wizard and haven't had any issues.  Once I set the value by hand on the printer or in the firmware, it seems stable.  I did notice that my offset somehow changed from -1.85 to -2.85, but once I set it, it's been fine ever since.

midnightsmith89, 2022-05-10:
> I have the same issue with a SKR 1.4 Turbo. It will be fine then for some reason it just starts to ignore the probe offset. If I re-flash it will be fine untill suddenly it isn't.

So much this! a fresh install works for an undetermined length of time, until suddenly it doesnt!

DiskDoctor, 2022-05-10:
Start this video here and see how the z-offset can be done via command line. Notice he turns off soft endstops while setting. I haven't tried this procedure to test if it holds the offset yet. 
[https://youtu.be/ONpKxkil16Q?t=605](url)
What results is anyone seeing if you follow these steps?

ManuelMcLure, 2022-05-10:
If the offset is being lost, it might be something wrong with your EEPROM. First, make sure that the setting of `NOZZLE_TO_PROBE_OFFSET` is correct in your `Configuration.h`. Then upload the firmware and do `M502`/`M500` to ensure the offset is stored correctly in EEPROM.
Then if you start seeing the problem again, do `M851` and see if the value is still correct - if it isn't then something's clobbering EEPROM. If so, you should be able to just do another `M502`/`M500` to get back the value from firmware.

[76 more replies]

closed 2024-06-02 by thisiskeithb