MarlinFirmware/Marlin #22325: [BUG] M666 command issue ?

Guilouz, 2021-07-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

M666 [X<adj>], [Y<adj>] or [Z<adj>] commands have no effect when used from terminal (pronterface), commands are received correctly but there is no change when sending M503. Value are always echo:  M666 X0.0000 Y-0.0000 Z0.0000

When I use Marlin menu in Delta settings and change Ex, Ey or Ez changes are taken into account correctly.

### Version of Marlin Firmware

Latest bugfix

### Electronics

BigTreeTech SKR 2 with BigTreeTech TFT70

ellensp, 2021-07-11:
actually I take that bug confirmed back.

In M666 is this line

```if (v * Z_HOME_DIR <= 0) delta_endstop_adj[i] = v;```

which translate to 
if the provided value * Z_HOME_DIR is <= 0  only then change the delta_endstop_adj

So if we have Z_HOME_DIR = 1
eg 

M666 X10 
v=10, v*Z_HOME_DIR = 10  this is not less than or equal to 0 so it is not updated.
M666 X-10
v=-10, v*Z_HOME_DIR = -10 and this is applied 

So if we have Z_HOME_DIR = -1

eg
M666 X10 
v=10, v*Z_HOME_DIR = -10  and this is applied
M666 X-10
v=-10, v*Z_HOME_DIR = 10 and this is applied  this is not less than or equal to 0 so it is not updated.

So I suspect your not entering negative values when z homing is to Max

Guilouz, 2021-07-11:
@ellensp How do you explain that's working when I change value from TFT but not when I enter value in Terminal ?

When I set a values from TFT, exemple Ex 0.6, Ey 0.4 and Ez 0.2, M503 return correct values for M666 (echo: M666 X0.6000 Y0.4000 Z0.2000) but when I try to change this values, nothing change I have always echo: M666 X0.6000 Y0.4000 Z0.2000. Changes are not taken into account when using the terminal but working only when they defined on TFT.

I use this settings : 

#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR 1

#define X_MIN_POS -135
#define Y_MIN_POS -135
#define Z_MIN_POS 0
#define X_MAX_POS 135
#define Y_MAX_POS 135
#define Z_MAX_POS 320

ellensp, 2021-07-11:
The LCD code just limits the range from -5 to 5, it doesn't look at Z_HOME_DIR

eg
```
  #define EDIT_ENDSTOP_ADJ(LABEL,N) EDIT_ITEM_P(float43, PSTR(LABEL), &delta_endstop_adj.N, -5, 5, _recalc_delta_settings)
  EDIT_ENDSTOP_ADJ("Ex", a);
```

This look to be in error. not the M666 part

ellensp, 2021-07-11:
So the question is should M666 allow the same range as the LCD input?
Or should the LCD be limited to the range of the current M666?

They definitely should match better than they do currently.

Thoughts anyone?

ellensp, 2021-07-11:
I think Im going to go with this diff
Will limit LCD same way as M666

```DIFF
diff --git a/Marlin/src/lcd/menu/menu_delta_calibrate.cpp b/Marlin/src/lcd/menu/menu_delta_calibrate.cpp
index 195afecc1b..1acec5b6aa 100644
--- a/Marlin/src/lcd/menu/menu_delta_calibrate.cpp
+++ b/Marlin/src/lcd/menu/menu_delta_calibrate.cpp
@@ -105,7 +105,7 @@ void lcd_delta_settings() {
   START_MENU();
   BACK_ITEM(MSG_DELTA_CALIBRATE);
   EDIT_ITEM(float52sign, MSG_DELTA_HEIGHT, &delta_height, delta_height - 10, delta_height + 10, _recalc_delta_settings);
-  #define EDIT_ENDSTOP_ADJ(LABEL,N) EDIT_ITEM_P(float43, PSTR(LABEL), &delta_endstop_adj.N, -5, 5, _recalc_delta_settings)
+  #define EDIT_ENDSTOP_ADJ(LABEL,N) EDIT_ITEM_P(float43, PSTR(LABEL), &delta_endstop_adj.N, (Z_HOME_DIR < 0)?0:-10 , (Z_HOME_DIR < 0)?10:0 , _recalc_delta_settings)
   EDIT_ENDSTOP_ADJ("Ex", a);
   EDIT_ENDSTOP_ADJ("Ey", b);
   EDIT_ENDSTOP_ADJ("Ez", c);
```

Guilouz, 2021-07-11:
Concretely what does this mean? That there is a problem on LCD input and not on the M666 side ? Will this allow to be able to change the values ​​via the M666 command? Because for me, the only problem is command sending with terminal.

thinkyhead, 2021-07-11:
It appears that on a DELTA machine M666 only allows negative values to be used. The change for Marlin 2.0.x was made in 6d5fcac54a based on #7697 by @LVD-AC. I'm not sure why, for a delta, we need to multiply the `Z_HOME_DIR` but maybe we were thinking of a future situation where the Z homing on a Delta was done as an extra step using a probe.

In any case, when homing normally on a delta, the M666 command only accepts negative values.

LVD-AC, 2021-07-11:
The delta homing procedure physicly moves each carriage after the endstops are triggered with the values set by M666; moving down is no problem, but moving up (positive values) can potentially destroy the printer. There's probably is a little legroom, but not a lot. The LCD accepting such positive values probably has been overlooked.

thinkyhead, 2021-07-11:
I have pushed a change modifying the LCD delta endstop adjustment code so that you can only set a negative or zero value, and I will soon merge a PR that improves the output and error handling of M666.

Guilouz, 2021-07-11:
@thinkyhead Just try your Pr and it's working as expected now. Thanks.

ellensp, 2021-07-12:
This commit assumes delta is homing to max. Although this is true for 99% of deltas. it is not 100% true
It should determine correct direction of movement based on z_home_dir

ellensp, 2021-07-13:
Will see if its a problem and reopen if required.

closed 2021-07-13 by ellensp