MarlinFirmware/Marlin #22326: [BUG] Tevo Tornado Custom Bootscreen displays garbage

yet-another-average-joe, 2021-07-10:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Tevo Tornado Custom Bootscreen displays garbage.

### Bug Timeline

No idea. Didn't use the feature since 1.1.7 or so !

### Expected behavior

The Custom Bootscreen shouyld display a tornado...

### Actual behavior

It displays garbage.

### Steps to Reproduce

Copy the _Bootscreen from the examples directory (Tevo Tornado 1 or 2)
Define SHOW_CUSTOM_BOOTSCREEN in Configuration.h
Boot Marlin

-> custom bootscreen displays garbage.

### Version of Marlin Firmware

2.0.9.1

### Printer model

Tevo Tornado v1

### Electronics

MKS Base 1.6

### Add-ons

none

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

In _Bootscreen.h : 

`#define CUSTOM_BOOTSCREEN_BMPWIDTH      112`

is not correct.

Should be 120. (15 bytes)

`#define CUSTOM_BOOTSCREEN_BMPWIDTH      120`

solves the problem.

ellensp, 2021-07-10:
issue is in https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Tevo/Tornado/V2%20(MKS%20GEN-L)/_Bootscreen.h 
and
https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Tevo/Tornado/V1%20(MKS%20Base)/_Bootscreen.h

thisiskeithb, 2021-07-10:
It used to work. I'll put in a fix if @ellensp isn't already on it.

ellensp, 2021-07-10:
You go fot it

thisiskeithb, 2021-07-10:
Fixed in https://github.com/MarlinFirmware/Configurations/pull/539

ellensp, 2021-07-10:
Since this is for the Configuration repository and a PR has been created to fix it. Closing this Marlin issue.

closed 2021-07-10 by ellensp