MarlinFirmware/Marlin #22327: [Filament Runout Sensor] (False Positive Immediately at the start of a print)

nnw522, 2021-07-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

At compile time whatever the value is that you put into "FILAMENT_RUNOUT_DISTANCE_MM" on the printer afterwards, the printer will immediately call out a filament change,"False Positive." I have gone back through my Configuration.h file. Everything looks good. I then issued a M412 command in the terminal. Terminal indicated a 7mm setting. I then changed the setting to 100mm and issued a M412 R command. Started a new print, i have not had a filament change come up, at this point I am 50% into a print. I'm not a coder by far, but I feel that there may be an issue with the script in the Configuration.h

### Bug Timeline

07/5/21

### Expected behavior

Runout sensor only triggering on a filament break, jam, or tangle

### Actual behavior

Marlin immediately triggers a filament change as soon as the print begins

### Steps to Reproduce

1. Compile Marlin with filament runout sensor and filament motion sensor uncommented.
2. Flash Marlin
3. Start a print

### Version of Marlin Firmware

2.9.1

### Printer model

Ender 5 Pro

### Electronics

Creality 4.2.2 Board

### Add-ons

Big Tree Tech Smart Filament Sensor

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/6794600/Configuration_adv.zip)

CRCinAU, 2021-07-10:
I've never used `M412 R` before.... If you only use `M412 S1` in your start g-code, does it still trigger?

nnw522, 2021-07-11:
It supposed to reset the code that controls that part of the marlin code? I
am not a programmer by any means.

CRCinAU, 2021-07-11:
I get what its supposed to do - but I've never used it myself and also never seen the problem you're reporting - as such I want to rule out that including `M412 R` isn't the cause of the problem...

tispokes, 2021-07-12:
Same on SWX1

omexlu, 2021-07-12:
Same with artillery Genius on bigtreetech TFT Firmware but only with prusaslicer, cura Works without Problem.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2040#issuecomment-878228560

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2027

nnw522, 2021-07-12:
> bigtreetech/BIGTREETECH-TouchScreenFirmware#2040
> bigtreetech/BIGTREETECH-TouchScreenFirmware#2027

I use cura 4.10. Yes the Big Tree Tech filament sensor is acting up again.
It doesn't matter if I have the runout distance set to the default of 7mm
or 100mm. I recompiled marlin again last night to no avail. Immediately at
the start of the print, after the purge line is complete, Marlin wants a
filament change. I run my last 2 prints with the sensor off. With a full
spool of carbon fiber PLA. I'm not running out of filament. I have not had
any actual filament brakes or filament tangle issues. Maybe if some could
take a look at my configuration.h and my configuration.adv files? Maybe I
have something off in my settings?

nnw522, 2021-07-13:
As far as the M412 code, can you explain a little more?

mriscoc, 2021-07-14:
Could be that your filament sensor has the logic inverted? Change #define FIL_RUNOUT_STATE in your Configuration.h file.

omexlu, 2021-07-14:
> Could be that your filament sensor has the logic inverted? Change #define FIL_RUNOUT_STATE in your Configuration.h file.

Nope.

Sonicboom247, 2021-07-26:
Same with artillery sidewinder X1, I have tested pretty much every logic of inverted True/false. Sensor off/smart and it still triggers about 20 Second into a print, I finally gave up and set it to off in settings

nnw522, 2021-07-28:
Same here, I have a regular sensor I'm trying out and seeing tonight.

Sonicboom247, 2021-07-29:
is it possible the "fix" was left out of the MKS boards? seems like this is more related to the Artillery line which uses the MKS boards. Just tested last night with M75\n start gcode as instructed and still get alarm

thinkyhead, 2021-07-30:
Is this with a directly-connected filament sensor, or one which connects to the screen, or one which is on an add-on board? I am getting the sense that it has more to do with the thing not being directly-connected on some MKS systems. This issue is a duplicate of one or two others.

omexlu, 2021-07-30:
With me directly connected to the screen.

thinkyhead, 2021-07-30:
OK. Perhaps there is something funky with the pin state on the screen device, such that we need to wait until the pin state is properly initialized by the screen and ignore it until that period of time has passed.

GMagician, 2021-07-31:
@nnw522 may you try proposed fix?

Sonicboom247, 2021-07-31:
just had one false positive almost 7 hrs into a 17 hr print with my settings of filament sensor inverted =true, runout time = 1000ms, and start gcode of M75.  not bad i would say so i may bump it up to 2000ms to be safe as if still have about 300mm left of filament at runout on the Sidewinder X1.

Sonicboom247, 2021-08-01:
update 2, after leaving the LED on, the print finished with no further filament runout alarms.

nnw522, 2021-08-02:
I'm going to give this a try tonight.

Sonicboom247, 2021-08-06:
Oddly enough i cant get one of my two sidewinders to behave the same as the other, it immediately gives a filament run-out error at start of print. Only difference is one has a clear sensor and one black. Tested inverted, and even setting it off in TFT gives error

nnw522, 2021-08-20:
I ordered a new Big Tree Tech Smart Sensor today, just to try to eliminate that from this issue. I did open up the control box to see if there is a issue with the wiring or if the plug had fallen out. Those were both fine. I tried inverting the logic on the pin. no change. Immediate call for a filament change at the beginning of the print. Also I haven't discounted the possibility that there may be an issue with the board.

Sonicboom247, 2021-08-20:
On one of my sidewinders the filament sensor seems to have gone bad and only reads closed or green so that one is dual LED and is clear plastic. The other Sidewinder has a black filament sensor with only one blue led, that one does not trigger so not sure if it is an issue with a dual LED. I will test my new firmware changes with BTT with their dual LED genius settings

Sonicboom247, 2021-08-23:
Update, so this time I removed FIL_RUNOUT_PIN 4 from config and reflashed, turned off the LED so I can go to sleep and not 5 min passed i get filament run-out alarm.  Ran in and turned on LED, clicked resume, and no issues. Will be adding FIL_RUNOUT_PIN 4 back into my firmware, should we request a #define FIL_RUNOUT_PIN be added to configuration.h?