MarlinFirmware/Marlin #22328: [BUG] (bug summary)CUSTOM_USER_BUTTONS not working

espadan3d, 2021-07-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

hi
I active CUSTOM_USER_BUTTONS for pin 42&44 to have a pause button but when connect GND to d42 nothing happens(also i check by 5v but nothing happens)
[image]

I change true/false & low/high but nothing happens

### Version of Marlin Firmware

bugfix-2.0.x

espadan3d, 2021-07-10:
For test I even entered the PIN number in this tab but nothing happens
[image]

ellensp, 2021-07-10:
_WHEN_PRINTING should "true" or "false" not  "False"

Pins debugging just shows the pin name in a M43, it does not set the pin.

espadan3d, 2021-07-10:
oh
 if I want pause print when connect GND to d42 and resume when connect GND to d44 below code is correct?
[image]

ellensp, 2021-07-10:
no. you need 
#define BUTTON1_WHEN_PRINTING true
#define BUTTON2_WHEN_PRINTING true

espadan3d, 2021-07-10:
Thanks, solved
But in my sample m109 does not work right
I wired so that push pause button  disconnected heater now receive error and failed print 
You have a solution?

ellensp, 2021-07-10:
What are you trying to get M109 to do?

closed 2021-07-10 by thisiskeithb

espadan3d, 2021-07-10:
May be need pause a long time and I want cool heater when machine is pause that the piece does not break by pause also disconnect heater
when change button of pause to resume connect heater this time I need M109 that not resume print before heating

espadan3d, 2021-07-11:
If i open a new issues for this problem (false alarm) ,Is there a solution?

thisiskeithb, 2021-07-11:
> If i open a new issues for this problem (false alarm) ,Is there a solution?

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).