MarlinFirmware/Marlin #22330: TFT mode no printing6

AntoszHUN, 2021-07-10:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Hi! I downloaded the TFT firmware from GitHub 2-3 days ago. (Not the official release!) Since I am developing I am not using the official release. This is the current firmware on GitHubon. The machine does not print in TFT mode. Prints only in Marlin mode. What's the mistake? This is what he does in the video. It heats up the devices and does nothing afterwards. What the hell?

### Version of Marlin Firmware

bugfix_2.0.x

thisiskeithb, 2021-07-10:
This has nothing to do with Marlin.

closed 2021-07-10 by thisiskeithb

AntoszHUN, 2021-07-10:
O sorry I pushed aside.