MarlinFirmware/Marlin #22335: [BUG] Z_SAFE_HOMING

espadan3d, 2021-07-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hi
When I enable Z_SAFE_HOMING below (pic) 
by autohome or when stat print move x&y to 0 than have a annoying sound and not move to center and start moving Z axis (head is in x0 y0) and start print in x0y0 and hits the sides!!!
Everything works fine with disabling
[image]

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

corexy

ellensp, 2021-07-11:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-07-11 by ellensp

espadan3d, 2021-07-11:
I raised in 2 forums but did not receive a response
https://discord.gg/n5NJ59y
RepRap.org Marlin Forum