MarlinFirmware/Marlin #22339: In Marlin 2.0.9.1, 250000 baud rate printer stops randomly. In Marlin 2.0.7 and before, the baud rate of 250000 was running smoothly

TimeJammer, 2021-07-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hello to everyone. I have mks E3D card and mks tft 35. In Marlin 2.0.9.1, 250000 baud rate printer stops randomly. In Marlin 2.0.7 and before, the baud rate of 250000 was running smoothly for 1.5 years.
In version 2.0.9.1 it is stable only at 115200. What is the reason of this ?

I did not cancel the original screen of the printer (Ender 3) and I use it together with tft 35. //#define SERIAL_PORT_2 The -1 setting does not work in the new marlin and I disabled it because it was said to be disabled in the new suggestions. Could this be the reason? While other versions do not have this problem, what could be the reason why 250000 baud rate does not work stable only in 2.0.9.1. Could it be caused by something or a change that I did wrong in this version? Thanks for your responses.

Motherboard: Mks E3D(tmc2209)
Screen: Mks TFT35 with wireless.
Ank mks 3d touch sensor.

Thanks in advance for your returns.

### Bug Timeline

New Bug. In Marlin 2.0.9.1

### Version of Marlin Firmware

Marlin 2.0.9.1

### Printer model

Ender 3 MKS E3D Board + TFT35

### Your Slicer

Cura

### Host Software

Other (explain below)

ellensp, 2021-07-12:
Please attach your configuration files.

TimeJammer, 2021-07-12:
> Please attach your configuration files.

First of all, thanks for the reply.

My Config Files is here. Thakns again.

https://drive.google.com/drive/folders/1It5f6x7WfwWC8Tr32RBmpx9jTejCS1e9?usp=sharing

ellensp, 2021-07-12:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6800026/Configuration.zip)
Config files stored locally.

TimeJammer, 2021-07-12:
Ok thans for

> [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6800026/Configuration.zip)
> Config files stored locally.

Thanks for the warning. 
 
The baud rate in the config file I sent is 115200.
I'm having a random stop at 250000. The previous marlin version didn't have this.

ellensp, 2021-07-12:
Did you test absolutely latest bugfix ?
as this was added 5 days ago https://github.com/MarlinFirmware/Marlin/pull/22292

TimeJammer, 2021-07-12:
Also the same is happening in the example config file.

It is attached to my files.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6800375/Configuration.zip)

TimeJammer, 2021-07-12:
Yes, I tested it too. It's giving the same error. Only 2.0.7 and earlier, Only version 2.0.7 and earlier have no problem.

TimeJammer, 2021-07-12:
> Did you test absolutely latest bugfix ?
> as this was added 5 days ago #22292

Yes, I tested it too. It's giving the same error. Only 2.0.7 and earlier, Only version 2.0.7 and earlier have no problem.

ellensp, 2021-07-12:
What exact error, screen shot? error log?

TimeJammer, 2021-07-12:
By error I mean the Printer freezes periodically and then continues. It does not encounter an error during compilation.

Only during printing, the printer stops at 250000 baud rate. But previous versions of marlin did not have this problem. I can only run 115200 with 2.0.9.1.

I use Tft 35 and original Ender 3 screen together.

TimeJammer, 2021-07-12:
> What exact error, screen shot? error log?

By error I mean the Printer freezes periodically and then continues. It does not encounter an error during compilation.

Only during printing, the printer stops at 250000 baud rate. But previous versions of marlin did not have this problem. I can only run 115200 with 2.0.9.1.

I use Tft 35 and original Ender 3 screen together.

CRCinAU, 2021-07-13:
Enable ADVANCED_OK and take note around the time of the freezes (assuming USB printing) of the B and P values, then attached them.

TimeJammer, 2021-07-14:
> Enable ADVANCED_OK and take note around the time of the freezes (assuming USB printing) of the B and P values, then attached them.

MKS TFT 35 üzerinden yazdırıyorum.Pc den yazdırmıyorum. Yinede bu işlemi yapabilirmiyim ?

TimeJammer, 2021-07-14:
> Enable ADVANCED_OK and take note around the time of the freezes (assuming USB printing) of the B and P values, then attached them.

I print via MKS TFT 35. I do not print from PC. Can I still do this?

ldursw, 2021-07-18:
Maple and STSTM32 have different implementations for serial communication. To make the stm32 HAL work like stm32f1 it would need some refactoring as mentioned here https://github.com/MarlinFirmware/Marlin/pull/21927#discussion_r633132913.

I've managed to achieve zero resends by using 57600 baud and doing the timer changes from the PR. That part was not merged so you will have to change them manually.

Another option is to go back to libmaple by changing the environment to `mks_robin_e3_maple` but it's deprecated and may be removed in a future version.

TimeJammer, 2021-07-19:
> Maple and STSTM32 have different implementations for serial communication. To make the stm32 HAL work like stm32f1 it would need some refactoring as mentioned here [#21927 (comment)](https://github.com/MarlinFirmware/Marlin/pull/21927#discussion_r633132913).
> 
> I've managed to achieve zero resends by using 57600 baud and doing the timer changes from the PR. That part was not merged so you will have to change them manually.
> 
> Another option is to go back to libmaple by changing the environment to `mks_robin_e3_maple` but it's deprecated and may be removed in a future version.

Thanks, I'll try and post the result.

TimeJammer, 2021-07-23:
> Thanks, I'll try and post the result.

Unfortunately it didn't work.

thinkyhead, 2021-07-30:
Serial stuff is pretty low level, so I'm surprised if anything in Marlin itself is directly responsible, but maybe the interrupts are running tighter and stepping on serial processing at some low level.

To narrow down the problem I suggest turning off every optional feature, and even disabling and disconnecting the screen and other extra peripherals, and see if the problem clears up. Then start adding things back one by one until the problem reappears.

TimeJammer, 2021-08-11:
> Serial stuff is pretty low level, so I'm surprised if anything in Marlin itself is directly responsible, but maybe the interrupts are running tighter and stepping on serial processing at some low [...]

Okey i will try and let you know

closed 2022-04-12 by github-actions[bot]