MarlinFirmware/Marlin #22341: [BUG] (bug summary)POWER_LOSS_RECOVERY

espadan3d, 2021-07-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

HI
I enable POWER_LOSS_RECOVERY by pin 40 but see 3 problem till now:
1-DIRECT_PIN_CONTROL was enable and I use of P4 ,but after power loss and command resume disable this pin

2-after power loss and command resume, resume print by default value heater 0 bed 0!!

3-by resume, move head to X0Y0 and return to first place this is not a good idea
It may lose place even a step I prefer to start from the current location

### Version of Marlin Firmware

bugfix-2.0.x

riff-electronics, 2021-07-27:
The problem same with me, on mks nano v3

thinkyhead, 2021-07-30:
This is a brittle feature and breaks fairly often. See if you can determine the most recent point in time where this was working, and that will help us to pinpoint the change(s) that caused it to break.

Meanwhile, there is a detailed debugging feature for Power Loss (by defining `DEBUG_POWER_LOSS_RECOVERY`), so you should use that to see whether data is being stored correctly in the "PLR" file, and you can also do simulated power outages with the debug feature in order to test it. The G-code `M413` has these special parameters:

```cpp
  #if ENABLED(DEBUG_POWER_LOSS_RECOVERY)
    if (parser.seen("RL")) recovery.load();
    if (parser.seen_test('W')) recovery.save(true);
    if (parser.seen_test('P')) recovery.purge();
    if (parser.seen_test('D')) recovery.debug(PSTR("M413"));
    #if PIN_EXISTS(POWER_LOSS)
      if (parser.seen_test('O')) recovery._outage();
    #endif
    if (parser.seen_test('E')) SERIAL_ECHOPGM_P(recovery.exists() ? PSTR("PLR Exists\n") : PSTR("No PLR\n"));
    if (parser.seen_test('V')) SERIAL_ECHOPGM_P(recovery.valid() ? PSTR("Valid\n") : PSTR("Invalid\n"));
  #endif
```

riff-electronics, 2021-07-30:
I have already enable debug power loss recovery, I have already check PLR file on flashdisk I think its already correct, I printing from flashdisk no sd card. When I turn off power, appear notification power outage on lcd and then Z axis raise up and I think its already correct too. But when I turn on power again, not appear choice for stop or resume printing, display default on idle screen

thinkyhead, 2021-07-31:
So, the next question is: What is the most recent commit in `bugfix-2.0.x` where it still works?

Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until the exact day where it broke is found.

If you started from a point 256 commits in the past, it would take no more than 8 steps to find the exact commit that broke it.

riff-electronics, 2021-08-18:
I have already try newest bug fix yesterday, and still not working.

thinkyhead, 2021-09-13:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until the exact day where it broke is found.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

Burockk, 2021-09-15:
Any way to fix this issue I am using skr mini v2?

closed 2021-11-25 by github-actions[bot]