MarlinFirmware/Marlin #22345: [BUG] No fysetc s6 boot with small bootloader

GMagician, 2021-07-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Uploading small bootloader and compiling Marlin with S6_8000 env program doesn't work. No new usb appears on PC.
Same Marlin configuration on 10000 env works (properly compiled).

Original precompiled Marlin on fysect repo works as well, even with S6_8000

Attached cnf is default with usb and board changed

### Version of Marlin Firmware

bugfix 2.0.9.1

### Host Software

Repetier Host

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6803205/Configuration.zip)

kyon03, 2021-07-13:
i thought my board was defective hoping for a fix

ellensp, 2021-07-13:
@GerogeFu can you take a look into this?

GerogeFu, 2021-07-22:
Let me try

GMagician, 2021-07-22:
Tested but it seems not to work

GMagician, 2021-07-22:
What I see, loading elf file, is that address seems wrong. Entry point is 45 20 01 08 and loading elf file with st cude it's loaded at 0x80010000

GerogeFu, 2021-07-23:
How did you test ? Different platformio env need different bootloader. If you use FYSETC_S6_8000 env,you need `Bootloader-FYSETC_S6.hex`, FYSETC_S6 env you need  `Bootloader-FYSETC_S6_10000.hex`. You can find the bootloader [here](https://github.com/FYSETC/FYSETC-S6/tree/main/bootloader).

GMagician, 2021-07-23:
I tested loading small boot with st cube and then compiled FYSETC_S6_8000 env (and also FYSETC_S6), uploaded via PlatformIO and via stcude as well

thisiskeithb, 2021-07-26:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22421

closed 2021-07-26 by thisiskeithb