MarlinFirmware/Marlin #22346: BTT GTR V1.0

martins6413, 2021-07-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

1 : K-Temp  Temperatur max error
2 : Thermal Probe Compensation

### Expected behavior

K-Temp temperature max error-wrong value is displayed, board switches off-works 2.0.7.1
Thermal Probe Compensation -works 2.0.7.1

### Version of Marlin Firmware

Marlin 2.0.9.1

### Electronics

GTR V1.0

slowbro, 2021-07-13:
Hi, can you please upload your configs? Are you using the Adafruit MAX31855 library?

martins6413, 2021-07-13:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6811049/Configuration.zip)
[pins_BTT_GTR_V1_0.zip](https://github.com/MarlinFirmware/Marlin/files/6811059/pins_BTT_GTR_V1_0.zip)
[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/6811060/platformio.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/6811061/Configuration_adv.zip)

martins6413, 2021-07-13:
Hello, I'm not a programmer, don't use Adafruit MAX31855 library, version 2.0.7.1 works fine, version 2.0.9.1 has been rewritten.

slowbro, 2021-07-13:
Thanks, I'll take a look asap.

slowbro, 2021-07-14:
Did you try on bugfix, and if so, did you make sure that the pins file was updated when you did? The THERMO_* pin names are obsoleted by more recent changes. Unfortunately I don't have a GTR (yet!) to test with here, but will try with another STM32F4 board.

There were other changes in bugfix related to this, see https://github.com/MarlinFirmware/Marlin/issues/22188 - some Software SPI fixes, etc.

martins6413, 2021-07-16:
Hallo, habe das Bugfix übernommen und auf GTR umgestellt. Nun bekomme ich den Fehler :  "Max6675_SS_PIN was not declared in this scope""  und "SENSITIVE_PINS is not declared"".

thinkyhead, 2021-07-16:
We will have to compare the code in 2.0.7.1 to see where the issue began. Can you test Marlin 2.0.8 to see if the problem also exists in that version? The more sample versions we have, the easier it will be to isolate the breaking changes. Thanks!

slowbro, 2021-07-16:
> Max6675_SS_PIN

If you are getting errors about this pin, then you likely have old files somewhere in your build directory. The bugfix code makes no reference to to this except for checking for it in Conditionals/SanityCheck. What files are you copying over to the bugfix branch when building? If you are copying over the 'pins_BTT_GTR..' file, or anything else other than Configuration, then you will likey run into this issue.

Can you please copy/paste the entire error message? It can help point us to where it is being used.

martins6413, 2021-07-16:
Habe das Bugfix übernommen und auf GTR umgestellt ohne Verenderungen. Werde es mit 2.0.8 nochmals versuchen .Marlin 2.0.7 funtioniert gut ,aber die TMC5160 machen die Motoren sehr heiss.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6833652/Marlin.zip)

slowbro, 2021-07-17:
@martins6413 the version of Marlin you are using is out of date. Many of the files are old versions. Please download the latest bugfix here, and try with that. You will need to re-configure it (Configuration.h, Configuration_adv.h):

https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.0.x.zip

martins6413, 2021-07-17:
Habe diese Version neu  auf GTR konfiguriert und erhalte nun diese Fehler :
	- "'TEMP_0_MOSI_PIN' was not declared in this scope; did you mean 'TEMP_0_MISO_PIN'?",
	- "Der Bezeichner \"\"TEMP_0_MOSI_PIN\"\" ist nicht definiert.",
	- "template argument 2 is invalid",
	- "request for member 'init' in 'max_tc_spi', which is of non-class type 'int'",
	- "request for member 'receive' in 'max_tc_spi', which is of non-class type 'int'",

slowbro, 2021-07-18:
Alright, thank you - I see the problem. Can you try this patch and let me know if it works?

```diff
diff --git a/Marlin/src/module/temperature.cpp b/Marlin/src/module/temperature.cpp
index fd2c260fd3..44da35f2c1 100644
--- a/Marlin/src/module/temperature.cpp
+++ b/Marlin/src/module/temperature.cpp
@@ -107,6 +107,27 @@
 #if (TEMP_SENSOR_0_USES_SW_SPI || TEMP_SENSOR_1_USES_SW_SPI) && !HAS_MAXTC_LIBRARIES
   #include "../libs/private_spi.h"
   #define HAS_MAXTC_SW_SPI 1
+
+  // determine the pins to use
+  #if TEMP_SENSOR_0_USES_SW_SPI
+    #define SW_SPI_SCK_PIN  TEMP_0_SCK_PIN
+    #define SW_SPI_MISO_PIN TEMP_0_MISO_PIN
+
+    #if PIN_EXISTS(TEMP_0_MOSI)
+      #define SW_SPI_MOSI_PIN TEMP_0_MOSI_PIN
+    #else
+      #define SW_SPI_MOSI_PIN SD_MOSI_PIN
+    #endif
+  #else
+    #define SW_SPI_SCK_PIN  TEMP_1_SCK_PIN
+    #define SW_SPI_MISO_PIN TEMP_1_MISO_PIN
+
+    #if PIN_EXISTS(TEMP_1_MOSI)
+      #define SW_SPI_MOSI_PIN TEMP_1_MOSI_PIN
+    #else
+      #define SW_SPI_MOSI_PIN SD_MOSI_PIN
+    #endif
+  #endif
 #endif
 
 #if ENABLED(PID_EXTRUSION_SCALING)
@@ -198,7 +219,8 @@ const char str_t_thermal_runaway[] PROGMEM = STR_T_THERMAL_RUNAWAY,
     // Initialize SoftSPI for non-lib Software SPI; Libraries take care of it themselves.
     template<uint8_t MisoPin, uint8_t MosiPin, uint8_t SckPin>
       SoftSPI<MisoPin, MosiPin, SckPin> SPIclass<MisoPin, MosiPin, SckPin>::softSPI;
-    SPIclass<TEMP_0_MISO_PIN, TEMP_0_MOSI_PIN, TEMP_0_SCK_PIN> max_tc_spi;
+    SPIclass<SW_SPI_MISO_PIN, SW_SPI_MOSI_PIN, SW_SPI_SCK_PIN> max_tc_spi;
+
   #endif
 
   #define MAXTC_INIT(n, M) \

```

martins6413, 2021-07-18:
Great, the patch works well. I started the first print and the temperature looks good.
Thanks again for the help.

[temperature.zip](https://github.com/MarlinFirmware/Marlin/files/6835909/temperature.zip)

PR #22389 (2021-07-19): Fix Software SPI when TEMP_n_MOSI_PIN is not defined
### Description

From the temperature.cpp changes in the last month or so, a bug was introduced with Software SPI. If a MAX TC is used which does not have a MOSI pin, and thus a MOSI pin is not defined, SW SPI breaks compilation. This will default back to the old way - using SD_MOSI_PIN - even if MOSI is never used.

### Requirements

MAX31855, using software SPI, with only MISO, SCK, and SS defined.

### Benefits

Fixes a bug.

### Configurations

See 22346

### Related Issues

Fixes #22346
closed 2021-07-18 by martins6413