MarlinFirmware/Marlin #22351: [FR] Pause print in between layers automatically

logiclrd, 2021-07-13:
### Describe the feature you want

Sometimes I need to adjust a print while it is running, and sometimes this adjustment requires actually swapping out the G code, and to accomplish this, I can tell Marlin "Stop Print", keeping the bed warm and the steppers on, then put a new G code file onto the SD card that starts directly at the current Z height, and tell it to print that file. But, it will always be partway into whatever layer I stop it at. I would like a way to tell Marlin to stop the current print (without shutting off heaters or motors) at the end of the current layer. Marlin could detect this by looking for a shift in Z that is below a certain threshold. As long as Z hop exceeds this threshold, it would not be detected as a false positive.

This feature would be in the LCD menu and also available via a G code command over the serial link. After this command was issued, the next time Z increased by less than a configured threshold, it would halt the print and write that next Z value to the serial link and the LCD screen's status.

This obviously would not work for people who have Z hop set to the same as or less than their layer height, but I suspect Z hop is usually higher than the layer height to ensure that the nozzle never leaves scars on the printed surface. This would also be a trickier feature to use with variable layer heights, but that's a v2 problem :-P

Another option would be to introduce a new G code command to signal the start of a layer, and provide an option for this new command to wait until it sees that specific command.

Proposed syntax (exact command numbers subject to being properly assigned):
```
G123 Z3.6 D0.2
```
Stop the print when Z is increased to >= 3.6 in a step no larger than 0.2 (i.e., starting from >= 3.4). A print that is at Z 3.2 with Z hop set to 0.4 would set Z to 3.6 without triggering this function because of D0.2.
```
G123 I46
```
Stop the print when the G code indicates it is beginning layer 46.
```
G124 I46
```
Indicate that layer 46 is starting.

logiclrd, 2021-07-14:
Additional proposal: `G123 D0.2` without specifying a Z would stop the print the next time Z was adjusted by 0.2 or less, regardless of the absolute value of Z. Z hop adjustments, if configured to more than 0.2mm, would not trigger this.

thinkyhead, 2021-07-30:
Interesting concept. You may try hiring someone and paying them to implement this for you if it is needed urgently. We have a channel on the Marlin Firmware Discord where you can post a job offer or just ask around, and maybe one of the developers in the channel will be keen to build this for you.