MarlinFirmware/Marlin #22352: [BUG] Probe Z Offset only one direction (only Z raise)

riff-electronics, 2021-07-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I tring adjust porbe Z offset up and down via LCD reprap discount, stepper only doing up. I turn left rotary encoder Z is up, and I turn right rotary encoder then the move direction is the same before.

### Bug Timeline

new

### Expected behavior

Probe Z offset can doing up and down direction

### Actual behavior

Probe Z offset only one direction move

### Version of Marlin Firmware

Marlin 2.0.9

### Printer model

Custom CoreXY

### Electronics

MKS Robin Nano V3

### Add-ons

Reprap DIscount LCD

### Your Slicer

Cura

### Host Software

SD Card (headless)

riff-electronics, 2021-07-23:
What needs to calibrated?

thinkyhead, 2021-07-30:
This issue is related to a calibration feature, capisce?

The problem you are seeing is related to `BABYSTEPPING` and the `DIR` signal. You can try enabling and adjusting the configuration options `MINIMUM_STEPPER_POST_DIR_DELAY` and `MINIMUM_STEPPER_PRE_DIR_DELAY`
and you could also try turning on `INTEGRATED_BABYSTEPPING` to see if that method behaves better.

riff-electronics, 2021-07-30:
> This issue is related to a calibration feature, capisce?
>
> The problem you are seeing is related to `BABYSTEPPING` and the `DIR` signal. You can try enabling and adjusting the configuration options [...]

Thanks, its working now, I enable MINIMUM_STEPPER_POS_DELAY, MINIMUM_STEPPER_PRE_DIR_DELAY  with value delay  for tmc2209. And I enable too INTEGRATED_BABYSTEPPING

thinkyhead, 2021-07-31:
Good news! I see that the `MINIMUM_STEPPER_(PRE|POST)_DIR_DELAY` are set automatically in `Conditionals_adv.h` if they are not defined, so also make sure your `*_DRIVER_TYPE` settings are all good.

closed 2021-07-31 by thinkyhead