MarlinFirmware/Marlin #22353: [BUG] "power.cpp" fails to compile if board does not have PS_ON_PIN

marciot, 2021-07-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The "power.cpp" file generates compilation errors if PS_ON_PIN and PSU_POWERUP_DELAY if the board does not support the feature.

It seems like the "power.cpp" file needs a guard condition on whether that feature is available.

### Version of Marlin Firmware

bugfix-2.0.x branch

### Printer model

Unreleased Prototype Printer

### Electronics

Ultimachine Archim 2 board

### Additional information & file uploads

Configuration files:

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6811011/configuration.zip)

Compilation errors:

[compilation_errors.txt](https://github.com/MarlinFirmware/Marlin/files/6811036/compilation_errors.txt)

slowbro, 2021-07-14:
I'm not able to reproduce this with your configs (in fact, the configs fail compilation for a number of other reasons). I tried to replicate it in other ways, i.e. enabling PSU_CONTROL with PS_ON_PIN unset, or set to -1, and it was properly caught with a `error: #error "PSU_CONTROL requires PS_ON_PIN."`. Make sure that your platformio.ini & ini/features files are up to date, as the most recent version excludes the power.cpp file unless PSU_CONRTOL is enabled.

marciot, 2021-07-14:
@slobro: Thank you for this information, you actually explained the situation and gave me the hint I needed on how to fix it.

I am using the Makefile for compilation, not platform.io, so every file gets compiled. This was the way Marlin used to work prior to the switch to platform.io. Because few people use the Makefile anymore, the proper #ifdef guards often aren't updated.

I think I am the only one who uses or tests that, so I'll create a PR to take care of the problem.

slowbro, 2021-07-14:
Ah, that makes sense. Glad I could point you in the right direction!

slobro, 2021-07-14:
You're welcome.

thisiskeithb, 2021-07-15:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22361

closed 2021-07-15 by thisiskeithb