MarlinFirmware/Marlin #22357: MKS NANO V3 Latest Bugfix Fails to compile[BUG] (bug summary)

Savage79, 2021-07-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Attempting to compile latest Bugfix ( July-13-2021) w/current config,  identical settings to Bugfix from July-06-2021.

[image]
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6813281/Configuration.zip)

### Bug Timeline

New Bugfix 2.0.9.x (July 13-2021)

### Expected behavior

Expected to compile as normal

### Actual behavior

Error during compile.

### Steps to Reproduce

Copy settings from working config files using current config files.
Compile

### Version of Marlin Firmware

Marlin Bugfix 2.0.X

### Printer model

Flsun QQS Pro

### Electronics

MKS Nano V3 w/TS35 V2 and TMC2209 Uart

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

slowbro, 2021-07-14:
This is fixed by https://github.com/MarlinFirmware/Marlin/pull/22354 (pending merge)

thisiskeithb, 2021-07-14:
> This is fixed by #22354

Merged. 

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

closed 2021-07-14 by thisiskeithb