MarlinFirmware/Marlin #22360: [BUG] MarlinUI - Change Filament menu doesn't show preheat options correctly

The-EG, 2021-07-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

If 'Change Filament' is selected while the hotend is cold, the preheat menu is shown but items can't be selected because the 'Change Filament' alert screen is repeated. The only way to exit is to select 'No'. The only way to initiate a filament change from the LCD is to preheat the hotend separately.

On a `CR10_STOCKDISPLAY`:
<details>
<summary>video</summary>

[image]

</details>

On the Ender 3 v2's DWIN (#22211, note the preheat items are visible):
<details>
<summary>video</summary>

[image]

</details>

### Bug Timeline

Not sure, found while testing #22211

### Expected behavior

Selecting 'Change Filament' while the hotend is not heated should show the preheat items and allow for selection.

### Actual behavior

Selecting a preheat item is not possible.

### Steps to Reproduce

With `ADVANCED_PAUSE_FEATURE` enabled:

1. Start with nozzle temperature off
2. Select 'Change Filament' from LCD menu

### Version of Marlin Firmware

2.0.9.1 and bugfix-2.0.x

### Printer model

Ender 3 Pro & Ender 3 V2

### Additional information & file uploads

Config:
[Ender3v2.zip](https://github.com/MarlinFirmware/Marlin/files/6816031/Ender3v2.zip)

thisiskeithb, 2021-07-14:
Interestingly, this works fine with `REPRAP_DISCOUNT_SMART_CONTROLLER` enabled.

The-EG, 2021-07-14:
> Interestingly, this works fine with `REPRAP_DISCOUNT_SMART_CONTROLLER` enabled.

A somewhat random question: do you happen to have `FILAMENT_LOAD_UNLOAD_GCODES` enabled?

thisiskeithb, 2021-07-14:
> A somewhat random question: do you happen to have FILAMENT_LOAD_UNLOAD_GCODES enabled?

I sure do! [Here's a link to the full Marlin build](https://github.com/thisiskeithb/Marlin/tree/myconfig/MK3S-BTT002-TMC2209).

The-EG, 2021-07-14:
Confirmed: with `FILAMENT_LOAD_UNLOAD_GCODES` enabled this issue doesn't occur.

With that, 'Change Filament' becomes a submenu, and the 'Change Filament' item in that menu functions correctly.

So that's something.

thisiskeithb, 2021-07-15:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22370

PR #22370 (2021-07-15): Use goto_screen for Change Filament preheat menu
### Description

The Change Filament menu item shows a preheat menu when the nozzle is not heated. When `FILAMENT_LOAD_UNLOAD_GCODES` is **not** enabled, the preheat menu isn't properly shown. See #22360

This fixes that issue.

### Requirements

Any MarlinUI LCD with `ADVANCED_PAUSE_FEATURE` enabled and `FILAMENT_LOAD_UNLOAD_GCODES` **not** enabled.

### Benefits

Fixes #22360

### Configurations

[Ender 3 pro v427.zip](https://github.com/MarlinFirmware/Marlin/files/6824152/Ender.3.pro.v427.zip)

### Related Issues

#22360

### Additional Notes [... cut]
closed 2021-07-15 by thisiskeithb