MarlinFirmware/Marlin #22362: [BUG] Marlin/src/MarlinCore.cpp:529:20: error: redeclaration of 'const millis_t ms'

hamster65, 2021-07-14:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin fails to compille when CUSTOM_USER_BUTTONS is defined.
Not a problem since I do not use custom user buttons.

Building in release mode
Compiling .pio/build/LPC1768/src/src/MarlinCore.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/bedlevel/ubl/ubl_motion.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/bltouch.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/controllerfan.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/e_parser.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/host_actions.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/hotend_idle.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/pause.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/runout.cpp.o
Marlin/src/MarlinCore.cpp: In function 'void manage_inactivity(bool)':
Marlin/src/MarlinCore.cpp:529:20: error: redeclaration of 'const millis_t ms'
  529 |     const millis_t ms = millis();
      |                    ^~
Marlin/src/MarlinCore.cpp:435:18: note: 'const millis_t ms' previously declared here
  435 |   const millis_t ms = millis();
      |                  ^~
Marlin/src/MarlinCore.cpp:528:16: warning: unused variable 'printer_not_busy' [-Wunused-variable]
  528 |     const bool printer_not_busy = !printingIsActive();
      |                ^~~~~~~~~~~~~~~~
*** [.pio/build/LPC1768/src/src/MarlinCore.cpp.o] Error 1
========================================================================================= [FAILED] Took 3.61 seconds ===============================================================================

### Bug Timeline

new, did compile a couple of days ago.

### Version of Marlin Firmware

bugfix

slowbro, 2021-07-14:
Hmm, looks like there are multiple areas where `const millis_t ms;` is defined, just behind different guard conditions. I think the single declaration of it at the top is enough. I'll collapse them down to just the one.

thisiskeithb, 2021-07-15:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22363

Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

closed 2021-07-15 by thisiskeithb