MarlinFirmware/Marlin #22374: [BUG] Setting temperature pin without analog capabilities crashes

G-Pereira, 2021-07-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When defining a new board or editing an existing one and a temperature pin is set which doesn't have ADC capabilities makes the board not respond to any serial commands after flashing.

### Bug Timeline

yesterday

### Expected behavior

The sanity checks could be able to prevent this

### Actual behavior

The build process succeded

### Steps to Reproduce

1. Go to any board pins file
2. set any temperature pin to a not capable ADC pin
3. build and flash
4. try to talk with the board via serial

### Version of Marlin Firmware

2.0.9

### Printer model

Index Machines Index PnP

### Electronics

Index Machines Motherboard REV03

slowbro, 2021-07-18:
When I do this, I am getting an immediate MAXTEMP error on boot, which is kind of legit:

```
Error:MAXTEMP triggered, system stopped! Heater_ID: E0

echo:Err: MAXTEMP
Error:Printer halted. kill() called!
```

It happens really early though, so it's hard to tell after the fact what caused the printer to halt. Are you seeing something different and/or is MAXTEMP not on the LCD (if installed)?

It would probably be wise to make sure that pins that are expected to be ADC temperature probes are actually on ADC pins, I agree.

G-Pereira, 2021-07-19:
In this case, I don't have an LCD installed and I don't have any machine at the moment with one that I can test with.

slowbro, 2021-07-19:
If you could test something: set `#define BOGUS_TEMPERATURE_GRACE_PERIOD 10000` in your config, recompile & and see if it dies with a MAXTEMP error after 10 seconds. At least that way we can see if Marlin is actually booting or not.

G-Pereira, 2021-07-20:
I can say that just changing the pin assignment made the board work as intended.

slowbro, 2021-07-20:
Alright. I opened #22409 regarding the feature request for ADC sanity checking as I think that is a great idea. I'll close this ticket for now as I'm not able to reproduce anything that I would consider a bug.

closed 2021-07-20 by slowbro