MarlinFirmware/Marlin #22375: [BUG] SKR Mini E3 V2.0 FAST_PWM_FAN ignores FAST_PWM_FAN_FREQUENCY (f=1kHz)

blazewicz, 2021-07-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Part cooling fan emits loud 1 kHz noise, investigation shown that PWM frequency is exactly 1 kHz. This doesn't change with `FAST_PWM_FAN_FREQUENCY`.

[image]

Board uses pin `PC6` for part cooling fan. This pin is channel 1 of timer `TIM3`. This board also has second PWM controlled fan output which is on pin `PC7` and its channel 2 of timer `TIM3`. (Arduino framework configures it to `TIM3`).

[image]

Probably `TIM3` is used by some ISR which is set to 1 kHz before PWM is initialized.

### Expected behavior

PWM frequency is configurable with `FAST_PWM_FAN_FREQUENCY`.

### Actual behavior

PWM frequency doesn't change and is always 1 kHz.

### Steps to Reproduce

Build with example config for Ender 3 with BTT SKR Mini E3 V2.0, set  `FAST_PWM_FAN` and `FAST_PWM_FAN_FREQUENCY` settings.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

doesn't matter

### Electronics

BTT SKR Mini E3 V2.0

blazewicz, 2021-07-16:
adding

```diff
--- a/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.h
+++ b/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/variant.h
@@ -131,6 +131,8 @@ extern "C" {
 #define HAL_DAC_MODULE_ENABLED
 #endif

+#define PWM_FREQUENCY 31000
+
 #ifdef __cplusplus
 } // extern "C"
 #endif
```

does change fan PWM frequency

thisiskeithb, 2021-07-17:
Please attach your configuration files.

Also, what is the exact MCU on the board and which PIO environment are you compiling with?

blazewicz, 2021-07-17:
Its GD32F103RET6 (STM32F103RET6 clone). I use `STM32F103RE_btt_USB`, also tested `STM32F103RC_btt_USB`, same behavior.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6834371/Configuration.zip)

thinkyhead, 2021-07-30:
Four different boards use `MARLIN_F103Rx` so we cannot make this change unless the other boards would not be affected. If only certain boards need the define, it should be okay to add it to those boards' environments with `-DPWM_FREQUENCY=31000` on the build flags.

blazewicz, 2021-11-25:
Maybe fixed in #23125, needs verification.

thisiskeithb, 2022-01-05:
Closing since the GD-based SKR Mini cannot be added here.

closed 2022-01-05 by thisiskeithb