MarlinFirmware/Marlin #22376: [LỖI] (tóm tắt lỗi)Unable to install LCD #define REPRAP_DISCOUNT_SMART_CONTROLLER Or #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Haiang, 2021-07-17:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Currently I can't install LCD even though I'm using Mega 2560 and Ramps 1.4
#define REPRAP_DISCOUNT_SMART_CONTROLLER
Or
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
If I do LCD REPRAP_DISCOUNT_SMART_CONTROLLER setting it gives error like this

Arduino: 1.8.7 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Print the file included from sketch\ultralcd.cpp:100:0:

ultralcd_impl_HD44780.h:196:17: error: 'LCD_PINS_RS' was not declared in this scope

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                 ^~~~~~~~~~

sketch\ultralcd_impl_HD44780.h:196:17: note: suggested alternative: 'LCD_PINS_D4'

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                 ^~~~~~~~~~

                 LCD_PINS_D4

ultralcd_impl_HD44780.h:196:30: error: 'LCD_PINS_ENABLE' was not declared in this scope

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                              ^~~~~~~~~~~~~~~~

sketch\ultralcd_impl_HD44780.h:196:30: note: suggested alternative: 'LCD_PINS_D4'

   LCD_CLASS lcd(LCD_PINS_RS, LCD_PINS_ENABLE, LCD_PINS_D4, LCD_PINS_D5, LCD_PINS_D6, LCD_PINS_D7); //RS,Enable,D4,D5,D6,D7

                              ^~~~~~~~~~~~~~~~

                              LCD_PINS_D4

Multiple libraries were found for "LiquidCrystal.h"
 Used: C:\Users\Administrator\Documents\Arduino\libraries\LiquidCrystal
 Not used: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal
exit status 1
'LCD_PINS_RS' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

### Bug Timeline

New

### Expected behavior

I don't know how to fix it

### Actual behavior

Looking forward to helping fix the error

### Version of Marlin Firmware

Marlin 1.1.9

### Printer model

Prusia

### Electronics

Cartesian

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

ellensp, 2021-07-17:
Please attach your configuration files.
This tells us what controller you have, so we can check it has pin definitions for that LCD
YOu also need to tell us the type of CD this is, REPRAP_DISCOUNT_SMART_CONTROLLER and REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER are very different lcd's

Haiang, 2021-07-17:
> Vui lòng đính kèm tệp cấu hình của bạn.
> Điều này cho chúng tôi biết bạn có bộ điều
> khiển nào, vì vậy chúng tôi có thể kiểm tra nó có định nghĩa pin cho màn hình LCD đó Bạn cũng cần cho chúng tôi biết loại LCD này là, REPRAP_DISCOUNT_SMART_CONTROLLER và REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER là lcd rất khác nhau

Haiang, 2021-07-17:
[](url)

ellensp, 2021-07-17:
just noticed "Marlin 1.1.9" sorry that marlin is no longer supported.

closed 2021-07-17 by ellensp

Haiang, 2021-07-17:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6834416/Marlin.zip)

ellensp, 2021-07-17:
this is for a Arduino Mega with ResinShield 
you would need to add the pins to pins_RESIN.h from the circuit diagram https://github.com/alch3my/printy/blob/master/Electrical/ResinShield%20sch.pdf

Haiang, 2021-07-17:
> điều này dành cho Arduino
> Mega với ResinShield, bạn sẽ cần thêm các chân vào pins_RESIN.h từ sơ [đồ mạch https://github.com/alch3my/printy/blob/master/Electrical/ResinShield%20sch.pdf](https://github.com/alch3my/printy/blob/master/Electrical/ResinShield%20sch.pdf)

I am having a hard time adding pins can you help me add pins for
REPRAP_DISCOUNT_SMART_CONTROLLER
OK

ellensp, 2021-07-17:
Important

Both LCD types require 5v on pin 10 of J6, But this is connected to D34  on your board.

You cannot just plug this in. You need to make a custom cable to get 5v for somewhere else.

Your board
```CPP
/*	               J6		               J9

 *    (very bad) D34 |10 9 | GND                  NC |10 9 | GND
 *               D35 | 8 7 | D36                 D47 | 8 7 | D44 
 *               D37 | 6 5 | D38          (MOSI) D51   6 5   D46
 *               D39 | 4 3 | D43            (SS) D53 | 4 3 | D49
 *               D41 | 2 1 | D45           (SCK) D52 | 2 1 | D50 (MISO)
 */

// Very bad Pin 10 on J6 needs to be 5v. cannot use IO pin for power. 
 
  #define EXP1_03_PIN                         35
  #define EXP1_04_PIN                         36
  #define EXP1_05_PIN                         37
  #define EXP1_06_PIN                         38
  #define EXP1_07_PIN                         39
  #define EXP1_08_PIN                         43
  #define EXP1_09_PIN                         41
  #define EXP1_10_PIN                         45

  #define EXP2_03_PIN                         47
  #define EXP2_04_PIN                         44
  #define EXP2_05_PIN                         51
  #define EXP2_06_PIN                         46
  #define EXP2_07_PIN                         53
  #define EXP2_08_PIN                         49
  #define EXP2_09_PIN                         52
  #define EXP2_10_PIN                         50
  
// for REPRAP_DISCOUNT_SMART_CONTROLLER

  #define LCD_PINS_RS          EXP1_07_PIN
  #define LCD_PINS_ENABLE      EXP1_08_PIN
  #define LCD_PINS_D4          EXP1_06_PIN
  #define LCD_PINS_D5          EXP1_05_PIN
  #define LCD_PINS_D6          EXP1_04_PIN
  #define LCD_PINS_D7          EXP1_03_PIN
  #define BEEPER_PIN           EXP1_10_PIN
  #define BTN_EN1              EXP2_08_PIN
  #define BTN_EN2              EXP2_06_PIN
  #define BTN_ENC              EXP1_09_PIN
  #define SD_DETECT_PIN        EXP2_04_PIN
  #define KILL_PIN             EXP2_03_PIN

```

ellensp, 2021-07-17:
Important

Both LCD types require 5v on pin 1 of J6, But this is connected to D34  on your board.

You cannot just plug this in. You need to make a custom cable to get 5v from somewhere else.

Your board
```CPP
/*	               J6		               J9

 *    (very bad) D34 | 1 2 | GND                  NC | 1 2 | GND
 *               D35 | 3 4 | D36                 D47 | 3 4 | D44 
 *               D37 | 5 6 | D38          (MOSI) D51 | 5 6 | D46
 *               D39 | 7 8 | D43            (SS) D53 | 7 8 | D49
 *               D41 | 9 10| D45           (SCK) D52 | 9 10| D50 (MISO)
 */

// Very bad Pin 1 on J6 needs to be 5v. cannot use IO pin for power. 
 
  #define EXP1_03_PIN                         35
  #define EXP1_04_PIN                         36
  #define EXP1_05_PIN                         37
  #define EXP1_06_PIN                         38
  #define EXP1_07_PIN                         39
  #define EXP1_08_PIN                         43
  #define EXP1_09_PIN                         41
  #define EXP1_10_PIN                         45

  #define EXP2_03_PIN                         47
  #define EXP2_04_PIN                         44
  #define EXP2_05_PIN                         51
  #define EXP2_06_PIN                         46
  #define EXP2_07_PIN                         53
  #define EXP2_08_PIN                         49
  #define EXP2_09_PIN                         52
  #define EXP2_10_PIN                         50
  
// for REPRAP_DISCOUNT_SMART_CONTROLLER

  #define LCD_PINS_RS          EXP1_07_PIN
  #define LCD_PINS_ENABLE      EXP1_08_PIN
  #define LCD_PINS_D4          EXP1_06_PIN
  #define LCD_PINS_D5          EXP1_05_PIN
  #define LCD_PINS_D6          EXP1_04_PIN
  #define LCD_PINS_D7          EXP1_03_PIN
  #define BEEPER_PIN           EXP1_10_PIN
  #define BTN_EN1              EXP2_08_PIN
  #define BTN_EN2              EXP2_06_PIN
  #define BTN_ENC              EXP1_09_PIN
  #define SD_DETECT_PIN        EXP2_04_PIN
  #define KILL_PIN             EXP2_03_PIN

```

Haiang, 2021-07-17:
> Important
> 
> Both LCD types require 5v on pin 1 of J6, But this is connected to D34 on your board.
> 
> You cannot just plug this in. You need to make a custom cable to get 5v from somewhere else.
> 
> Your board
> 
> ```c++
> /*	               J6		               J9
> 
>  *    (very bad) D34 | 1 2 | GND                  NC | 1 2 | GND
>  *               D35 | 3 4 | D36                 D47 | 3 4 | D44 
>  *               D37 | 5 6 | D38          (MOSI) D51 | 5 6 | D46
>  *               D39 | 7 8 | D43            (SS) D53 | 7 8 | D49
>  *               D41 | 9 10| D45           (SCK) D52 | 9 10| D50 (MISO)
>  */
> 
> // Very bad Pin 1 on J6 needs to be 5v. cannot use IO pin for power. 
>  
>   #define EXP1_03_PIN                         35
>   #define EXP1_04_PIN                         36
>   #define EXP1_05_PIN                         37
>   #define EXP1_06_PIN                         38
>   #define EXP1_07_PIN                         39
>   #define EXP1_08_PIN                         43
>   #define EXP1_09_PIN                         41
>   #define EXP1_10_PIN                         45
> 
>   #define EXP2_03_PIN                         47
>   #define EXP2_04_PIN                         44
>   #define EXP2_05_PIN                         51
>   #define EXP2_06_PIN                         46
>   #define EXP2_07_PIN                         53
>   #define EXP2_08_PIN                         49
>   #define EXP2_09_PIN                         52
>   #define EXP2_10_PIN                         50
>   
> // for REPRAP_DISCOUNT_SMART_CONTROLLER
> 
>   #define LCD_PINS_RS          EXP1_07_PIN
>   #define LCD_PINS_ENABLE      EXP1_08_PIN
>   #define LCD_PINS_D4          EXP1_06_PIN
>   #define LCD_PINS_D5          EXP1_05_PIN
>   #define LCD_PINS_D6          EXP1_04_PIN
>   #define LCD_PINS_D7          EXP1_03_PIN
>   #define BEEPER_PIN           EXP1_10_PIN
>   #define BTN_EN1              EXP2_08_PIN
>   #define BTN_EN2              EXP2_06_PIN
>   #define BTN_ENC              EXP1_09_PIN
>   #define SD_DETECT_PIN        EXP2_04_PIN
>   #define KILL_PIN             EXP2_03_PIN
> ```

Thank you so much

Haiang, 2021-07-17:
> > quan trọng
> > Cả hai loại LCD đều yêu cầu 5v trên chân 1 của J6, Nhưng điều này được kết nối với D34 trên bảng của bạn.
> > Bạn không thể chỉ cắm cái này vào. Bạn cần tạo một cáp tùy chỉnh để có được 5v từ một nơi khác.
> > Bảng của bạn
> > ```c++
> > /*	               J6		               J9
> > 
> >  *    (very bad) D34 | 1 2 | GND                  NC | 1 2 | GND
> >  *               D35 | 3 4 | D36                 D47 | 3 4 | D44 
> >  *               D37 | 5 6 | D38          (MOSI) D51 | 5 6 | D46
> >  *               D39 | 7 8 | D43            (SS) D53 | 7 8 | D49
> >  *               D41 | 9 10| D45           (SCK) D52 | 9 10| D50 (MISO)
> >  */
> > 
> > // Very bad Pin 1 on J6 needs to be 5v. cannot use IO pin for power. 
> >  
> >   #define EXP1_03_PIN                         35
> >   #define EXP1_04_PIN                         36
> >   #define EXP1_05_PIN                         37
> >   #define EXP1_06_PIN                         38
> >   #define EXP1_07_PIN                         39
> >   #define EXP1_08_PIN                         43
> >   #define EXP1_09_PIN                         41
> >   #define EXP1_10_PIN                         45
> > 
> >   #define EXP2_03_PIN                         47
> >   #define EXP2_04_PIN                         44
> >   #define EXP2_05_PIN                         51
> >   #define EXP2_06_PIN                         46
> >   #define EXP2_07_PIN                         53
> >   #define EXP2_08_PIN                         49
> >   #define EXP2_09_PIN                         52
> >   #define EXP2_10_PIN                         50
> >   
> > // for REPRAP_DISCOUNT_SMART_CONTROLLER
> > 
> >   #define LCD_PINS_RS          EXP1_07_PIN
> >   #define LCD_PINS_ENABLE      EXP1_08_PIN
> >   #define LCD_PINS_D4          EXP1_06_PIN
> >   #define LCD_PINS_D5          EXP1_05_PIN
> >   #define LCD_PINS_D6          EXP1_04_PIN
> >   #define LCD_PINS_D7          EXP1_03_PIN
> >   #define BEEPER_PIN           EXP1_10_PIN
> >   #define BTN_EN1              EXP2_08_PIN
> >   #define BTN_EN2              EXP2_06_PIN
> >   #define BTN_ENC              EXP1_09_PIN
> >   #define SD_DETECT_PIN        EXP2_04_PIN
> >   #define KILL_PIN             EXP2_03_PIN
> > ```
> 
> Cảm ơn rất nhiều

Hi do you have gmail can I discuss some more issues with you via Gmail?
Can you give me your Gmail address

ellensp, 2021-07-17:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Haiang, 2021-07-17:
> Hàng đợi vấn đề này dành cho báo cáo lỗi Marlin và các vấn đề liên quan đến phát triển và chúng tôi không muốn xử lý các câu hỏi hỗ trợ người dùng ở đây. (Như đã lưu ý [trên trang này](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) Để có kết quả tốt nhất khi nhận trợ giúp về cấu hình và khắc phục sự cố, vui lòng sử dụng các tài nguyên sau:
> 
> * MarlinFW.org tài [liệu Marlin](http://marlinfw.org)
> * RepRap.org diễn [đàn Marlin](http://forums.reprap.org/list.php?415)
> * Nhóm Facebook ["Phần mềm Marlin"](https://www.facebook.com/groups/1049718498464482/)
> * Tập đoàn Facebook ["Phần mềm Marlin cho máy in 3D"](https://www.facebook.com/groups/3Dtechtalk/)
> * [Cấu hình Marlin](https://www.youtube.com/results?search_query=marlin+configuration) trên YouTube
> * Máy chủ Marlin Discord. Liên kết tham gia: https://discord.gg/n5NJ59y
> 
> Sau khi tìm kiếm sự giúp đỡ từ cộng đồng, nếu sự đồng thuận chỉ ra một lỗi ở Marlin, thì bạn nên đăng một [báo cáo lỗi](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Yes