MarlinFirmware/Marlin #22379: [BUG] DOGLCD - Fonts for additional languages does not load

MESlammer, 2021-07-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

You can specify additional languages for Marlin UI, with `LCD_LANGUAGE_2, LCD_LANGUAGE_3, LCD_LANGUAGE_4, LCD_LANGUAGE_5`. However, fonts for these languages are not uploaded to controller, so if you will change language in menu, text on display will be broken, unless it uses font of `LCD_LANGUAGE`
Bug was found on DOGLCD displays, i have no hardware to test it on TFT_COLOR_UI or TOUCH_UI_FTDI_EVE.

### Bug Timeline

First found in 2.0.9.1

### Expected behavior

Readable text on display in different languages

### Actual behavior

Random symbols instead of text

### Steps to Reproduce

1. Enable graphical controller in configuration.h (ULTI_CONTROLLER, for example)
2. Enable additional languages in configuration_adv.h - uncomment LCD_LANGUAGE_2 and set it to language with not default font - ru, for example
3. Repeat same for LCD_LANGUAGE_3, LCD_LANGUAGE_4, LCD_LANGUAGE_5 - set them to uk, jp_kana and ko_KR
4. Compile, upload
5. Change language through menu

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

BLV mgn Cube (unrelevant)

### Electronics

Arduino DUE, RuRAMPS v.1.3, SSD1309 DIY controller 

### Additional information & file uploads

[configurations.zip](https://github.com/MarlinFirmware/Marlin/files/6835155/configurations.zip)

thinkyhead, 2021-07-17:
Thanks for the report! Please test #22381 to see if it fixes the issue.

MESlammer, 2021-07-17:
@thinkyhead Sure!
#22381 - loads font only for the last language in list. For understanding, if there 5 choices - it will load font only for LCD_LANGUAGE_5, if there is 3 choices - LCD_LANGUAGE_3. Fonts in between do not load.

thinkyhead, 2021-07-17:
Since most of the language fonts are empty I assumed that the `langdata_xx.h` files are simply providing the additional characters needed to augment the standard Western font, and not a complete font. I had hoped they could all be installed at the start, but it appears that the call to `uxg_SetUtf8Fonts` does a clear of the font data before installing the additional characters. So, I will simply adjust `set_language` to call `uxg_SetUtf8Fonts` on the fly.

thinkyhead, 2021-07-17:
Ok, I have made the change, so theoretically it should now work!

MESlammer, 2021-07-18:
Okay, looks like it's working now. All fonts load correctly.

thinkyhead, 2021-07-19:
Thanks, it works here too, and I have also changed the `LANGUAGE` string for `zh_CN` to English so it can always be drawn.

PR #22381 (2021-07-19): Load all characters for multi-language
Fixes #22379

Currently the multi-language fonts for DOGLCD displays are not loaded when needed. This PR adds code to include all required fontdata and to re-initialize the font data whenever the language is changed.
closed 2021-07-19 by thinkyhead