MarlinFirmware/Marlin #22386: [BUG] Failing to build marlin on SKR_PRO_1.2

Alchoni, 2021-07-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

At first I need to admit that I'm fresh in vscode,  C++ and 3d-printing, with that being said I have a trouble when building Marlin. After reading 20 github threads and other sources I just gave up, I think I saw solution for similar error but I lost it :c

**ERROR:**
--------------------------------------------------------------------------------------------------------------
```
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfigPre.h:38,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\szymo\downloads\marlin-2.0.x\marlin-2.0.x\marlin\src\core\boards.h:430:64: error: operator '==' has no right operand
  430 | #define _MB_1(B)  (defined(BOARD_##B) && MOTHERBOARD==BOARD_##B)
      |                                                                ^
c:\users\szymo\downloads\marlin-2.0.x\marlin-2.0.x\marlin\src\core\macros.h:164:29: note: in expansion of macro '_MB_1'
  164 | #define _DO_1(W,C,A)       (_##W##_1(A))
      |                             ^
c:\users\szymo\downloads\marlin-2.0.x\marlin-2.0.x\marlin\src\core\macros.h:179:28: note: in expansion of macro '_DO_1'
  179 | #define __DO_N(W,C,N,V...) _DO_##N(W,C,V)
      |                            ^~~~
c:\users\szymo\downloads\marlin-2.0.x\marlin-2.0.x\marlin\src\core\macros.h:180:28: note: in expansion of macro '__DO_N'
  180 | #define _DO_N(W,C,N,V...)  __DO_N(W,C,N,V)
      |                            ^~~~~~
c:\users\szymo\downloads\marlin-2.0.x\marlin-2.0.x\marlin\src\core\macros.h:181:29: note: in expansion of macro '_DO_N'
  181 | #define DO(W,C,V...)       (_DO_N(W,C,NUM_ARGS(V),V))
      |                             ^~~~~
c:\users\szymo\downloads\marlin-2.0.x\marlin-2.0.x\marlin\src\core\boards.h:431:19: note: in expansion of macro 'DO'
  431 | #define MB(V...)  DO(MB,||,V)
      |                   ^~
c:\users\szymo\downloads\marlin-2.0.x\marlin-2.0.x\marlin\src\pins\pins.h:588:7: note: in expansion of macro 'MB'
  588 | #elif MB(BTT_SKR_PRO_V1_2)
      |       ^~
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\szymo\downloads\marlin-2.0.x\marlin-2.0.x\marlin\src\pins\pins.h:763:6: error: #error "Unknown MOTHERBOARD value set in Configuration.h"  
  763 |     #error "Unknown MOTHERBOARD value set in Configuration.h"
      |      ^~~~~
CalledProcessError: Command '"C:\Users\szymo\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-g++.exe" -D__MARLIN_FIRMWARE__ -DNDEBUG -DUSBCON -DUSBD_USE_CDC -DTIM_IRQ_PRIO=13 -DADC_RESOLUTION=12 -DSTM32F407_5ZX -DVECT_TAB_OFFSET=0x8000 -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "C:\Users\szymo\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 175:
    env.SConscript(item, exports="env")
  File "C:\Users\szymo\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\szymo\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\szymo\Downloads\Marlin-2.0.x\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 309:
    apply_features_config()
  File "C:\Users\szymo\Downloads\Marlin-2.0.x\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 136:
    if not env.MarlinFeatureIsEnabled(feature):
  File "C:\Users\szymo\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "C:\Users\szymo\Downloads\Marlin-2.0.x\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 285:
    load_marlin_features()
  File "C:\Users\szymo\Downloads\Marlin-2.0.x\Marlin-2.0.x\buildroot\share\PlatformIO\scripts\common-dependencies.py", line 273:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "C:\Users\szymo\.platformio\python3\lib\subprocess.py", line 424:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\szymo\.platformio\python3\lib\subprocess.py", line 528:
    raise CalledProcessError(retcode, process.args,
=========================================================== [FAILED] Took 12.97 seconds ===========================================================
Environment      Status    Duration
---------------  --------  ------------
BIGTREE_SKR_PRO  FAILED    00:00:12.974
====================================================== 1 failed, 0 succeeded in 00:00:12.974 ======================================================The
```
--------------------------------------------------------------------------------------------------------------

I supose it's the cascade error but anyway I decided to post everything. I think it's worth mentioning that Im using SKR_PRO_1.2 and that Im not using any bought 3d printer. Also, I have already tried to updating verbose, platformIO and few other things but maybe I have done something wrong so if it may help, I will be happy to receive feedback on what to update/change. I dunno if it does have any matter but ask me anything and I will respond quickly, thanks :)

### Bug Timeline

15/07/2021

### Expected behavior

It was compiling a ~week ago but I think I broke something :c

### Version of Marlin Firmware

2.0.9.1

### Printer model

Custom design 

### Electronics

SKR-PRO-V1.2

### Add-ons

TMC2209-V1.2

### Host Software

SD Card (headless)

ellensp, 2021-07-18:
" #error "Unknown MOTHERBOARD value set in Configuration.h""

You have broken the Configuration.h

ellensp, 2021-07-18:
It should say 
#define MOTHERBOARD BOARD_BTT_SKR_PRO_V1_2

closed 2021-07-18 by ellensp

Alchoni, 2021-07-18:
Welp, that's weird because I downloaded Marlin circa 10 times in past two days, I also reinstalled Vscode a few times :?
I will try to download Marlin one more time but I don't think it will suceed

ellensp, 2021-07-18:
Code you posted has nothing to do with the error in your Configuration.h

Alchoni, 2021-07-18:
What am I supposed to post then, to help get forward ?

ellensp, 2021-07-18:
Please attach your configuration files.
NB Zip them up and attach them.

ellensp, 2021-07-18:
no reply, so closing and advising where you should be going to get general home. 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

Alchoni, 2021-07-18:
Jesus, I am devastated by my stupidity, here's solution by one of the Discord members: 

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_SKR_PRO_V1_2
#endif

I had accidentally deleted "MOTHERBOARD" it appears... Thank you all

ellensp, 2021-07-18:
quoting 5 hours ago...
"It should say
#define MOTHERBOARD BOARD_BTT_SKR_PRO_V1_2"