MarlinFirmware/Marlin #22390: [BUG] Bugfix-2.0.x v020008 - Random Freezes on BTT E3 V2.0 / Ender 3 V2 / Hemera

RipperDrone, 2021-07-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

This might be related to https://github.com/MarlinFirmware/Marlin/issues/21010 and previous freeze issues reported here:

I am getting freezes irrespective of 

- USB connected to PC or not 
- TMC_Debug on or off
- Trinamic MONITR_DRIVER_STATUS on/off

Freeze can happen anytime, during probing or during printing. Had it once oeven before start of axes movements!

Looks it's getting worse with leaving the printer on for longer, so was suspecting a temperature issue. However, fans are all running, and steppers feel warm but not superhot even when running at higher speeds.

Attaching my config here:

### Bug Timeline

Started after replacing BTT E3 Mini 2.0 for the stock Creality 4.2.2 MB

### Expected behavior

Complete prints

### Actual behavior

Random freezes

### Steps to Reproduce

1. Flash firmware containing the configs uploaded further down to printer
2. Load xyz cube in Cura
3. Print over USB OR copy to SD card on PC and run print job from SD card via printer menu
4. wait and hope
5. meet the freeze randomly

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Creality Ender 3 V2

### Electronics

BTT E3 Mini v2.0

### Add-ons

Hemera extruder, BLTouch - not likely to be involved though

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6836094/Marlin.zip)

arminth, 2021-07-18:
I also started to get random freezes and TMC driver errors onmy skr14 turbo and BTT TFT 3.5 E3 display.
I have switched  both baudrates ffrom 250000 to 115500. Then my print finished. Maybe, you can try this.

Cheers
Armin

RipperDrone, 2021-07-18:
@arminth thank you for this hint, but I'm afraid my serial comm is running at 115k bake already and still fails/hangs 😒

CRCinAU, 2021-07-19:
Unless you're using the hardware UART, the baud rate means nothing. The USB controller of these boards runs at USB speeds - not any serial speed. That means it tops out at about 1.1Mbit.

When you're talking about freezes, are you talking about pauses in print, controller hangs, or something else?

I run a BTT E3 V2.0 and haven't seen that issue. Which screen are you guys using?

RipperDrone, 2021-07-19:
@CRCinAU Using the Creality stock Ender **3V2** LCD screen which has a serial comm to the mainboard.

Freeze means fans continue to spin, but print stops, printer does not respond to inputs over jog dial wheel anymore... needs cold reboot.

CRCinAU, 2021-07-19:
So this is using the custom cable to connect the stock DWIN based screen to the SKR board?

RipperDrone, 2021-07-19:
No of course not, it is re-mapped to correct Pinot via breadboard jumper cable. Works beautifully. If printer didn't freeze, that is 😄

CRCinAU, 2021-07-19:
That's fine - I'd add that to your original request that you're using the DWIN screen etc - as I know that this isn't a very well tested combination.

RipperDrone, 2021-07-19:
> @CRCinAU Using the Creality stock Ender **3V2** LCD screen which has a serial comm to the mainboard.
> 
> Freeze means fans continue to spin, but print stops, printer does not respond to inputs over jog dial wheel anymore... needs cold reboot.

SKR Mini E3 2.0 is using hardware serial AFAIK...

RipperDrone, 2021-07-19:
> Unless you're using the hardware UART, the baud rate means nothing. The USB controller of these boards runs at USB speeds - not any serial speed. That means it tops out at about 1.1Mbit.
>

SKR E3 2.0 IS using hardware UART afaik....

CRCinAU, 2021-07-19:
The USB port is not, the screen connector is a hardware UART.

RipperDrone, 2021-07-19:
Ruling out a hardware issue next, ordered a replacement mainboard to test. BTT support suspects the display cable, maybe these ribbon cables catch a lot of noise. The jumper cable extension with an extra 200mm length needed to be routed inside the mainboard compartment, maybe this is part of the issue on the hardware side...

thisiskeithb, 2021-07-20:
Does your SKR Mini E3 V2 have an STM32F103RCT6 or RET6 processor? Which environment are you compiling with?

RipperDrone, 2021-07-20:
103RC, this env used is STM32F103RC_btt

thisiskeithb, 2021-07-20:
The configs you attached are from 2.0.8 and do not build with the latest bugfix. Can you attach the latest version?

Also, please confirm that you flashed & tested with the latest bugfix so we're not chasing old bugs.

RipperDrone, 2021-07-20:
No need to test software any further before I will have the final answer to hardware vs software flaw. Since freezes occurred irrespective of LCD display connected or not, USB cable to PC connected or not, more frequent as printer gets warmer etc. I am suspecting defective hardware for now and will try a replacent mainboard next. Test has been with 2.0.8 bugfix so far, next I can test with latest version - as soon as I will have the new mainboard in my hands (pbly next week)

blazewicz, 2021-07-21:
I'll just add here that I use the same combo i.e. Ender 3 V2, DWIN Display, BTT SKR Mini E3 V2.0 and I haven't seen such issue. My board has GD32F103RET6 MCU. I use `STM32F103RE_btt_USB` env.

I have other issues though:

* serial connection over USB is not stable, breaks after few hours
* sometimes knob push event is registered even though I haven't touched anything
* sometimes I get random z-offset which ruins first layer

This combo doesn't seem that stable after all.

CRCinAU, 2021-07-21:
@blazewicz I seem to remember a while back, there was issues with having both USB data and serial multiplexed over the one link.

I would be very interested in testing and see if you see the same problems with the `STM32F103RE_btt` env.

marcwingduck, 2021-07-23:
I am experiencing similar issues during probing and printing using the same setup. Though in my case the board does not freeze, it soft resets. The fans turn off, the progress bar appears on the screen and after it has disappeared I can control the printer again. In the serial output the same notifications appear as if I would have switched the printer on normally. There are no error messages.

I am using target STM32F103RC_btt according to my MCU. Next to the latest bugfix_2.0.x, I also compiled and tested the official BTT fork. I have also tried both versions of the firmware with disconnected screen, initializing prints from the SD card via serial connection. Also, I am already using the second board, a different SD card, and unfortunately, the issue persists.

With the second board not working, I am running out of ideas on how to further isolate the problem.

RipperDrone, 2021-07-23:
Hmm, has anyone tried to supply 24V DC to the mainboard from an external power source, just to check if we can rule out a PSU issue / voltage drop?

RipperDrone, 2021-07-23:
Alternatively, I start worrying if my silent fan mods (including replacing the metal mainboard and PSU housings by 3D printed plastic ones) might have a detrime talk EMI effect on the circuit boards, picking up too much EMI noise from e.g. PWM operated fans or heaters?

CRCinAU, 2021-07-23:
I can't say as to causes, but I run a BTT E3 Mini v2.0 board - but with the stock text screen as well as the TFT35. I haven't seen anything like the problems stated.

RipperDrone, 2021-07-23:
@CRCinAU So it does work if done correctly and hardware is ok, it seems 👍. Can you pls post so e details on our config?

- Marlin version
- build environment in platform.io
- settings for USB / serial ports
- any changes in HAL libs etc?

CRCinAU, 2021-07-23:
I use the latest bugfix-2.0.x - the STM32F103RC_btt env, everything else is mostly defaults.

There is a ton of changes to my hardware, but that's all stuff like thermistor types and enabling PID etc along with MIN_X_POS etc etc - which I wouldn't think would make any difference on this at all...

RipperDrone, 2021-07-23:
Got it. Your 'stock text screen' refers to (stock Ender3V2) Creality DWIN screen or to an Ender 3 stock screen (non DWIN). Your printer is an Ender 3 (non-V2) then?

CRCinAU, 2021-07-23:
Yeah - the Ender 3 Pro screen - I forget its definition at the moment....

RipperDrone, 2021-07-23:
@marcwingduck The more users post their setups, the more I'm getting the impression that the Ender 3 He's Creality DWIN LCD display is messing things up (serial ports, USB communication, timer/IRQ racing, whatever it is).

To narrow things further down, could you test a Marlin version compiled without setting the serial port up for the DWIN display serial2 left at default I stead of setting it to 1 for DWIN display) and serving g-code print comma es over USB? Just unplugging the display cable might still leave the serial communication faulty...

I will pbly get my replacement E3 Mini within next 2-3 days, can then re-engage into testing as well...

marcwingduck, 2021-07-23:
@RipperDrone The BTT firmware I used had the SERIAL_PORT_2 set to -1, which is default I assume. But I can check if the issue persists with the latest bugfix-2.0.x on Sunday or Monday!

A little more info about my setup:

I also have the silent mod, and thus printed main board and PSU covers using Noctua fans. The voltage regulator I used is a MP1584EN. I am using a Hydra direct drive mod, also with a Noctua fan and two Winsinn blower fans instead of the original one.
From day one I had issues with the LCD, where the rotary encoder (or some interferences along the wires) issued ghost enter presses during prints. This problem disappeared at a completely random point.

Further observations regarding the issue:

Since a soft reset screams for undervoltage, I thought it may occur when the bed or nozzle start heating, or even when the blower fans kick in, but as you already mentioned, the stops seem completely random.
What speaks against randomness though: Using the BTT firmware, the printer stopped two out of three times at the same point in a print (the third time it stopped just 10 mm before that point). I am going to test this print again using the latest bugfix-2.0.x within the next days.

looxonline, 2021-07-23:
> @RipperDrone The BTT firmware I used had the SERIAL_PORT_2 set to -1, which is default I assume. But I can check if the issue persists with the latest bugfix-2.0.x on Sunday or Monday!
>
> A little [...]

A soft reset is often caused by a timeout when reading from the SD card. Try starting a print and then manually removing the SD card and you should be able to replicate it fairly easily. With this in mind you may want to try a variety of different cards and also be sure to format them using the official SD card formatting tool (google that phrase and it should be the first link). Since your board also had ghost presses that disappeared without a reason I would say that you may have an EMI issue. Check that your machine has a good Earth connection and check that you are Earthing the chassis. Also check that your wires to and from the PSU are well secured.

There are HAL changes going on at the moment and more than that there is a major shift from the maple framework to the STM core framework but I have been running dev builds which have been working great regardless of the changes (aside from the lingering USB issues with the GD variant).

blazewicz, 2021-07-24:
> @blazewicz I seem to remember a while back, there was issues with having both USB data and serial multiplexed over the one link.
>
> I would be very interested in testing and see if you see the same problems with the STM32F103RE_btt env.

@CRCinAU I changed my env to `STM32F103RE_btt` (with Arduino fix for USB single buffer) and it seems to be stable now - no issues with ~5h print, I'll test it on something longer tonight.

> The BTT firmware I used had the SERIAL_PORT_2 set to -1, which is default I assume. But I can check if the issue persists with the latest bugfix-2.0.x on Sunday or Monday!

@marcwingduck `SERIAL_PORT_2` won't work with DWIN display due to #22299. You need to set `SERIAL_PORT` to -1 and leave `SERIAL_PORT_2` undefined.

RipperDrone, 2021-07-24:
Alrighty, I'm back in the game. Just installed my replacement BTT E3 Mini 2.0 board, flashed actual bugfix Marlin version with Ender 3V2 configs from examples directory, configs adopted for SKK board acc. changes by @blazewicz: https://github.com/MarlinFirmware/Configurations/issues/535.

(Then manually edited for some Hemera extruder magic which is not the point here)

Mainboard-to-display cable pins re-mapped to match CREALITY_DWIN_LCD display - display works ok.
My replacement board has the STM32F103RC chip affirmatively - so says the chip ink print at least ;-).

env = STM32F103RC_btt_USB: Did compile with a minor warning (auto assignment not suitable in context)
env = STM32F103RC_btt: Did NOT compile w/o errors

Firmware seems fully functional, bedsize and offset need some trimming. Then have to get into testing some prints... :-)

Is there anything I should consider in the current state of Marlin (I saw many fragments of Jyers enhancements are being implemented right now, remixed with 2 more devs' DWIN enhancements, as it seems - confusing for me as a coding noob to guess if in the current transition phase there is a stable enough Marlin bugfix version available that is worthwhile stress testing for USB / serial freezes due to IRQ racing or other timing issues in software OR hardware issues with the BTT mainboard we had observed. As things stand right now, there seems to be some of you (particularly @blazewicz ) who are getting more and more happy with the Ender 3V2 + DWIN LCD + SKR E3 MINI board - maybe we have overcome the teethign issues now!? :-)

RipperDrone, 2021-07-24:
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/6873025/Configurations.zip)

Here's my config files, just for ref :-)

RipperDrone, 2021-07-24:
1st observation: Part fan always running @100%, cannot be tuned down via menu anymore. HAve the two fan terminals been swapped on the BTT MINI E3 board in the meantime? It used to be the upper terminal 'FAN1', is it now back to 'Fan0'? Or did I miss a software switch somewhere? 

[image]

RipperDrone, 2021-07-24:
> The configs you attached are from 2.0.8 and do not build with the latest bugfix. Can you attach the latest version?
> 
> Also, please confirm that you flashed & tested with the latest bugfix so we're not chasing old bugs.

@thisiskeithb catching up now - so answer to both your questions is YES now:
- STM32F103RC chip
- used latest bugfix

will have to find time and test longer prints now - very short tests have successfully finished. If it stays like this, it must have been a faulty mainboard - the replacement board seems to be good so far.

blazewicz, 2021-07-25:
My printer just completed a 8.5h print from Octopi. In total I did over 20h over USB on this setup without any issues.

Once again, my setup:

* BTT SKR Mini E3 V2.0 with GD32F103RET6 chip
* stock Creality DWIN display
* env: `STM32F103RE_btt`
* Marlin `bugfix-2.0.x` from  22-07-2021
* config as in https://github.com/MarlinFirmware/Configurations/issues/535
* Arduino framework modified to use single buffer for USB

RipperDrone, 2021-07-25:
Have done ~20 prints now, never seen a single freeze anymore. Seems fixed with replacement mainboard - latest Marlin bugfix running stable, staying with STM*_btt environment for now. :-)

Thank you for all the good hints here! Closing this issue now...

closed 2021-07-25 by RipperDrone