MarlinFirmware/Marlin #22392: [BUG] Thermal Probe Compensation

martins6413, 2021-07-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

	 "'_btc_sample_res' does not name a type; did you mean '_ptc_sample_res'?",
	  "Der explizite Typ fehlt (\"int\" wird angenommen).",
        "'_btc_sample_res' was not declared in this scope; did you mean '_ptc_sample_res'?",
	

### Bug Timeline

Marlin 2.0.7 funktioniert

### Actual behavior

Firmware wird nicht kompiliert

### Version of Marlin Firmware

Marlin 2.0.9.1

### Electronics

GTR V1.0

### Additional information & file uploads

Verwende letzte Version 2.0.9.1 (mit original Prusa Pinda V2).  Funtioniert in 2.0.7 einwandfrei.

slowbro, 2021-07-19:
Looks like it's missing an `auto` - submitting a PR now. 

```diff
diff --git a/Marlin/src/inc/SanityCheck.h b/Marlin/src/inc/SanityCheck.h
index ae4aba14db..9ced6e4e42 100644
--- a/Marlin/src/inc/SanityCheck.h
+++ b/Marlin/src/inc/SanityCheck.h
@@ -623,7 +623,7 @@ static_assert(COUNT(arm) == LOGICAL_AXES, "AXIS_RELATIVE_MODES must contain " _L
     static_assert(_test_btc_sample_start != 12.3f, "BTC_SAMPLE_START must be a whole number.");
   #endif
   #ifdef BTC_SAMPLE_RES
-    constexpr _btc_sample_res = BTC_SAMPLE_RES;
+    constexpr auto _btc_sample_res = BTC_SAMPLE_RES;
     constexpr decltype(_btc_sample_res) _test_btc_sample_res = 12.3f;
     static_assert(_test_btc_sample_res != 12.3f, "BTC_SAMPLE_RES must be a whole number.");
   #endif
```

martins6413, 2021-07-19:
Habe es geändert und mit G76 getestet, funktioniert perfekt.
Danke für die Hilfe .

closed 2021-07-19 by martins6413